LuckfoxTECH/luckfox-pico #112: Guide: Connecting to RV1106 MCU JTAG

luyi1888, 2024-05-21:
Please add documentation tag.

The MCU use cJTAG protocol. So you need J-Link v10+ or something other.
If you don't have them, I have a project which use Raspberry Pi Pico as cJTAG debugger. 
https://github.com/luyi1888/pico-dirtyJtag
OpenOCD repo: https://github.com/luyi1888/openocd

**Step1: pinout**
You will lost UART2, which is default console.
TCKC - GPIO1_B2_d/UART2_TX
TMSC - GPIO1_B3_u/UART2_RX

**Step2: make sure your mcu is running**
`./mcuload execute 0xff6c0000 rv1106-mcu-led-blinky.bin`

**Step3: set pinmux to hpmcu jtag**
`devmem 0xff538008 32 0xff003300`

**Step4: connect to debugger**
```
ubuntu@ubuntu-virtual-machine:~/openocd/tcl$ sudo ../src/openocd   -f interface/dirtyjtag-cjtag.cfg -f target/rv1106-mcu.cfg 
Open On-Chip Debugger 0.12.0+dev-g22ffe21b7 (2024-05-21-19:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : DirtyJTAG: cJTAG protocol set to 1
sw_reset_halt
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at very low speed (100 kHz).
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Info : dirtyjtag version 2
Info : clock speed 100 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0xdeb11001 (mfg: 0x000 (<invalid>), part: 0xeb11, ver: 0xd)
Info : datacount=2 progbufsize=6
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40001104
Info : [riscv.cpu] Examination succeed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections

```

bwooster0, 2024-12-29:
is mcuload / devmem part of uboot?

Thanks for any help.

yichen115, 2025-01-09:
> is mcuload / devmem part of uboot?
> 
> Thanks for any help.

你可以在 https://github.com/luyi1888/rv1106-mcu 找到 mcutool.c 和 Makefile，使用 https://github.com/deerpi/arm-rockchip830-linux-uclibcgnueabihf 编译，在 luckfox 的 buildroot 镜像运行：./mcutool execute 0xff6c0000 rv1106-mcu-led-blinky.bin。devmem 可以直接在 linux 下运行

skyandray, 2025-08-20:
> rv1106-mcu-led-blinky.bin
这个BIN文件的DEMO工程有吗？