LuckfoxTECH/luckfox-pico #113: Using the Debian image how do I disable RNDIS so I can use a USB Wifi Adapter

jockm, 2024-05-22:
I want to apply power to VBUS, login via serial, and use a USB WiFi adapter for a network connection.  To do this I have do disable RNDIS.  I tried `systemctl disable usb-gadget-rndis` but got the message:

> Failed to disable unit: Unit file usb-gadget-rndis.service does not exist.

so clearly RNDIS is being enabled in a different way.

PJ5168, 2024-05-29:
Where is the Debain image?
The wiki told you to rebuild the kernel....again...

jockm, 2024-05-29:
@PJ5168 If you look at the bottom of [this page](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-prepare/) you will find the Google Cloud Link.  It says ubuntu, but U is debian derivied.  

I ended up getting an answer on [the forum](https://forums.luckfox.com/viewtopic.php?p=2047&hilit=rndis#p2047):

> Hello, if you need to disable RNDIS, please modify RNDIS_EN=on to RNDIS_EN=off in /etc/init.d/S50usbdevice on the Luckfox Pico, and reboot the system to apply the settings.
> 
> If you need to use a WIFI USB Adapter, you also need to change the USB mode. The system sets the USB mode to peripheral by default, and you need to modify the device tree to switch the USB to Host mode. For details, please refer to the wiki: https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-USB/.

closed 2024-05-29 by jockm

PJ5168, 2024-05-30:
@jockm Sure, I knew that.
As I said, for Luckfox Pico Max, without device tree overlay, you need to keep building and flashing.... for any change/test you want to make.
For Milk-V Duo256 with Debian image v1.3.0, you can use USB gadget to turn off the RNDIS and use USB WiFi.
For LicheeRV nano, there is build-in WiFi model available. Sure, you can use USB WiFi as well.
And there is Ubuntu 22.04 v0.0.4 image available for Milk-V and LicheeRV nano, too.