LuckfoxTECH/luckfox-pico #115: openWRT for luckfox series initial support

pengphei, 2024-05-28:
After some hack for openwrt, now I can launch openwrt for luckfox rv1103/rv1106 series boards, features as below:

- [x] sd image generate (based on glibc openwrt)
- [x] sd card boot up 
- [x] **kmpp** kernel module build, **mpp-interface** branch of **rkmpp** build ok.
- [x] **librga** library build
- [ ] **camera_engine_rkaiq** built ok, but segment always

pengphei, 2024-05-28:
[openwrt-rockchip-cortexa7-luckfox_pico-max-ext4-rootfs.zip](https://github.com/LuckfoxTECH/luckfox-pico/files/15466942/openwrt-rockchip-cortexa7-luckfox_pico-max-ext4-rootfs.zip)

openwrt image for **luckfox pico pro/max**

shawnchain, 2024-06-18:
Goo job!

plan44, 2024-06-18:
> [openwrt-rockchip-cortexa7-luckfox_pico-max-ext4-rootfs.zip](https://github.com/LuckfoxTECH/luckfox-pico/files/15466942/openwrt-rockchip-cortexa7-luckfox_pico-max-ext4-rootfs.zip)
> 
> openwrt image for **luckfox pico pro/max**

Does this work for Luckfox pico max with NAND flash (no SD card), too?

plan44, 2024-06-20:
Hi @pengphei, I'm trying to build openwrt for mtd (nand flash), in squashfs/jffs2 configuration, but I am not successful yet ;-) I am quite used to OpenWrt builds, but only for the MT7688 and RPi targets. I do not yet understand the specifics of the RV110x build, in particular the uboot parts.

I'd be very much interested to help in further developing OpenWrt for the luckfox rv1103/rv1106 series boards, but I don't know where to find more information. Any ideas how to proceed best?

plan44, 2024-06-27:
In the meantime, I managed to build trough OpenWrt 22.03 for the RV1106 and creating `openwrt-rockchip-cortexa7-luckfox_pico-max-squashfs-boot.img` and `openwrt-rockchip-cortexa7-luckfox_pico-max-squashfs-rootfs.img`. The former seems to be the kernel, it starts with a device tree according to `binwalk`, the latter is the squashfs rootfs.

What I am stuck with now is how to install these two in the NAND flash of my LuckFox Pico Max. The entire setup seems different from the mtd setup I know from MT7688, where kernel and squashfs are baked together as an uimage.

Any hints or pointers where to read up? Thx in advance!

plan44, 2024-07-08:
Again two weeks and some experiments later 😉 

Today I finally managed to get `SocToolkit.exe` running on an old i86 Windows box (unfortunately, it does not run on any ARM based machines, and neither does `upgrade_tool`), and could flash the boot and rootfs partition with the kernel/rootfs as produced by the openwrt built.

However, there seems to be a mismatch between the kernel config and the squashfs/jffs2 rootfs image.

I found that CONFIG_MTD_SPI_NAND and CONFIG_MTD_PHYSMAP are required, but apparently something else is missing, because when I start from uboot command line with:

```
setenv sys_bootargs " loglevel=7 root=/dev/mtdblock6 rootfstype=squashfs,jffs2"
boot
```

...the boot log shows that the kernel never even attempts to use mtd and start the mtd partition scanners that would create the /dev/mtdblockxx devices. So the kernel just hangs waiting for `/dev/mtdblock6`.

plan44, 2024-07-08:
Just for context, below is the complete bootlog starting in the uboot console. I would expect `mtd` to parse the partitions around `[0.389703]` but it does not happen. I must be missing some essential detail here 😢 

```
=>     setenv sys_bootargs " loglevel=7 root=/dev/mtdblock6 rootfstype=squashfs,jffs2"
=>     printenv
arch=arm
autoload=no
baudrate=115200
board=evb_rv1106
board_name=evb_rv1106
bootargs=storagemedia=mtd androidboot.storagemedia=mtd androidboot.mode=normal 
bootcmd=boot_fit;boot_android ${devtype} ${devnum};
bootdelay=0
cpu=armv7
devnum=1
devtype=mtd
eth1addr=26:93:a5:5a:0e:d7
ethaddr=22:93:a5:5a:0e:d7
fdt_addr_r=0x00c00000
kernel_addr_c=0x00808000
kernel_addr_r=0x00008000
mtdparts=spi-nand0:256K(env),256K@256K(idblock),512K(uboot),4M(boot),30M(oem),10M(userdata),210M(rootfs)
pxefile_addr_r=0x00c00000
ramdisk_addr_r=0x000e00000
rkimg_bootdev=if mmc dev 1 && rkimgtest mmc 1; then setenv devtype mmc; setenv devnum 1; echo Boot from SDcard;elif mmc dev 0; then setenv devtype mmc; setenv devnum 0;elif mtd_blk dev 0; then setenv devtype mtd; setenv devnum 0;elif mtd_blk dev 1; then setenv devtype mtd; setenv devnum 1;elif mtd_blk dev 2; then setenv devtype mtd; setenv devnum 2;elif rknand dev 0; then setenv devtype rknand; setenv devnum 0;elif rksfc dev 0; then setenv devtype spinand; setenv devnum 0;elif rksfc dev 1; then setenv devtype spinor; setenv devnum 1;else;setenv devtype ramdisk; setenv devnum 0;fi; 
scriptaddr=0x00b00000
serial#=31d1f0e3a3b8bdbb
soc=rockchip
stderr=serial
stdin=serial
stdout=serial
sys_bootargs= loglevel=7 root=/dev/mtdblock6 rootfstype=squashfs,jffs2
vendor=rockchip

Environment size: 1368/262140 bytes
=>     boot
## Booting FIT Image at 0xe8cb080 with size 0x0032ec00
Fdt Ramdisk skip relocation
No misc partition
## Loading kernel from FIT Image at 0e8cb080 ...
   Using 'conf' configuration
## Verified-boot: 0
   Trying 'kernel' kernel subimage
     Description:  unavailable
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x0e8d5280
     Data Size:    3295296 Bytes = 3.1 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x00008000
     Entry Point:  0x00008000
     Hash algo:    sha256
     Hash value:   2f58d0727960528255e2e204a38803ee38926374dfc8668db2b9c1608f32742d
   Verifying Hash Integrity ... sha256+ OK
## Loading fdt from FIT Image at 0e8cb080 ...
   Using 'conf' configuration
   Trying 'fdt' fdt subimage
     Description:  unavailable
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x0e8cb880
     Data Size:    39256 Bytes = 38.3 KiB
     Architecture: ARM
     Load Address: 0x00c00000
     Hash algo:    sha256
     Hash value:   f4a9e232ea144a4f5168e9fc7144a76b757dc75cc859ad784d26115d98b90dac
   Verifying Hash Integrity ... sha256+ OK
   Loading fdt from 0x0e8cb080 to 0x00c00000
   Booting using the fdt blob at 0x00c00000
[... 78 lines cut ...]
[    0.425054] /usbdrd: Failed to get clk index: 0 ret: -517
[    0.432595] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.440270] ehci-platform: EHCI generic platform driver
[    0.446461] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    0.453564] ohci-platform: OHCI generic platform driver
[    0.460360] usbcore: registered new interface driver cdc_acm
[    0.467145] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[    0.476579] i2c /dev entries driver
[    0.483627] softdog: initialized. soft_noboot=0 soft_margin=60 sec soft_panic=0 (nowayout=0)
[    0.494324] softdog:              soft_reboot_cmd=<not set> soft_active_on_boot=0
[    0.503743] Synopsys Designware Multimedia Card Interface Driver
[    0.514869] NET: Registered protocol family 10
[    0.522627] Segment Routing with IPv6
[    0.526834] NET: Registered protocol family 17
[    0.531928] 8021q: 802.1Q VLAN Support v1.8
[    0.537853] /usbdrd: Failed to get clk index: 0 ret: -517
[    0.545405] /usbdrd: Failed to get clk index: 0 ret: -517
[    0.551674] rk3x-i2c ff470000.i2c: deferred probe timeout, ignoring dependency
[    0.560169] dwmmc_rockchip ffaa0000.mmc: deferred probe timeout, ignoring dependency
[    0.569512] ALSA device list:
[    0.572832]   No soundcards found.
[    0.576642] Warning: unable to open an initial console.
[    0.582531] Waiting for root device /dev/mtdblock6...

```

pengphei, 2024-07-10:
Actually, openwrt use some hack patches for squashfs/jffs/ubifs overlay support.
1. patch the rk kernel with overlay referring patches from *target/linux/generic/pending-5.15/hacking-5.15*
2. remove **rootfs=** env for boot environments, and  the openwrt will detectes overlay automaticaly, which have stoped me from booting for a very long time.

have a good luck!

pengphei, 2024-07-10:
> > [openwrt-rockchip-cortexa7-luckfox_pico-max-ext4-rootfs.zip](https://github.com/LuckfoxTECH/luckfox-pico/files/15466942/openwrt-rockchip-cortexa7-luckfox_pico-max-ext4-rootfs.zip)
> > openwrt image for **luckfox pico pro/max**
> 
> Does this work for Luckfox pico max with NAND flash (no SD card), too?

Yes, I have made import some patches from "openwrt/target/linux/generic" and make this work.

1. first we have a openwrt image that bootup from external sd flash
2. then we build seperate parts of nand firmware binary
3. then we burn firmware to  the nand flashes from sd flash system
4. we remove sd and boot from nand system
 
that's all the steps, but I need time to clean my openwrt repo.

shawnchain, 2024-07-10:
Thanks for the initial works by @pengphei .

The openwrt ubi patches are required for  boot from nand flash with BSP kernel, and rk-uboot also requires additional hack to support dual-boot from spi-nand or sd/emmc.

Anyway it's a huge work to port ```22.03/target/linux/generic/pending-5.10``` to RK BSP kernel, despiting the uclibc issues on RK commercial libs.

pengphei, 2024-07-10:
> Thanks for the initial works by @pengphei .
>
> The openwrt ubi patches are required for boot from nand flash with BSP kernel, and rk-uboot also requires additional hack to support dual-boot from [...]

1. openwrt kernel patches for rk bsp kernel has been finished, anyway very ugly ;)
2. uclibc for openwrt to support binary compability with rk libs has been finished yet, also ugly ;)
3. I have made a feed for rockchip, especilly ugly ;(

that's why I need some time to clean my repo.

pengphei, 2024-07-10:
> Just for context, below is the complete bootlog starting in the uboot console. I would expect `mtd` to parse the partitions around `[0.389703]` but it does not happen. I must be missing some essential detail here 😢

1. please use the patched kernel **https://github.com/ticklab/rockchip-kernel/tree/openwrt-ticklab-5.10**
2. please config kernel with overlay/suqashfs/ubifs support
3. remove **rootfs=""** env

with the last kernel, it will boot.

plan44, 2024-07-12:
Hi @pengphei, thanks a lot for your help! With this, it *almost* work, but not quite yet (ubifs problem, see below)

I forked your patched kernel, because I had to [add a few fixes on top](https://github.com/plan44/rockchip-kernel/commits/openwrt-p44-5.10/) to make it build in my environment - macOS on M2 mac.

Then, I rebased the changes from your openwrt branch `rockchip-22.03` on the latest openwrt 22.03 commit having kernel 5.10.160 to make sure openwrt and kernel are compatible, in a new branch [p44-rockchip-rebased](https://github.com/plan44/openwrt/commits/p44-rockchip-rebased/).

With this, I applied config to use the external kernel:

```
CONFIG_KERNEL_GIT_CLONE_URI="git@github.com:plan44/rockchip-kernel.git"
CONFIG_KERNEL_GIT_LOCAL_REPOSITORY=""
CONFIG_KERNEL_GIT_MIRROR_HASH=""
CONFIG_KERNEL_GIT_REF="openwrt-p44-5.10"
```

This now builds 😄 , and when flashed to "boot" and "rootfs" areas on luckfox pico max the kernel starts😄 , correctly reads mtd 😄 , but then fails on attaching ubifs 😢 

```
...

[    1.388037] ubi0: attaching mtd6
[    1.595318] ubi0: scanning is finished
[    1.599195] ubi0 error: ubi_read_volume_table: the layout volume was not found
[    1.606747] ubi0 error: ubi_attach_mtd_dev: failed to attach mtd6, error -22
[    1.613832] UBI error: cannot attach mtd6

...
```

I guess I have something wrong in my changes in `target/rockchip/image/cortexa7.mk` to produce the ubifs rootfs. 

full boot log:

```
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.160 (luz@MBA15-2023.local) (arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 11.2.0 r0+19961-22333a94a0) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 PREEMPT Fri Jul 12 17:59:35 2024
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=50c53c7d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Luckfox Pico Max
[    0.000000] earlycon: uart8250 at MMIO32 0xff4c0000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 67584 KiB at 0x0be00000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: user_debug=31 storagemedia=mtd androidboot.storagemedia=mtd androidboot.mode=normal  earlycon=uart8250,mmio32,0xff4c0000 console=ttyS2,115200 mtdparts=spi-nand0:256K(env),256K@256K(idblock),512K(uboot),4M(boot),30M(oem),10M(userdata),210M(rootfs) ubi.mtd=6 root=ubi0:rootfs rootfstype=ubifs rk_dma_heap_cma=66M
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 184348K/262144K available (4994K kernel code, 277K rwdata, 1688K rodata, 192K init, 122K bss, 10212K reserved, 67584K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 30 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000008] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.008791] Switching to timer-based delay loop, resolution 41ns
[    0.015829] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=80000)
[... 161 lines cut ...]
[    1.679181] rockchip-suspend not set sleep-mode-config for mem-ultra
[    1.685546] rockchip-suspend not set wakeup-config for mem-ultra
[    1.693019] ALSA device list:
[    1.696078]   #0: rv-acodec
[    1.699095] dw-apb-uart ff4c0000.serial: forbid DMA for kernel console
[    1.705855] VFS: Cannot open root device "ubi0:rootfs" or unknown-block(31,6): error -19
[    1.713991] Please append a correct "root=" boot option; here are the available partitions:
[    1.722366] 1f00             256 mtdblock0 
[    1.722370]  (driver?)
[    1.728923] 1f01             256 mtdblock1 
[    1.728926]  (driver?)
[    1.735473] 1f02             512 mtdblock2 
[    1.735477]  (driver?)
[    1.742023] 1f03            4096 mtdblock3 
[    1.742026]  (driver?)
[    1.748569] 1f04           30720 mtdblock4 
[    1.748572]  (driver?)
[    1.755116] 1f05           10240 mtdblock5 
[    1.755119]  (driver?)
[    1.761661] 1f06          215040 mtdblock6 
[    1.761665]  (driver?)
[    1.768206] b300         7707648 mmcblk1 
[    1.768210]  driver: mmcblk
[    1.775013] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,6)
```

plan44, 2024-07-12:
@pengphei: just found the problem 😄 , need to use `append-ubi`, not `append-rootfs` for correctly generating ubi/ubifs. See [latest commit](https://github.com/plan44/openwrt/commit/55a369e5172920a8ddd9b6a58763eba493118f80).

pengphei, 2024-07-12:
> @pengphei: just found the problem 😄 , need to use `append-ubi`, not `append-rootfs` for correctly generating ubi/ubifs. See [latest commit](https://github.com/plan44/openwrt/commit/55a369e5172920a8ddd9b6a58763eba493118f80).

good luck, later I will update the uclibc referring patches, but it will have many patches including openwrt, and its referring packages.

plan44, 2024-07-18:
> good luck
Thanks 😄 

It boots with all of rootfs as ubifs, but I'd like to use the "squashfs+ubifs overlay in ubi" variant and am stuck with that - I just don't understand yet why it does not work, and now hope [someone over at the openwrt forum](https://forum.openwrt.org/t/creating-and-booting-ubinized-squashfs-ubifs-for-new-device/204368) will have an idea.

plan44, 2025-05-20:
Hi @pengphei , after a long pause I am trying to pick this up and get OpenWrt running on LuckFox. The build works fine an produces `*-rockchip-cortexa7-luckfox_pico-max-ubifs-rootfs.img`and `*-rockchip-cortexa7-luckfox_pico-max-ubifs-boot.img`.
Flashing these two into boot and rootfs partition of the luckfox starts the kernel, but whatever I try, **I could not figure out what I need to have in uboot env such that the kernel can find and mount the rootfs**.

The ubootenv is:

```uboot
arch=arm
autoload=no
baudrate=115200
blkdevparts=mmcblk0:32K(env),512K@32K(idblock),256K(uboot),32M(boot),512M(oem),256M(userdata),6G(rootfs)
board=evb_rv1106
board_name=evb_rv1106
bootargs=storagemedia=emmc androidboot.storagemedia=emmc androidboot.mode=normal 
bootcmd=boot_fit;boot_android ${devtype} ${devnum};
bootdelay=0
cpu=armv7
devnum=0
devtype=mmc
eth1addr=0a:1a:e3:2c:8a:6c
ethaddr=06:1a:e3:2c:8a:6c
fdt_addr_r=0x00c00000
kernel_addr_c=0x00808000
kernel_addr_r=0x00008000
pxefile_addr_r=0x00c00000
ramdisk_addr_r=0x000e00000
rkimg_bootdev=if mmc dev 1 && rkimgtest mmc 1; then setenv devtype mmc; setenv devnum 1; echo Boot from SDcard;elif mmc dev 0; then setenv devtype mmc; setenv devnum 0;elif mtd_blk dev 0; then setenv devtype mtd; setenv devnum 0;elif mtd_blk dev 1; then setenv devtype mtd; setenv devnum 1;elif mtd_blk dev 2; then setenv devtype mtd; setenv devnum 2;elif rknand dev 0; then setenv devtype rknand; setenv devnum 0;elif rksfc dev 0; then setenv devtype spinand; setenv devnum 0;elif rksfc dev 1; then setenv devtype spinor; setenv devnum 1;else;setenv devtype ramdisk; setenv devnum 0;fi; 
scriptaddr=0x00b00000
serial#=a9a1317e3e29c315
soc=rockchip
stderr=serial
stdin=serial
stdout=serial
sys_bootargs= root=/dev/mmcblk0p7 rootfstype=ext4 rk_dma_heap_cma=1M
vendor=rockchip
```

in uboot, `part list mmc 0` gives:

```
Partition Map for MMC device 0  --   Partition Type: ENV

Part	Start LBA	Size		Name
  1	0x00000000	0x00000040	env
  2	0x00000040	0x00000400	idblock
  3	0x00000440	0x00000200	uboot
  4	0x00000640	0x00010000	boot
  5	0x00010640	0x00100000	oem
  6	0x00110640	0x00080000	userdata
  7	0x00190640	0x00400000	rootfs
```

With that, boot fails with:

```
[    1.216927] VFS: Cannot open root device "mmcblk0p7" or unknown-block(0,0): error -6
[    1.224708] Please append a correct "root=" boot option; here are the available partitions:
[    1.233071] 0100            4096 ram0 
[    1.233074]  (driver?)
[    1.239189] 0101            4096 ram1 
[    1.239191]  (driver?)

...

[    1.318949]  (driver?)
[    1.325141] 010f            4096 ram15 
[    1.325143]  (driver?)
[    1.331339] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
```

mmcblk0p7 seems wrong.

I also tried different `sys_bootarg`s:

```boot
setenv sys_bootargs " ubi.mtd=6 root=ubi0:rootfs rootfstype=ubifs rk_dma_heap_cma=66M"
```

This gives:

```
[    1.064668] dwmmc_rockchip ffaa0000.mmc: DW MMC controller at irq 59,32 bit host data width,256 deep fifo
[    1.076866] rve: rve_irq, irq = 61, match scheduler
[    1.083253] rve: Driver loaded successfully rve[2] ver:[2021]58.1a
[    1.089523] rve: probe successfully
[    1.093153] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[    1.102966] rve: Module initialized. v1.0.5
[    1.135184] mmc_host mmc1: Bus speed (slot 0) = 300000Hz (slot req 300000Hz, actual 300000HZ div = 0)
[    1.199585] mmc_host mmc1: Bus speed (slot 0) = 200000Hz (slot req 200000Hz, actual 200000HZ div = 0)
[    1.211730] UBI error: cannot open mtd 6, error -19
[    1.219214] ALSA device list:
[    1.222511]   #0: rv-acodec
[    1.225548] dw-apb-uart ff4c0000.serial: forbid DMA for kernel console
[    1.232271] VFS: Cannot open root device "ubi0:rootfs" or unknown-block(0,0): error -19
[    1.240291] Please append a correct "root=" boot option; here are the available partitions:
[    1.248652] 0100            4096 ram0 
[    1.248655]  (driver?)
[    1.254771] 0101            4096 ram1 
[    1.254773]  (driver?)
[    1.260884] 0102            4096 ram2 
[    1.260886]  (driver?)
[    1.266999] 0103            4096 ram3 

...

[    1.346948] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
```

In both cases, the kernel cannot even open the mtd / mmcblk device, so it's not the contents of my rootfs partition.

I cannot see what the problem is, maybe you have an idea?

plan44, 2025-05-21:
@pengphei finally, after a lot of failed attempts, I made progress and got OpenWrt 22.03 built on aarch64 (apple silicon) and running ok with ethernet connectivity on the luckfox 86panel with emmc and squashfs/f2fs overlay the openwrt way 😄.

I need to clean up things a bit but will post the changes on my fork of this repo ASAP.

ysoyipek, 2025-06-20:
@pengphei and @plan44 ; I need to choose a SoC to use in a project. RV1106G3 is really nice product, but I need OpenWrt in the project. What is the latest status of your projects? I would like to support your projects by donating.

hlyi, 2025-08-16:
@pengphei Thanks for your contribution to the device. Could you please share .config of your build?

plan44, 2025-08-17:
@ysoyipek the status of my OpenWrt use on rv1106 is not much further than it was in June, unfortunately.

For the 86panel I use [this openwrt branch](https://github.com/plan44/openwrt/tree/luz-86panel) and [this rockchip-kernel branch](https://github.com/plan44/rockchip-kernel/tree/openwrt-p44-5.10). The diffconfig for my build is [this one](https://gist.github.com/plan44/94051d62ae47bbc3bc59ea1f1b26d38c).

This works ok for demo purposes, but there is an unresolved startup stability issue (there seems to be a undefined HW state that can block bootup) and there are also spurios unresolved drastic ethernet performance drops, probably also related to undefined HW state somewhere (more likely to happen with higher EMI  such as eMMC active, fast UART transfers ongoing).

At the moment I am wondering if taking current OpenWrt (24.10) and importing the rockchip specifics to make it run on rv1106 might be the better and more longterm useful investment of work than the other way around (trying to bring OpenWrt patches onto rockchip-kernel). I'm trying this right now, but got nowhere with it so far, mostly because I don't have enough experience with kernel porting, and no working JTAG/SWD setup yet for the rv1106.

hlyi, 2025-08-27:
@plan44 Great to hear that you booted 6.6 kernel! The Makefile in rockchip-kernel indicates that it is based upon 5.10.160. But when I checked it against android 5.10.160 kernel, they are quite different. Do you happen to know which version of android kernel  rockchip-kernel is based off?

plan44, 2025-08-28:
@hlyi looking at my notes from a while ago, I am now consfused about rockchip-kernel being a android kernel. Probably that's not true. It seems it's just a fork of mainline, with some android stuff added later (which is a good thing, it makes my forward-porting efforts of 5.10/6.1 rockchip specifics to 6.6.93 less difficult). Here's my notes:

# Rockchip kernel develop-5.10
- forked from Linux "master" mainline
- 2017-06-17
- at adc311034c356e884d180df25deb046cef3e8c75
- Author: Linus Torvalds <torvalds@linux-foundation.org>

- forked from **mainline at v5.10**
- at 2c85ebc57b3e1817b6ce1a6b703928e113a90442
- 2020-12-13
- Tag: v5.10
- Author: Linus Torvalds <torvalds@linux-foundation.org>

# Ticklab kernel
- 2022-09-27 forked from Rockchip develop-5.10
- cbf9da816c4a2d062c55272de83936e597df31d2
- **not relevant for openwrt because ticklab-openwrt was forked earlier**

# Ticklab openwrt kernel
- **detached from develop-5.10**
- 2021-12-18
- made new root commit b78b2876000406fe0f98b1423eb925e0d19de2d1
- from 54821e8d8549d69789b433192cb410734ebec992
    - in rockchip-kernel develop-5.10
    - Author: Wyon Bi <bivvy.bi@rock-chips.com>
    - Date:   Thu Dec 9 15:39:43 2021 +0800
    - **ticklab develop-5.10 was not yet forked back then!**

plan44, 2025-08-28:
weekend status: OpenWrt 24.10 with kernel 6.6.93 builds and boots into failsafe shell on LuckFox pico pro max 😀 !
No ethernet yet, and probably squashfs/ubifs overlay not yet working, but now where I have a shell debugging gets a lot easier!

And with standard luckfox uboot, for every start, I need to enter `setenv kernel_addr_r 0x01000000` in uboot console, because the predefined `kernel_addr_r` is 0x8000 and collides with `mmc@3f000`. **I have very little knowledge about uboot, any help at that point is very welcome!**

All this is **very WIP**, but it supports my assumption that starting with stock OpenWrt kernel and adding rockchip-related patches is much easier than the other way around.

Here's my [OpenWrt 24.10.2](https://github.com/plan44/openwrt/tree/luz-WIP) and [6.6.93 kernel](https://github.com/plan44/linux-stable/tree/luz-WIP) working branches. Eventually, I will clean up these changes and convert them to proper OpenWrt target and [p44build](https://github.com/plan44/p44build) patch sets, but until then I'll work on these forked repos.

To make OpenWrt use the pre-patched external kernel tree, set:

`CONFIG_EXTERNAL_KERNEL_TREE="/my/checkout/of/plan44/linux-stable"`

hlyi, 2025-08-29:
@plan44 Thanks for sharing your port on openwrt. I checked rockchip-kernel against both mainline and andriod kernels, it doesn't match either, but closer to android than mainline. 

In sysdrv/source/uboot/u-boot/include/configs/rv1106_common.h
```
#define ENV_MEM_LAYOUT_SETTINGS		\
	"scriptaddr=0x00b00000\0"	\
	"pxefile_addr_r=0x00c00000\0"	\
	"fdt_addr_r=0x00c00000\0"	\
	"kernel_addr_c=0x00808000\0"	\
	"kernel_addr_r=0x00008000\0"	\
	"ramdisk_addr_r=0x000e00000\0"
#endif
```
Maybe you can change `kernel_addr_r=0x00008000` to `kernel_addr_r=0x01000000`?

plan44, 2025-08-29:
@hlyi thanks for the suggestion! Yes, that would most likely solve the problem, however for now I am reluctant to touch luckfox factory uboot. I'd prefer to find a way to set the load address via uboot env changes, but `kernel_addr_r` is a special env var that cannot be persisted.

plan44, 2025-09-01:
> Great progress👍 is wired ethernet stable?

As far as I could test, yes 😄 

And iperf3 looks good:

```
iperf3 -c 192.168.59.252
Connecting to host 192.168.59.252, port 5201
[  5] local 192.168.59.183 port 44312 connected to 192.168.59.252 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  11.2 MBytes  94.3 Mbits/sec    0   66.5 KBytes       
[  5]   1.00-2.00   sec  11.2 MBytes  94.4 Mbits/sec    0   66.5 KBytes       
[  5]   2.00-3.00   sec  11.1 MBytes  93.3 Mbits/sec    0   66.5 KBytes       
[  5]   3.00-4.00   sec  11.2 MBytes  94.4 Mbits/sec    0   66.5 KBytes       
[  5]   4.00-5.00   sec  11.1 MBytes  93.3 Mbits/sec    0   66.5 KBytes       
[  5]   5.00-6.00   sec  11.2 MBytes  94.4 Mbits/sec    0   66.5 KBytes       
[  5]   6.00-7.00   sec  11.1 MBytes  93.3 Mbits/sec    0   66.5 KBytes       
[  5]   7.00-8.00   sec  11.4 MBytes  95.5 Mbits/sec    0    110 KBytes       
[  5]   8.00-9.00   sec  11.2 MBytes  94.3 Mbits/sec    0    110 KBytes       
[  5]   9.00-10.00  sec  11.1 MBytes  93.2 Mbits/sec    0    110 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   112 MBytes  94.0 Mbits/sec    0             sender
[  5]   0.00-10.02  sec   112 MBytes  93.9 Mbits/sec                  receiver
```

that's about what we can expect from a 100baseT PHY.

hlyi, 2025-09-03:
@plan44 Thanks for the tip and config file! You are right, the issue was CONFIG_EXTERNAL_KERNEL_TREE. After setting it correctly, the kernel compilation went through. Now it failed with this error message:
```
mkenvimage -s 0x40000 -p 0x0 -o /build/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/image/rv1106-nand-env.img ./rv1106-uboot.env.spi-nand.txt
Can't open output file "/build/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/image/rv1106-nand-env.img": No such file or directory
make[5]: *** [Makefile:203: /build/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-rockchip_cortexa7/tmp/openwrt-24.10.2-r28739-d9340319c6-rockchip-cortexa7-luckfox_pico-max-squashfs-env.img] Error 1
make[5]: Leaving directory '/build/openwrt/target/linux/rockchip/image'
```
Any suggestion?

maxl95, 2025-10-06:
@plan44 thank you very much for your effort and dedication. Thanks to your help, I was able to successfully build an OpenWRT image. Unfortunately, I’m stuck now. How do I flash the built images onto the Pico?

plan44, 2025-10-23:
My time was cosumed entirely by another urgent project in the past few days, so I did not get any further, unfortunately. Hopefully that will be different next week, I'll keep you posted.

Looking at your log, my guess would be that mtd4 is not really the right partion. How does the kernel command line look like (should be among the first ~30lines of the bootlog)?

maxl95, 2025-10-23:
@plan44 Here are the relevant excerpts from my log

> Kernel command line: storagemedia=mtd androidboot.storagemedia=mtd androidboot.mode=normal  earlycon=uart8250,mmio32,0xff4c0000 console=ttyS2,115200 mtdparts=spi2.0:256K(env),256K@256K(idblock),512K(uboot),8M(boot),-(ubi) ro rootwait ubi.mtd=ubi ubi.block=0,rootfs root=/dev/ubiblock0_0 rootfstype=squashfs rk_dma_heap_cma=40M androidboot.fwver=ddr-v1.15-306b9977f5,uboot-10/17/2025

> [    1.092760] 5 cmdlinepart partitions found on MTD device spi2.0
> [    1.099446] Creating 5 MTD partitions on "spi2.0":
> [    1.104816] 0x000000000000-0x000000040000 : "env"
> [    1.112809] 0x000000040000-0x000000080000 : "idblock"
> [    1.121423] 0x000000080000-0x000000100000 : "uboot"
> [    1.129781] 0x000000100000-0x000000900000 : "boot"
> [    1.141196] 0x000000900000-0x000008000000 : "ubi"