LuckfoxTECH/luckfox-pico #121: pulling Ubuntu source error

crackerjacques, 2024-06-08:
Hi.
Luckfox Pico Max.
When I chose Ubuntu SD Card Boot.

```
 ==sysdrv== build ubuntu   
 ==ubuntu== 
test -f /home/jackheinlein/luckfox-pico/sysdrv/tools/board/ubuntu/luckfox-ubuntu-22.04.3.tar.gz || (\
	pushd  /home/jackheinlein/luckfox-pico/sysdrv/tools/board/ubuntu/ ;\
	./split_and_check_md5.sh merge ./luckfox-ubuntu-22.04.3.tar.gz ;\
	popd ;\
);
/bin/bash: line 2: pushd: /home/jackheinlein/luckfox-pico/sysdrv/tools/board/ubuntu/: No such file or directory
/bin/bash: line 3: ./split_and_check_md5.sh: No such file or directory
/bin/bash: line 5: popd: directory stack empty
make: *** [Makefile:624: ubuntu] Error 1
make: Leaving directory '/home/jackheinlein/luckfox-pico/sysdrv'
[build.sh:error] Running build_sysdrv failed!
[build.sh:error] exit code 2 from line 644:
[build.sh:info]     make -C ${SDK_SYSDRV_DIR}
```

This is...
```
# SUBMODULES : github/gitee
export LF_SUBMODULES_BY=gitee
```

which can occur in either case.

What could be the cause?
So,I'm using the luckfox-ubuntu-22.04.3.tar.gz.split.* from github.com/luckfox-eng33 by manually copying it there.

best regards.

luckfox-eng33, 2024-08-19:
This is when my colleague updated the SDK and accidentally deleted the link. It has been fixed now.

14Og, 2024-08-19:
Hi! I used to build sd image for my pico mini b using docker environment and im facing issues right now when  i pulled last updates on this repo:
```
[build.sh:error] Running build_sysdrv failed!
[build.sh:error] exit code 2 from line 653:
[build.sh:info]     make -C ${SDK_SYSDRV_DIR}
```
my PATH value is:
`/home/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin:/home/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin:/home/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin:/home/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin`
Is this my mistake or new commits are hiding some erros? The only change i've made is i've added
`export MAKEFLAGS = "-j $(nproc)" ` in .bashrc file inside of docker container.
Thank you!