LuckfoxTECH/luckfox-pico #125: SC3336 sensor format question and how to grab raw data

vjoco, 2024-06-12:
I need tha sensor raw deata for an app
 I have 
```
cat /proc/rkisp-vir0
rkisp-vir0 Version:v02.00.00
clk_isp_core 198000000
aclk_isp   198000000
hclk_isp   148500000
clk_isp_core_vicap 0
Interrupt  Cnt:8 ErrCnt:0
Input      rkcif-mipi-lvds Format:SBGGR10_1X10 Size:2304x1296@25fps Offset(0,0)
```
is the BG10 ( SBGGR10_1X10 ) the pixel direct data...

with 
``` v4l2-ctl --device=/dev/video0 --set-fmt-video=width=2304,height=1296,pixelformat=BG10 --stream-mmap --stream-to=aaa.BG10 --stream-count=1 ```
i am able to grab an image.
 But when I try to compile a c program, I cannot set any format on /dev/video0 . I can succeed only with /dev/video11 ... 

any idea/help?

luckfox-eng33, 2024-06-13:
I'm not sure what's wrong with your program. Generally speaking, it can be read based on v4l2. 
Normally, we don't directly read images that have not gone through ISP.

vjoco, 2024-06-13:
two (dummy) questions: the ISP is on the camera or on the rv1103?  would be possible to get frame before passing it through the ISP?

luckfox-eng33, 2024-06-13:
1. RV1103
2. It should be possible. Maybe you need to close the process of the relevant ISP. You could not read it before or it may be that the ISP is occupying it. You need to test it by yourself.

vjoco, 2024-06-14:
I was digging more:
 - the sc3336 driver (i2c) is setting a MEDIA_BUS_FMT_SBGGR10_1X10 format

 -the media controller maps this to /dev/video0-10
  there is another mapping to video11-21 for the ISP, that generates an NV12 format out of the BGGR10
 
I expect that the video0 gives the raw MEDIA_BUS_FMT_SBGGR10_1X10  data, I'll try to decode that one.

the media-ctl generates this graph: 
[image]

closed 2024-12-15 by vjoco

vjoco, 2025-06-10:
I'll come back with the solution in 1 day. I did it. I just need a pc, not the phone to be able to answer.

vjoco, 2025-06-13:
what I observed:
- the data is almost BG10, there is rubbish data at the end of each line (if I remember well, maybe was the case only for NV21).. probably related with the ALIGN() . 
- 
You took nice the 4 Bayer values (x1..x4)

you can also define a class: 
class Packed10Bit:
    def __init__(self, data):
        self.value1 = data & 0x3FF
        self.value2 = (data >> 10) & 0x3FF
        self.value3 = (data >> 20) & 0x3FF
        self.value4 = (data >> 30) & 0x3FF

--- check this full python script that can be used to convert it to png:

import numpy as np
import cv2
from PIL import Image
import argparse
import os
import sys
from colour_demosaicing import (
    ROOT_RESOURCES_EXAMPLES,
    demosaicing_CFA_Bayer_bilinear,
    demosaicing_CFA_Bayer_Malvar2004,
    demosaicing_CFA_Bayer_Menon2007,
    demosaicing_CFA_Bayer_DDFAPD,
    mosaicing_CFA_Bayer)

class Packed10Bit:
    def __init__(self, data):
        self.value1 = data & 0x3FF
        self.value2 = (data >> 10) & 0x3FF
        self.value3 = (data >> 20) & 0x3FF
        self.value4 = (data >> 30) & 0x3FF

def read_bayer_image(filename, width, height):
    try:
        with open(filename, "rb") as file:
            num_pixels = width * height
            num_bytes = 3072 * height

            buffer = file.read(num_bytes)
            if not buffer:
                print("Error reading file")
                return None

            bayer16 = np.zeros(4 * num_pixels, dtype=np.uint16)

            j = 0
            k = 0
            jl = 0

            while j < num_bytes:
                p10 = Packed10Bit(int.from_bytes(buffer[j:j+5], byteorder='little'))

                v1 = p10.value1
                v2 = p10.value2
                v5 = p10.value3
                v6 = p10.value4

                j += 5
                jl += 5
                if jl == 2880:
                    j += 3072 - 2880
                    jl = 0

                bayer16[k] = v1 << 6
                bayer16[k + 1] = v2 << 6
                bayer16[k + 2] = v5 << 6
                bayer16[k + 3] = v6 << 6
                k += 4

            # with open("bayer16_again.bin", "wb") as file2:
            #     file2.write(bayer16[:k].tobytes())

            return bayer16[:k]

    except IOError:
        print("Error opening file")
        return None

def whitebalance_gammaCorrection(image):
    # Apply gamma correction
    gamma_corrected_image = cv2.convertScaleAbs(image, alpha=(255.0 / 65536.0))

    # Normalize the image to 0-255 range
    normalized_image = cv2.normalize(gamma_corrected_image, None, alpha=0, beta=255, norm_type=cv2.NORM_MINMAX, dtype=cv2.CV_8U)

    return normalized_image

def convertRAW2PNG(raw_data, width, height):
    #reshape array to matrix
    raw_image = raw_data.reshape((height, width))
    #apply white balance to get pixels with better color information
    normalized_image = whitebalance_gammaCorrection(demosaicing_CFA_Bayer_Malvar2004(raw_image, pattern="BGGR"))
    #normalize colors
    # bgr_image = cv2.cvtColor(normalized_image, cv2.COLOR_BGR2RGB)
    #flip horizontally
    flipped_img = cv2.flip(normalized_image, 1)
    #save (if needed)
    #cv2.imwrite(destination_path, flipped_img) #only if you want to keep it in a specific folder and name

    return flipped_img

# if __name__ == "__main__":
#     import sys

#     if len(sys.argv) != 4:
#         print(f"Usage: {sys.argv[0]} <input.raw> <width=2304> <height=1296>")
#         sys.exit(1)

#     width = int(sys.argv[2])
#     height = int(sys.argv[3])

#     bayer = read_bayer_image(sys.argv[1], width, height)
#     if bayer is None:
#         sys.exit(1)

#     rgb_image = convertRAW2PNG(bayer, width, height)
#     # Save the RGB image as a JPEG
#     image = Image.fromarray(rgb_image)
#     file_name_without_ext = os.path.splitext(sys.argv[1])[0]
#     image.save(file_name_without_ext + '.png')

if __name__ == "__main__":
    parser = argparse.ArgumentParser(description='Convert RAW Bayer image to PNG.')
    parser.add_argument('input_file', type=str, help='Input RAW file')
    parser.add_argument('width', type=int, help='Width of the image')
    parser.add_argument('height', type=int, help='Height of the image')
    parser.add_argument('output_file', type=str, help='Output PNG file name')

    args = parser.parse_args()

    bayer = read_bayer_image(args.input_file, args.width, args.height)
    if bayer is None:
        print("Failed to read the Bayer image.")
        sys.exit(1)

    rgb_image = convertRAW2PNG(bayer, args.width, args.height)
    if rgb_image is None:
        print("Failed to convert RAW to RGB.")
        sys.exit(1)

    # Save the RGB image as a PNG
    image = Image.fromarray(rgb_image,'RGB')
    image.save(args.output_file)
    print(f"Image saved as {args.output_file}")