LuckfoxTECH/luckfox-pico #129: upgrade_tool / SocToolKit replacement for Linux writing to SD card

Spiritdude, 2024-06-17:
I posted in the forum at https://forums.luckfox.com/viewtopic.php?t=786 already, but no feedback yet, so I try here.

I like to install Ubuntu on eMMC and SD card, but since `upgrade_tool` is binary only and doesn't support SD card but only eMMC, and SocToolKit.exe is MS-Windows only, I'm looking for some guidance how to config the SD card on the host (and not inserted in the board) with Linux.

I started a py script which reads the `.env.txt` but doesn't do anything else yet, e.g. 
```
% blkenvflash /dev/sdc
mmcblk1: env.img size:32,768/32K (offset:0/0B) imgsize:32,768 (32K)
mmcblk1: idblock.img size:524,288/512K (offset:32,768/32K) imgsize:182,272 (178K)
mmcblk1: uboot.img size:262,144/256K (offset:0/0B) imgsize:262,144 (256K)
mmcblk1: boot.img size:33,554,432/32M (offset:0/0B) imgsize:2,903,552 (2,903,552B)
mmcblk1: oem.img size:536,870,912/512M (offset:0/0B) imgsize:45,555,712 (44,488K)
mmcblk1: userdata.img size:268,435,456/256M (offset:0/0B) imgsize:9,999,360 (9,765K)
mmcblk1: rootfs.img size:6,442,450,944/6G (offset:0/0B) imgsize:1,094,365,184 (1,068,716K)
mmcblk0: idblock.img size:524,288/512K (offset:32,768/32K) imgsize:182,272 (178K)
mmcblk0: uboot.img size:4,194,304/4M (offset:0/0B) imgsize:262,144 (256K)
```

So, as I understand, the `mmcblk0` is the eMMC, and the `mmcblk1` is the SD card here; are there any kind of security/encryption involved, or is it simply partitioning the SD card with the values (size & offset) and `dd` the corresponding `.img` in there?

Why doesn't `upgrade_tool` support writing SD card while inserted in the board, what's the problem / difficulty in there?

I tried to run SocTookKit.exe via WINE emulator, but it failed - I won't run MS-Windows just to write/config a SD card.

AlexanderSchuetz97, 2024-06-17:
I had the exact same problem as you. I ended up using dd with a makefile provided by some helpful man.

See: (Dont worry about the name of the issue, initally I failed to earase flash on linux, then I had the sd card issue.)
https://github.com/LuckfoxTECH/luckfox-pico/issues/66

luyi1888, 2024-06-18:
There is the Linux version of SocToolKit.
I used it on Ubuntu 20.04, no problem.
remember to use sudo to launch.

https://github.com/LuckfoxTECH/luckfox-pico/blob/main/tools/linux/SocToolKit/SocToolKit

Spiritdude, 2024-06-18:
@luyi1888 thanks for the hint, I tried it but it failed under Ubuntu 22.04 LTS as the SD card did not show up as 'Usb Disk'.

@AlexanderSchuetz97 thanks for the link, I extended my py script according the info I was able to get from it, and created a `disk.img`  which I then `dd`ed unto the SD card, and ta-tah:

[image]

[image]

I gonna cleanup my script and then make it available.

Spiritdude, 2024-06-18:
For those interested in my quick put together `blkenvflash` script: https://gist.github.com/Spiritdude/da36d2cf064e49094c870e0a8b9f972f
and looks like this:
```
% sudo ./blkenvflash /dev/sdX
== blkenvflash 0.0.1 ==
writing to /dev/sdX
   mmcblk1: env.img size:32,768/32K (offset:0/0B) imgsize:32,768 (32K)
   mmcblk1: idblock.img size:524,288/512K (offset:32,768/32K) imgsize:182,272 (178K)
   mmcblk1: uboot.img size:262,144/256K (offset:0/0B) imgsize:262,144 (256K)
   mmcblk1: boot.img size:33,554,432/32M (offset:0/0B) imgsize:2,903,552 (2,903,552B)
   mmcblk1: oem.img size:536,870,912/512M (offset:0/0B) imgsize:45,555,712 (44,488K)
   mmcblk1: userdata.img size:268,435,456/256M (offset:0/0B) imgsize:9,999,360 (9,765K)
   mmcblk1: rootfs.img size:6,442,450,944/6G (offset:0/0B) imgsize:1,094,365,184 (1,068,716K)
done.
```

NOTE: do not just copy / paste, inquery which device your SD card is mapped to with `lsblk` and then replace `/dev/sdX` with it.

luckfox-eng33, 2024-06-19:
It looks like the problem is solved

closed 2024-06-19 by Spiritdude

RobertH1993, 2026-10-15:
For anyone landing on this problem, a solution was already given but i came up with this python script:

```
#!/usr/bin/env python3
# sdwrite.py - Writes Rockchip SDK-images to SD-card based on blkdevparts
import re, sys, os, argparse

UNITS = {"": 1, "K": 1024, "M": 1024**2, "G": 1024**3}

def to_bytes(s):
    m = re.fullmatch(r"(\d+)([KMGkmg]?)", s)
    return int(m.group(1)) * UNITS[m.group(2).upper()]

def parse(env_path):
    text = open(env_path).read()
    m = re.search(r"blkdevparts=\S*?:(\S+)", text)
    if not m:
        sys.exit("blkdevparts not found " + env_path)
    parts, pos = [], 0
    for item in m.group(1).split(","):
        mm = re.fullmatch(r"(-|\d+[KMGkmg]?)(?:@(\d+[KMGkmg]?))?\((\w+)\)", item)
        if not mm:
            sys.exit("Unable to parse: " + item)
        size, off, name = mm.groups()
        if off:
            pos = to_bytes(off)
        parts.append((name, pos, None if size == "-" else to_bytes(size)))
        if size != "-":
            pos += to_bytes(size)
    return parts

ap = argparse.ArgumentParser()
ap.add_argument("device", help="/dev/sdX (full block device, not a partition)")
ap.add_argument("--img", default="output/image")
ap.add_argument("--write", action="store_true", help="No dry-run, write")
a = ap.parse_args()

parts = parse(os.path.join(a.img, ".env.txt"))
for name, off, size in parts:
    f = os.path.join(a.img, name + ".img")
    if not os.path.exists(f):
        print(f"[skip] {name}: geen {f}")
        continue
    fsize = os.path.getsize(f)
    if size and fsize > size:
        sys.exit(f"{name}: image ({fsize}) bigger then partition: ({size})")
    print(f"{name:10} offset {off:>12} (sector {off//512:>8})  {fsize} bytes")

if not a.write:
    print("\nDry-run. Add --write to write to disk")
    sys.exit()

if input(f"Overwrite Everything on {a.device} type 'yes': ") != "yes":
    sys.exit("Canceled")

with open(a.device, "r+b") as dev:
    for name, off, size in parts:
        f = os.path.join(a.img, name + ".img")
        if not os.path.exists(f):
            continue
        dev.seek(off)
        with open(f, "rb") as src:
            while chunk := src.read(1 << 20):
                dev.write(chunk)
        print("Written:", name)
    dev.flush()
    os.fsync(dev.fileno())
print("Done, run 'sync' before removing SD card")
```