LuckfoxTECH/luckfox-pico #130: WiFi not working when Ubuntu is built from SDK

x-magic, 2024-06-20:
When using the SDK to build buildroot image for Pico Ultra W, wlan0 is functional (although udhcpc is not automatically running on the interface thus must manually invoke for DHCP - that's another story). However when choosing `BoardConfig-EMMC-Ubuntu-RV1106_Luckfox_Pico_Ultra_W-IPC.mk` as build target, wlan0 is no where to be found. 

It seems mmc1 is detected, but the wireless module on it is not. 

选Ubuntu作为Pico Ultra W的build target时WiFi模块检测不到，无法驱动。Buildroot没问题。

Boot log from UART: 

```
DDR V1.10 ec2fae0c96 wesley.yao 22/11/15-10:58:09
S5P1
f967
F
DDRConf2
DDR3, BW=16 Col=10 Bk=8 CS0 Row=14 CS=1 Die BW=16 Size=256MB
924MHz
DDR bin out

U-Boot SPL board init
U-Boot SPL 2017.09 (Jun 20 2024 - 19:46:36)
unrecognized JEDEC id bytes: 00, 00, 00
unknown raw ID 0 0 0
Trying to boot from MMC2
Card did not respond to voltage select!
mmc_init: -95, time 16
spl: mmc init failed with error: -95
Trying to boot from MMC1
ENVF: Primary 0x00000000 - 0x00008000
ENVF: OK
Trying fit image at 0x440 sector
## Verified-boot: 0
## Checking uboot 0x00200000 (lzma @0x00400000) ... sha256(76694e7803...) + sha256(b6b41ac172...) + OK
## Checking fdt 0x0025bed0 ... sha256(51bba657c3...) + OK
Total: 308.915 ms

Jumping to U-Boot(0x00200000)

U-Boot 2017.09 (Jun 20 2024 - 19:46:36 +1000)

Model: Rockchip RV1106 EVB Board
PreSerial: 2, raw, 0xff4c0000
DRAM:  256 MiB
Sysmem: init
Relocation Offset: 0fd85000
Relocation fdt: 0edfacb0 - 0edfede0
CR: M/C/I
Using default environment

mmc@ffa90000: 0, mmc@ffaa0000: 1
ENVF: Primary 0x00000000 - 0x00008000
ENVF: OK
Bootdev(atags): mmc 0
MMC0: HS200, 200Mhz
PartType: ENV
DM: v2
No misc partition
boot mode: normal
FIT: no signed, no conf required
DTB: rk-kernel.dtb
HASH(c): OK
Model: Luckfox Pico Ultra W
CLK: (sync kernel. arm: enter 816000 KHz, init 816000 KHz, kernel 0N/A)
  apll 816000 KHz
  dpll 924000 KHz
  gpll 1188000 KHz
  cpll 1000000 KHz
  aclk_peri_root 400000 KHz
  hclK_peri_root 200000 KHz
  pclk_peri_root 100000 KHz
  aclk_bus_root 300000 KHz
[... 532 lines cut ...]

Welcome to Ubuntu 22.04.3 LTS (GNU/Linux 5.10.110 armv7l)

 * Documentation:  https://wiki.luckfox.com
 * Management:     https://www.luckfox.com

This system has been minimized by removing packages and content that are
not required on a system that users do not log into.

To restore this content, you can run the 'unminimize' command.
Last login: Thu Jun 20 20:33:35 CST 2024 on ttyFIQ0
root@luckfox:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether <REDACTED> brd ff:ff:ff:ff:ff:ff
3: usb0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether <REDACTED> brd ff:ff:ff:ff:ff:ff
    inet 172.32.0.70/16 brd 172.32.255.255 scope global usb0
       valid_lft forever preferred_lft forever
```

Any suggestions? Thank you.

luckfox-eng33, 2024-06-21:
ubuntu 暂时还没有适配wifi模块，我们优先适配buildroot，因为ubuntu无法使用RK有关的媒体库。
如果你非要使用ubuntu使用wifi的话，你可以试着把aic8800dc的驱动（.ko）和固件放到相同位置，并且挂载驱动

x-magic, 2024-06-21:
> ubuntu 暂时还没有适配wifi模块，我们优先适配buildroot，因为ubuntu无法使用RK有关的媒体库。 如果你非要使用ubuntu使用wifi的话，你可以试着把aic8800dc的驱动（.ko）和固件放到相同位置，并且挂载驱动

谢谢，我试试。文档里写一下就更好了，毕竟Ubuntu的board config里有wifi配置，根本就没往驱动那个方面去想。

luckfox-eng33, 2024-06-21:
> > ubuntu 暂时还没有适配wifi模块，我们优先适配buildroot，因为ubuntu无法使用RK有关的媒体库。 如果你非要使用ubuntu使用wifi的话，你可以试着把aic8800dc的驱动（.ko）和固件放到相同位置，并且挂载驱动
> 
> 谢谢，我试试。文档里写一下就更好了，毕竟Ubuntu的board config里有wifi配置，根本就没往驱动那个方面去想。

我和同事说一下，让他们更新一下

noon92, 2024-10-23:
Does wifi work yet with pico mini?

closed 2025-06-17 by x-magic