LuckfoxTECH/luckfox-pico #131: USB port next to ethernet on Pico Ultra W not working

x-magic, 2024-06-20:
The USB port next to the ethernet port on Pico Ultra W is not functional with either buildroot or Ubuntu images. The USB gadgets through Type-C port is still working, USB network, adb, etc. 

Pico Ultra W无论是buildroot还是Ubuntu，板载的USB口都检测不到设备，dmesg也没有反应，但Type-C口的功能正常。官方Google Drive里的镜像还是自己编译的（就是默认配置）都这个问题。

Seems the usbcore is initialised in the kernel, but no root hub can be found: 
```
[root@luckfox root]$ lsusb
[root@luckfox  root]$
```

Boot log from UART: 
```
DDR V1.10 ec2fae0c96 wesley.yao 22/11/15-10:58:09
S5P1
f967
F
DDRConf2
DDR3, BW=16 Col=10 Bk=8 CS0 Row=14 CS=1 Die BW=16 Size=256MB
924MHz
DDR bin out

U-Boot SPL board init
U-Boot SPL 2017.09 (Jun 20 2024 - 18:42:26)
unrecognized JEDEC id bytes: 00, 00, 00
unknown raw ID 0 0 0
Trying to boot from MMC2
Card did not respond to voltage select!
mmc_init: -95, time 16
spl: mmc init failed with error: -95
Trying to boot from MMC1
ENVF: Primary 0x00000000 - 0x00008000
ENVF: OK
Trying fit image at 0x440 sector
## Verified-boot: 0
## Checking uboot 0x00200000 (lzma @0x00400000) ... sha256(b1b296a34c...) + sha256(e5bbb79ba0...) + OK
## Checking fdt 0x0025bed0 ... sha256(51bba657c3...) + OK
Total: 309.506 ms

Jumping to U-Boot(0x00200000)

U-Boot 2017.09 (Jun 20 2024 - 18:42:26 +1000)

Model: Rockchip RV1106 EVB Board
PreSerial: 2, raw, 0xff4c0000
DRAM:  256 MiB
Sysmem: init
Relocation Offset: 0fd85000
Relocation fdt: 0edfacb0 - 0edfede0
CR: M/C/I
Using default environment

mmc@ffa90000: 0, mmc@ffaa0000: 1
ENVF: Primary 0x00000000 - 0x00008000
ENVF: OK
Bootdev(atags): mmc 0
MMC0: HS200, 200Mhz
PartType: ENV
DM: v2
No misc partition
boot mode: normal
FIT: no signed, no conf required
DTB: rk-kernel.dtb
HASH(c): OK
Model: Luckfox Pico Ultra W
CLK: (sync kernel. arm: enter 816000 KHz, init 816000 KHz, kernel 0N/A)
  apll 816000 KHz
  dpll 924000 KHz
  gpll 1188000 KHz
  cpll 1000000 KHz
  aclk_peri_root 400000 KHz
[... 687 lines cut ...]

luckfox : set usb0 ip

luckfox : set usb0 ip

luckfox : set usb0 ip

luckfox : set usb0 ip

usb0 config error
Starting SMB services: OK
Starting NMB services: OK
[   38.536536] of_dma_request_slave_channel: dma-names property of node '/serial@ff4b0000' missing or empty
[   38.536678] dw-apb-uart ff4b0000.serial:Device setup complete
 failed to request DMA, use interrupt mode
/root/main.py and /root/boot.py not exist ,pass...
OK

Welcome to Luckfox Pico
luckfox login: udhcpc: broadcasting discover
udhcpc: broadcasting discover
udhcpc: broadcasting discover
udhcpc: broadcasting discover
udhcpc: broadcasting discover
```

luckfox-eng33, 2024-06-21:
应该在wiki中有描述，A口和C口是共用一个USB的，中间没有hub（hub无法做主从切换）
当USB-C口有供电，则USB会切换到USB-C上
如果USB-C口没有供电，则USB回切换到USB-A上面（主要配合POE模块或者外部供电模块一起使用）

x-magic, 2024-06-21:
> 应该在wiki中有描述，A口和C口是共用一个USB的，中间没有hub（hub无法做主从切换） 当USB-C口有供电，则USB会切换到USB-C上 如果USB-C口没有供电，则USB回切换到USB-A上面（主要配合POE模块或者外部供电模块一起使用）

如果是指[Wiki这篇文章的话](https://wiki.luckfox.com/zh/Luckfox-Pico/Luckfox-Pico-USB/)（英文版也是），确实无法联想到你指出的这个问题，因为文档里除了概述页，Ultra相关的资料基本上是没有的。其他型号的Pico并没有两个物理的USB口，所以我就想当然的认为两个口可以共用。我试试单独供电，谢谢

luckfox-eng33, 2024-06-21:
> > 应该在wiki中有描述，A口和C口是共用一个USB的，中间没有hub（hub无法做主从切换） 当USB-C口有供电，则USB会切换到USB-C上 如果USB-C口没有供电，则USB回切换到USB-A上面（主要配合POE模块或者外部供电模块一起使用）
> 
> 如果是指[Wiki这篇文章的话](https://wiki.luckfox.com/zh/Luckfox-Pico/Luckfox-Pico-USB/)（英文版也是），确实无法联想到你指出的这个问题，因为文档里除了概述页，Ultra相关的资料基本上是没有的。其他型号的Pico并没有两个物理的USB口，所以我就想当然的认为两个口可以共用。我试试单独供电，谢谢
> 
> 这个板子和我之前接触的大多数SBC真的太不一样了（更像个MCU）

应该参考Ultra的[wiki](https://wiki.luckfox.com/zh/Luckfox-Pico/Luckfox-Pico-RV1106/Luckfox-Pico-Ultra-W/Luckfox-Pico-USB)

x-magic, 2024-06-21:
> 应该参考Ultra的[wiki](https://wiki.luckfox.com/zh/Luckfox-Pico/Luckfox-Pico-RV1106/Luckfox-Pico-Ultra-W/Luckfox-Pico-USB)

是不是Wiki更新了？我记得今天上午还不是这样的🤦

luckfox-eng33, 2024-06-21:
> 是不是Wiki更新了？我记得今天上午还不是这样的🤦

可能有？十一点的时候看是已经这样样子了
最近wiki可能会持续更新，如果发现有不合理的地方，我们就会修改.

x-magic, 2024-06-21:
> > 是不是Wiki更新了？我记得今天上午还不是这样的🤦
> 
> 可能有？十一点的时候看是已经这样样子了 最近wiki可能会持续更新，如果发现有不合理的地方，我们就会修改.

👍 也就是说，我将USB设置为host模式的话，USB A口就应该可以用了吧？peripheral模式和otg模式的区别是什么？是和USB gadget或adb有关吗？谢谢

luckfox-eng33, 2024-06-21:
> > > 是不是Wiki更新了？我记得今天上午还不是这样的🤦
> > 
> > 
> > 可能有？十一点的时候看是已经这样样子了 最近wiki可能会持续更新，如果发现有不合理的地方，我们就会修改.
> 
> 👍 也就是说，我将USB设置为host模式的话，USB A口就应该可以用了吧？peripheral模式和otg模式的区别是什么？是和USB gadget或adb有关吗？谢谢

根据你要连接的设备，host代表作为主机，用来读取或者驱动设备，具体可以看下usb的几个角色[维基百科](https://en.wikipedia.org/wiki/USB_On-The-Go)
adb通常是peripheral才能使用

x-magic, 2024-06-21:
> > > > 是不是Wiki更新了？我记得今天上午还不是这样的🤦
> > > 
> > > 
> > > 可能有？十一点的时候看是已经这样样子了 最近wiki可能会持续更新，如果发现有不合理的地方，我们就会修改.
> > 
> > 
> > 👍 也就是说，我将USB设置为host模式的话，USB A口就应该可以用了吧？peripheral模式和otg模式的区别是什么？是和USB gadget或adb有关吗？谢谢
> 
> 根据你要连接的设备，host代表作为主机，用来读取或者驱动设备，具体可以看下usb的几个角色[维基百科](https://en.wikipedia.org/wiki/USB_On-The-Go) adb通常是peripheral才能使用

明白了
如果只有host和peripheral模式我还能理解，otg在列表里搞得我有点懵

x-magic, 2024-06-22:
> 应该在wiki中有描述，A口和C口是共用一个USB的，中间没有hub（hub无法做主从切换） 当USB-C口有供电，则USB会切换到USB-C上 如果USB-C口没有供电，则USB回切换到USB-A上面（主要配合POE模块或者外部供电模块一起使用）

还真的是USB-C口什么都不能插，供电都不行，否则A口没反应🫠