LuckfoxTECH/luckfox-pico #132: i2c and spi device

nopnop2002, 2024-06-22:
I'm using the official buildroot image of Mini board.

This i2c device exists, but it cannot be used as an i2c.

[image]

No SPI device.

[image]

x-magic, 2024-06-22:
It seems PWM pins needs to be explicitly enabled before accessible in `/sys/class/pwm`
https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-RV1106/Luckfox-Pico-Pro-Max/Luckfox-Pico-PWM#22-enable-pwm

nopnop2002, 2024-06-22:
[image]

nopnop2002, 2024-07-03:
The cause is that this file is not configured correctly.
```
$ cat rv1103g-luckfox-pico-mini-b.dts
// SPDX-License-Identifier: (GPL-2.0+ OR MIT)
/*
 * Copyright (c) 2023 Luckfox Electronics Co., Ltd.
 */

/dts-v1/;

#include "rv1103.dtsi"
#include "rv1106-evb.dtsi"
#include "rv1103-luckfox-pico-ipc.dtsi"

/ {
        model = "Luckfox Pico Mini B";
        compatible = "rockchip,rv1103g-38x38-ipc-v10", "rockchip,rv1103";
};

/**********SFC**********/
&sfc {
        status = "okay";
        flash@0 {
                compatible = "spi-nand";
                reg = <0>;
                spi-max-frequency = <75000000>;
                spi-rx-bus-width = <4>;
                spi-tx-bus-width = <1>;
        };
};

/**********SDMMC**********/
&sdmmc {
        max-frequency = <50000000>;
        no-sdio;
        no-mmc;
        bus-width = <4>;
        cap-mmc-highspeed;
        cap-sd-highspeed;
        disable-wp;
        pinctrl-names = "default";
        pinctrl-0 = <&sdmmc0_clk &sdmmc0_cmd &sdmmc0_det &sdmmc0_bus4>;
        status = "okay";
};

/**********ETH**********/
&gmac {
        status = "disabled";
};

/**********USB**********/
```