LuckfoxTECH/luckfox-pico #135: control  WS2812B LEDs

KaliAssistant, 2024-06-28:
is possible use `rpi_ws281x` and `adafruit-circuitpython-neopixel` to control ws2812b rgb led just like raspberry pi?

plan44, 2024-06-28:
I hope this is possible, maybe by leveraging pne of the PWMs of which 2 units apparently can send IR data? Is there any detail docs about the RV1106 PWM? I have written a [PWM based WS28xx kernel driver](https://github.com/plan44/plan44-feed/tree/main/p44-ledchain) for the Mediatek MT7688, and would very much like to port it to RV1106, once I can obtain docs.

KaliAssistant, 2024-06-28:
I will try it.

nopnop2002, 2024-07-04:
adafruit-circuitpython-neopixel library is based on Adafruit Blinka.

Adafruit Blinka supports this board.

KaliAssistant, 2024-07-04:
@nopnop2002 I installed `rpi_ws281x` and `adafruit-circuitpython-neopixel` , i run the code ,it says gpio not support this board

KaliAssistant, 2024-07-04:
it still using `RPI.gpio`

KaliAssistant, 2024-07-04:
I use 0.5 days to install pip pkg

nopnop2002, 2024-07-04:
https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/microcontroller/rockchip/rv1103/pin.py

https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/microcontroller/rockchip/rv1106/pin.py

KaliAssistant, 2024-07-04:
@nopnop2002 thanks i will try it.

nopnop2002, 2024-07-04:
Enabling SWAP will make it run a little faster.
```
$ sudo mkswap /dev/mmcblk1p8
Setting up swapspace version 1, size = 54726455296 bytes

$ sudo swapon /dev/mmcblk1p8
[  105.830584] Adding 53443804k swap on /dev/mmcblk1p8.  Priority:-2 extents:1 across:53443804k SS

$ free -h
               total        used        free      shared  buff/cache   available
Mem:            33Mi        13Mi       2.0Mi       0.0Ki        17Mi        16Mi
Swap:          8.7Gi        15Mi       8.7Gi
```

KaliAssistant, 2024-07-04:
This is very harmful to the sd card...

nopnop2002, 2024-07-04:
>This is very harmful to the sd card...

Why?

KaliAssistant, 2024-07-04:
> >This is very harmful to the sd card...
> 
> 
> 
> Why?

Just like you make a very slow ram on sd card...

KaliAssistant, 2024-07-05:
@nopnop2002 
same...
```
pico@foxjack:~/python3$ sudo ./.venv/bin/python3
[sudo] password for pico:
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gpiod
>>> import board
>>> import neopixel
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pico/python3/.venv/lib/python3.10/site-packages/neopixel.py", line 18, in <module>
    from neopixel_write import neopixel_write
  File "/home/pico/python3/.venv/lib/python3.10/site-packages/neopixel_write.py", line 41, in <module>
    raise NotImplementedError("Board not supported")
NotImplementedError: Board not supported
>>>
```

luckfox-eng33, 2024-07-05:
If you don't use SPI, you can theoretically use SPI's MOSI pins to control WS2812B

rpi_ws281x has two options
1. Based on the PWM register and DMA of Raspberry Pi, it is difficult to implement on RV1103/6
2. Using SPI for simulation is not difficult to implement.

KaliAssistant, 2024-07-05:
I run my code
```
pico@foxjack:~/python3$ sudo ./.venv/bin/python3 ./neotest 
[sudo] password for pico: 
Traceback (most recent call last):
  File "/home/pico/python3/./neotest", line 97, in <module>
    strip.begin()
  File "/home/pico/python3/.venv/lib/python3.10/site-packages/rpi_ws281x/rpi_ws281x.py", line 143, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
Segmentation fault
```
code:
```python
#!/usr/bin/env python3
# NeoPixel library strandtest example
# Author: Tony DiCola (tony@tonydicola.com)
#
# Direct port of the Arduino NeoPixel library strandtest example.  Showcases
# various animations on a strip of NeoPixels.

import time
from rpi_ws281x import PixelStrip, Color
import argparse

# LED strip configuration:
LED_COUNT = 1        # Number of LED pixels.
LED_PIN = 50          # SPI ? idk 
LED_FREQ_HZ = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA = 10          # ?
LED_BRIGHTNESS = 255  # Set to 0 for darkest and 255 for brightest
LED_INVERT = False    # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL = 0       # set to '1' for GPIOs 13, 19, 41, 45 or 53

# Define functions which animate LEDs in various ways.
def colorWipe(strip, color, wait_ms=50):
    """Wipe color across display a pixel at a time."""
    for i in range(strip.numPixels()):
        strip.setPixelColor(i, color)
        strip.show()
        time.sleep(wait_ms / 1000.0)

def theaterChase(strip, color, wait_ms=50, iterations=10):
    """Movie theater light style chaser animation."""
    for j in range(iterations):
        for q in range(3):
            for i in range(0, strip.numPixels(), 3):
                strip.setPixelColor(i + q, color)
            strip.show()
            time.sleep(wait_ms / 1000.0)
            for i in range(0, strip.numPixels(), 3):
                strip.setPixelColor(i + q, 0)

def wheel(pos):
    """Generate rainbow colors across 0-255 positions."""
    if pos < 85:
        return Color(pos * 3, 255 - pos * 3, 0)
    elif pos < 170:
        pos -= 85
        return Color(255 - pos * 3, 0, pos * 3)
    else:
        pos -= 170
        return Color(0, pos * 3, 255 - pos * 3)

def rainbow(strip, wait_ms=20, iterations=1):
    """Draw rainbow that fades across all pixels at once."""
    for j in range(256 * iterations):
        for i in range(strip.numPixels()):
            strip.setPixelColor(i, wheel((i + j) & 255))
        strip.show()
        time.sleep(wait_ms / 1000.0)

def rainbowCycle(strip, wait_ms=20, iterations=5):
    """Draw rainbow that uniformly distributes itself across all pixels."""
    for j in range(256 * iterations):
        for i in range(strip.numPixels()):
            strip.setPixelColor(i, wheel(
                (int(i * 256 / strip.numPixels()) + j) & 255))
        strip.show()
        time.sleep(wait_ms / 1000.0)

def theaterChaseRainbow(strip, wait_ms=50):
    """Rainbow movie theater light style chaser animation."""
    for j in range(256):
        for q in range(3):
            for i in range(0, strip.numPixels(), 3):
                strip.setPixelColor(i + q, wheel((i + j) % 255))
            strip.show()
            time.sleep(wait_ms / 1000.0)
            for i in range(0, strip.numPixels(), 3):
                strip.setPixelColor(i + q, 0)

# Main program logic follows:
if __name__ == '__main__':
    # Process arguments
    parser = argparse.ArgumentParser()
    parser.add_argument('-c', '--clear', action='store_true', help='clear the display on exit')
    args = parser.parse_args()

    # Create NeoPixel object with appropriate configuration.
    strip = PixelStrip(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL)
    # Intialize the library (must be called once before other functions).
    strip.begin()

    print('Press Ctrl-C to quit.')
    if not args.clear:
        print('Use "-c" argument to clear LEDs on exit')

    try:

        while True:
            print('Color wipe animations.')
            colorWipe(strip, Color(255, 0, 0))  # Red wipe
            colorWipe(strip, Color(0, 255, 0))  # Green wipe
            colorWipe(strip, Color(0, 0, 255))  # Blue wipe
            print('Theater chase animations.')
            theaterChase(strip, Color(127, 127, 127))  # White theater chase
            theaterChase(strip, Color(127, 0, 0))  # Red theater chase
            theaterChase(strip, Color(0, 0, 127))  # Blue theater chase
            print('Rainbow animations.')
            rainbow(strip)
            rainbowCycle(strip)
            theaterChaseRainbow(strip)

    except KeyboardInterrupt:
        if args.clear:
            colorWipe(strip, Color(0, 0, 0), 10)
```
idk what gpio num should i use in my code

nopnop2002, 2024-07-05:
Adafruit CircuitPython NeoPixel only supports these boards.
```
if detector.board.any_raspberry_pi:
    from adafruit_blinka.microcontroller.bcm283x import neopixel as _neopixel
elif detector.board.pico_u2if:
    from adafruit_blinka.microcontroller.rp2040_u2if import neopixel as _neopixel
elif (
    detector.board.feather_u2if
    or detector.board.feather_can_u2if
    or detector.board.feather_epd_u2if
    or detector.board.feather_rfm_u2if
    or detector.board.qtpy_u2if
    or detector.board.itsybitsy_u2if
    or detector.board.macropad_u2if
    or detector.board.qt2040_trinkey_u2if
    or detector.board.kb2040_u2if
):
    from adafruit_blinka.microcontroller.rp2040_u2if import neopixel as _neopixel
elif "sphinx" in sys.modules:
    pass
else:
    raise NotImplementedError("Board not supported")
```

plan44, 2024-07-06:
> If you don't use SPI, you can theoretically use SPI's MOSI pins to control WS2812B
The RV1106 has two SPI units - only SPI0 is exposed on the LuckFox pico's pins. I guess the internal NAND-Flash is also connected via SPI, is that on SPI1? If so, SPI0 would be free 😄 

> 2. Using SPI for simulation is not difficult to implement.

Yes, if you can set the SPI to ~2.8Mhz bit clock, then sending 0xC and 0x8 nibbles for H and L bits, respectively, works fine (implemented this on another platforms some years ago). The wiki has good information about SPI programming: https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-SPI#4-spi-communication-c-program

I will try this as soon as I get my OpenWrt flashed onto the pico max, but I am stuck with that right now...

luckfox-eng33, 2024-07-09:
> I will try this as soon as I get my OpenWrt flashed onto the pico max, but I am stuck with that right now...

It seems that someone has already ported openwrt in #115. You can refer to it.

plan44, 2024-07-09:
> @luckfox-eng33 It seems that someone has already ported openwrt in #115. You can refer to it.

Yes, thanks - I'm aware of that and am trying to work with and maybe help top improve it, because at this time it is SD-card only and uses the SDK kernel, not the one from openwrt. I'm currently trying to get the openwrt built kernel find a openwrt-style squashfs/jffs2 rootfs on the pico max, but with little success so far (see my comments in #115).

DhSufi, 2024-10-09:
Just in case someone finds it useful. As luckfox-eng33 said, we should be able to control them with SPI.

1. You should play with spi.max_speed_hz (between 2500000 and 3500000 depending on your LEDs).

2. Then connect the LEDs data pin to the MOSI pin of the Luckfox Pico Pro.

3. Finally you have to translate each bit of information to a 3 bits message in the SPI bus:
(This is because of WS2812 bit timings VS SPI bit timings, and is related to the frequency we set above in point 1)
    - if data bit is 1 you have to send 3 bits: 110 on the MOSI pin
    - if data bit is 0 you have to send 3 bits: 100 on the MOSI pin

A simple python script to set 8 leds strip to color red will be like this:
```
import spidev

spi = spidev.SpiDev()
spi.open(0, 0)
spi.max_speed_hz = 3300000  # Adjust between 2500000 and 3500000

# Create a list with the GRB color for each led
num_leds = 8
colors_list = []
for _ in range(num_leds):
    colors_list.extend([0, 255, 0])  # set to color RED -> format GRB

mosi_data = []
for color_byte in colors_list:
    
    # convert each color byte to list of bits
    byte_str = format(color_byte, '08b')
    bit_list = [int(bit) for bit in byte_str]

    # Translate each bit into the equivalent 3 bits of SPI MOSI
    for original_bit in bit_list :
        if original_bit == 1:
            mosi_data.extend([1, 1, 0])
        elif original_bit == 0:
            mosi_data.extend([1, 0, 0])

# spidev needs bytes in order to send data
byte_array = bytearray()
for i in range(0, len(mosi_data), 8):
    byte = mosi_data[i:i + 8]
    byte_value = sum(bit << (7 - j) for j, bit in enumerate(byte))
    byte_array.append(byte_value)
        
spi.writebytes(byte_array)
```

Hope it helps

DansDesigns, 2025-03-13:
managed to make some progress using PWM:

``` 
from periphery import PWM
import time

# Use a hardware PWM pin
pwm = PWM(10, 0)  # Chip 10, Channel 0 — adjust for your hardware

# Set frequency to match WS2812B protocol (800kHz signal)
pwm.frequency = 800000
pwm.enable()

# High and low timing for "1" and "0" (WS2812B protocol)
HIGH_TIME_1 = 0.0000008  # 0.8 µs high for "1"
LOW_TIME_1 = 0.00000045  # 0.45 µs low for "1"
HIGH_TIME_0 = 0.0000004  # 0.4 µs high for "0"
LOW_TIME_0 = 0.00000085  # 0.85 µs low for "0"

def set_pwm_duty_cycle(duty):
    pwm.duty_cycle = duty
    time.sleep(0.000001)  # 1µs for testing timing accuracy

def send_bit(bit):
    if bit:
        pwm.duty_cycle = 0.64  # "1" is 64% duty cycle
        time.sleep(HIGH_TIME_1)  # High time for "1"
        pwm.duty_cycle = 0.32  # Low time for "1"
        time.sleep(LOW_TIME_1)  # Low time for "1"
    else:
        pwm.duty_cycle = 0.32  # "0" is 32% duty cycle
        time.sleep(HIGH_TIME_0)  # High time for "0"
        pwm.duty_cycle = 0.64  # Low time for "0"
        time.sleep(LOW_TIME_0)  # Low time for "0"
        
        
def send_byte(byte):
    for i in range(8):
        bit = (byte & (1 << (7 - i))) != 0
        send_bit(bit)

def send_color(r, g, b):
    send_byte(g)
    send_byte(r)
    send_byte(b)

    
def reset():
    pwm.duty_cycle = 0.0  # Force LOW
    time.sleep(0.00005)       # 50ms reset pulse

    
    # Color cycle test
colors = [
    (255, 0, 0),  # Red
    (0, 255, 0),  # Green
    (0, 0, 255),  # Blue
    (255, 255, 0),  # Yellow
    (0, 0, 0)  # Off
]

while True:
    for r, g, b in colors:
        print(f"Color: R={r}, G={g}, B={b}")
        send_color(r, g, b)
        time.sleep(1)

```

However only green lights up, the other colours flash green then it goes blank...

KaliAssistant, 2025-06-29:
> Just in case someone finds it useful. As luckfox-eng33 said, we should be able to control them with SPI.
>
>
>
> 1. You should play with spi.max_speed_hz (between 2500000 and 3500000 depending on your [...]
@DhSufi Works
I should close this issue now
https://github.com/user-attachments/assets/b4c14869-0f77-4916-8e35-21703d17dc09

closed 2025-06-29 by KaliAssistant

plan44, 2025-07-03:
Also just in case:

While the rv1106 has only 2 SPIs and only one free, it has 6 UARTs. I figured that running an UART at 30Mbaud (standard baudrate available in modern UART drivers) and 7-N-1 data format, I can generate very accurate WS28xx timing. Only the direct output is inverted, so you need an inverting level shifter (for reliable operation, you need a 3.3V->5V level shifter anyway). Any HCT inverting logic will do, such as HCT04, HCT02, HCT00 etc.

The trick is, at 3Mbaud one bit has 333nS width, which is good for T0H and T1L. Two times 333nS=666nS works for T0L or T1H. Only for the older WS2811+12, 2.5Mbaud (also standard) might be the better match.

With 7-N-1 you have 9 bits, first always H (inverted start bit) which is either T0H or the first half of T1H, last always 0 (inverted stop bit) which is either T1L or second half of T0L. With the 7 variable bits in between we can make any 3 WS28xx bits.

As UARTs have DMA and output bytes nearly side by side (<400nS pause), this is a quite simple and efficient WS28xx output option.

Below is the C code, which is by now integrated in [vdcd](https://github.com/plan44/vdcd) and the simple [p44ledchaintest utility](https://github.com/plan44/plan44-feed/tree/main/p44ledchaintest).

```C
void sendToUart(int aUartFd, uint8_t* aLedData, size_t aLedDataSize)
{
  // We can generate appropriate WS28xx timing using 7-N-1 @ 3Mhz
  // - UART are LSBit first
  // - WS28xx are MSBit first
  // - we need 1 byte output for 3 bits LED data = 8 byte output for 3 byte LED data
  // - UART idle is H, start bit is L, stop bit is H
  // - as we need idle=L, start=H, stop=L for WS, we need to invert the UART data, and send inverted data
  // UART   : | start | Bit0 | Bit1 | Bit2 | Bit3 | Bit4 | Bit5 | Bit6 | Stop |
  // WS28xx : |   1   | LED7 |   0  |   1  | LED6 |   0  |   1  | LED5 |   0  |
  // WS28xx : |   1   | LED4 |   0  |   1  | LED3 |   0  |   1  | LED2 |   0  |
  // WS28xx : |   1   | LED1 |   0  |   1  | LED0 |   0  |   1  | next |   0  |
  size_t uartDataSize = (aLedDataSize*8+2)/3;
  uint8_t* uartData = malloc(uartDataSize);
  uint8_t* outP = uartData;
  int outBitMask = 0x01; // first LED data bit in UART output
  uint8_t uartByte = 0;
  while(aLedDataSize-- > 0) {
    uint8_t ledByte = *aLedData++;
    uint8_t ledBitMask = 0x80;
    while (ledBitMask) {
      if (outBitMask>1) uartByte |= (outBitMask>>1); // 2nd or 3rd bit in this uart byte: enable the bit position (not idle) by setting the always 1 sync bit (T0H)
      if (ledByte & ledBitMask) uartByte |= outBitMask; // actual data bit, extending T0H to become T1H in case bit is set
      ledBitMask >>= 1; // next input bit
      outBitMask <<= 3; // next output bit position
      if ((outBitMask & 0x7F)==0) {
        // uart byte complete
        *(outP++) = uartByte^0x7F; // send inverted, but only 7 bits
        uartByte = 0;
        outBitMask = 0x01;
      }
    }
  }
  // send
  write(aUartFd, uartData, uartDataSize);
  tcflush(aUartFd, TCIFLUSH);
  // done
  free(uartData);
}
```