LuckfoxTECH/luckfox-pico #136: ubuntu: not enough space available on /run/systemd

KaliAssistant, 2024-07-01:
I am running `apt update` and `apt upgrade`
```
pico@luckfox:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       7.9G  1.9G  5.7G  25% /
tmpfs            29M     0   29M   0% /dev/shm
tmpfs            12M  648K   11M   6% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
/dev/mmcblk1p5  488M   26M  456M   6% /oem
/dev/mmcblk1p6  240M  2.1M  233M   1% /userdata
tmpfs           5.7M     0  5.7M   0% /run/user/1000
```

```
pico@luckfox:~$ sudo apt update
[sudo] password for pico: 
Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease [128 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease [127 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main armhf Packages [888 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main Translation-en [324 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports jammy-updates/restricted armhf Packages [14.6 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports jammy-updates/restricted Translation-en [352 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe armhf Packages [825 kB]                    
Get:10 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe Translation-en [253 kB]                   
Get:11 http://ports.ubuntu.com/ubuntu-ports jammy-updates/multiverse armhf Packages [4256 B]                 
Get:12 http://ports.ubuntu.com/ubuntu-ports jammy-updates/multiverse Translation-en [10.8 kB]                
Get:13 http://ports.ubuntu.com/ubuntu-ports jammy-backports/main armhf Packages [66.7 kB]                    
Get:14 http://ports.ubuntu.com/ubuntu-ports jammy-backports/main Translation-en [11.0 kB]                    
Get:15 http://ports.ubuntu.com/ubuntu-ports jammy-backports/universe armhf Packages [26.0 kB]                
Get:16 http://ports.ubuntu.com/ubuntu-ports jammy-backports/universe Translation-en [16.5 kB]                
Fetched 3048 kB in 11s (284 kB/s)                                                                            
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
109 packages can be upgraded. Run 'apt list --upgradable' to see them.
```
```
pico@luckfox:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following NEW packages will be installed:
  libxcb-randr0
The following packages will be upgraded:
  alsa-ucm-conf apt apt-utils base-files bash binutils binutils-arm-linux-gnueabihf binutils-common bsdutils
  cmake cmake-data coreutils curl distro-info-data dns-root-data dnsmasq-base dpkg dpkg-dev ethtool fdisk
  gdb git git-man gtk-update-icon-cache less libapt-pkg6.0 libarchive13 libbinutils libblkid1 libbluetooth3
  libc-bin libc-dev-bin libc-devtools libc6 libc6-dbg libc6-dev libcryptsetup12 libctf-nobfd0 libctf0
  libcups2 libcurl3-gnutls libcurl4 libdpkg-perl libexpat1 libexpat1-dev libfdisk1 libgbm1
  libgdk-pixbuf-2.0-0 libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libglib2.0-0 libglib2.0-data libgnutls30
  libgtk-3-0 libgtk-3-bin libgtk-3-common libip4tc2 libldap-2.5-0 libldap-common libmm-glib0 libmount1
  libndp0 libnetplan0 libnghttp2-14 libnss-systemd libpam-modules libpam-modules-bin libpam-runtime
  libpam-systemd libpam0g libpulse0 libsmartcols1 libsqlite3-0 libssh-4 libssl-dev libssl3 libsystemd0
  libtiff5 libudev1 libuuid1 libuv1 libvpx7 libxml2 libxtables12 linux-libc-dev login mesa-va-drivers
  mesa-vdpau-drivers modemmanager mount netplan.io openssh-client openssh-server openssh-sftp-server openssl
  passwd systemd systemd-hwe-hwdb systemd-sysv tar tcpdump tzdata udev util-linux vim vim-common vim-runtime
  wget xxd
109 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 88.7 MB of archives.
After this operation, 3255 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main armhf libc6-dbg armhf 2.35-0ubuntu3.8 [11.1 MB]
Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main armhf libc-devtools armhf 2.35-0ubuntu3.8 [29.7 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main armhf libc6-dev armhf 2.35-0ubuntu3.8 [1332 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main armhf libc-dev-bin armhf 2.35-0ubuntu3.8 [19.0 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main armhf linux-libc-dev armhf 5.15.0-113.123 [1305 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main armhf libc6 armhf 2.35-0ubuntu3.8 [2901 kB]    
[... 104 lines cut ...]
Get:47 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main armhf passwd armhf 1:4.8.1-2ubuntu2.2 [742 kB]
Fetched 88.7 MB in 1min 38s (907 kB/s)                                                                       
Extracting templates from packages: 100%
Preconfiguring packages ...
(Reading database ... 48133 files and directories currently installed.)
Preparing to unpack .../0-libc6-dbg_2.35-0ubuntu3.8_armhf.deb ...
Unpacking libc6-dbg:armhf (2.35-0ubuntu3.8) over (2.35-0ubuntu3.4) ...
Preparing to unpack .../1-libc-devtools_2.35-0ubuntu3.8_armhf.deb ...
Unpacking libc-devtools (2.35-0ubuntu3.8) over (2.35-0ubuntu3.4) ...
Preparing to unpack .../2-libc6-dev_2.35-0ubuntu3.8_armhf.deb ...
Unpacking libc6-dev:armhf (2.35-0ubuntu3.8) over (2.35-0ubuntu3.4) ...
Preparing to unpack .../3-libc-dev-bin_2.35-0ubuntu3.8_armhf.deb ...
Unpacking libc-dev-bin (2.35-0ubuntu3.8) over (2.35-0ubuntu3.4) ...
Preparing to unpack .../4-linux-libc-dev_5.15.0-113.123_armhf.deb ...
Unpacking linux-libc-dev:armhf (5.15.0-113.123) over (5.15.0-89.99) ...
Preparing to unpack .../5-libc6_2.35-0ubuntu3.8_armhf.deb ...
Unpacking libc6:armhf (2.35-0ubuntu3.8) over (2.35-0ubuntu3.4) ...
Setting up libc6:armhf (2.35-0ubuntu3.8) ...
Failed to reload daemon: Refusing to reexecute, not enough space available on /run/systemd. Currently, 10.6M are free, but a safety buffer of 16.0M is enforced.
dpkg: error processing package libc6:armhf (--configure):
 installed libc6:armhf package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 libc6:armhf
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

luckfox-eng33, 2024-07-01:
It seems that it is caused by insufficient memory. You can try to expand the swap partition (please google it)

KaliAssistant, 2024-07-01:
@luckfox-eng33  Thanks. I fix it 
`mount -o remount,size=99M  /run/`

closed 2024-07-01 by KaliAssistant