LuckfoxTECH/luckfox-pico #138: Source for upgrade_tool?

plan44, 2024-07-07:
I would like to look into why `upgrade_tool` neither correctly works on modern M2 macs, nor in virtual machines, probably due to timing issues (after a promising start, always: "communication error"). Also, `upgrade_tool` segfaults when the current directory path contains spaces.

However, I only found prebuilt binaries - it would help a lot if the source code was available. Maybe it is, but I haven't found it?

luckfox-eng33, 2024-07-09:
rk does not provide the source code of `upgrade_tool`, which is also a regret for us.

plan44, 2024-07-09:
Ok, thanks @luckfox-eng33 for the answer.

So I guess there's little we can do except hope RK will eventually create versions that also work on ARM based PCs and macs.

Usually all Win tools I need for industrial embedded development run fine in a VM on my M2 mac, be it Win or Linux, native arm or x86. upgrade_tool (and SocToolkit) is one of the few exceptions that seems to work *only* when run on x86 win hardware 😢

plan44, 2025-06-03:
I guess the problem is not the emulation per se, but the handling of USB connections/disconnections.

Because both SocToolkit and upgrade_tool start their operation with the rockchip target in boot mode, but then cause the target to disconnect and reconnect with a different USB identity. Unfortunately, VMs are not prepared to handle the reconnect immediately, but usually need manual confirmation or even virtual re-plugging via menu. That takes way to long for SocToolkit/upgrade_tool, which only waits a short time for the reconnect to happen and then errors out.

Maybe there is a way to make qemu/UTM auto-re-connect somehow? I did some googling but did not find a solution yet. Still, I think qemu + some scripting should be able to do that. I just don't have the time right now to figure it out.

In the meantime, I dug out an old x86 HP laptop where I can use SocToolkit, and hope I'll get OpenWrt's self-update running soon, so I need the x86 brick only for initial flashing of new hardware 😉