LuckfoxTECH/luckfox-pico #147: run luckfox pico rv1106 as an hid device

Cosinusjustafonction, 2024-07-30:
hello i have a concern and i hope someone could give me an answer i would like to use my luckfoxpico rv1106 as a mouse (emulate mouse function) any idea ?

luckfox-eng33, 2024-08-02:
I think this should be easier to implement. It should be possible to modify the kernel and device tree. 
You can refer to the [tutorial of Raspberry Pi zero](https://www.isticktoit.net/?p=1383).

luckfox-eng33, 2024-08-02:
Although I haven't tested it specifically

danielw1990, 2025-05-31:
Where is this documented?
For me it does something weird with the keyboard. I don't know the full extent but the symptoms I noticed are:
spamming the M-key and doing something with caps lock.

Medak1, 2025-06-01:
> Where is this documented?
> For me it does something weird with the keyboard. I don't know the full extent but the symptoms I noticed are:
> spamming the M-key and doing something with caps lock.

Make sure that you have the correct descriptor for the device, and use a packet sniffer to make sure that the emulated device HID descriptor matches what you need, and to see what the reports are.

Good luck

danielw1990, 2025-06-01:
Ah, okay, this makes sense. I used the default settings of the image it came with. But well, it definitely just works. And now I even know what to look for.

And now I see that it generated a bunch of new contents in that config file after enabling hid and rebooting. Many of which I can already recognize as USB device settings.