LuckfoxTECH/luckfox-pico #150: ttyS3 crashes my luckfox pico ultra

Cosinusjustafonction, 2024-08-09:
when i run this command stty -F /dev/ttyS2 my luckfox pico ultra reboots it works fine with ttyS3

luckfox-eng33, 2024-08-09:
If you have a serial port module, it is best to provide log information.
And make sure your device tree does not conflict with other peripherals

Cosinusjustafonction, 2024-08-09:
`lsof /dev/ttyS2
1       /bin/busybox    0       /dev/console
1       /bin/busybox    1       /dev/console
1       /bin/busybox    2       /dev/console
86      /bin/busybox    0       socket:[5192]
86      /bin/busybox    1       /dev/null
86      /bin/busybox    2       /dev/null
86      /bin/busybox    3       /tmp/messages
90      /bin/busybox    0       /dev/null
90      /bin/busybox    1       /dev/null
90      /bin/busybox    2       /dev/null
90      /bin/busybox    3       socket:[5199]
101     /sbin/udevd     0       /dev/null
101     /sbin/udevd     1       /dev/null
101     /sbin/udevd     2       /dev/null
101     /sbin/udevd     3       socket:[5217]
101     /sbin/udevd     4       socket:[5218]
101     /sbin/udevd     5       /dev/kmsg
101     /sbin/udevd     6       anon_inode:inotify
101     /sbin/udevd     7       anon_inode:[signalfd]
101     /sbin/udevd     8       socket:[5221]
101     /sbin/udevd     9       socket:[5222]
101     /sbin/udevd     10      anon_inode:[eventpoll]
177     /usr/bin/dbus-daemon    0       /dev/null
177     /usr/bin/dbus-daemon    1       /dev/null
177     /usr/bin/dbus-daemon    2       /dev/null
177     /usr/bin/dbus-daemon    3       anon_inode:[eventpoll]
177     /usr/bin/dbus-daemon    4       socket:[5407]
177     /usr/bin/dbus-daemon    5       anon_inode:inotify
177     /usr/bin/dbus-daemon    6       socket:[5412]
177     /usr/bin/dbus-daemon    7       socket:[5413]
177     /usr/bin/dbus-daemon    8       socket:[5437]
184     /usr/libexec/bluetooth/bluetoothd       0       /dev/null
184     /usr/libexec/bluetooth/bluetoothd       1       /dev/null
184     /usr/libexec/bluetooth/bluetoothd       2       /dev/null
184     /usr/libexec/bluetooth/bluetoothd       3       anon_inode:[eventfd]
184     /usr/libexec/bluetooth/bluetoothd       4       socket:[5432]
184     /usr/libexec/bluetooth/bluetoothd       5       socket:[5433]
184     /usr/libexec/bluetooth/bluetoothd       6       socket:[5436]
184     /usr/libexec/bluetooth/bluetoothd       7       socket:[5441]
184     /usr/libexec/bluetooth/bluetoothd       8       socket:[5442]
184     /usr/libexec/bluetooth/bluetoothd       9       socket:[5443]
184     /usr/libexec/bluetooth/bluetoothd       10      anon_inode:[signalfd]
215     /usr/sbin/ntpd  0       /dev/null
215     /usr/sbin/ntpd  1       /dev/null
215     /usr/sbin/ntpd  2       /dev/null
215     /usr/sbin/ntpd  3       socket:[5548]
215     /usr/sbin/ntpd  5       pipe:[5540]
215     /usr/sbin/ntpd  16      socket:[5557]
215     /usr/sbin/ntpd  17      socket:[5568]
215     /usr/sbin/ntpd  18      socket:[5569]
215     /usr/sbin/ntpd  19      socket:[5570]
215     /usr/sbin/ntpd  20      socket:[5571]
215     /usr/sbin/ntpd  21      socket:[6281]
248     /bin/busybox    0       /dev/null
248     /bin/busybox    1       /dev/console
248     /bin/busybox    2       /dev/console
256     /usr/sbin/sshd  0       /dev/null
256     /usr/sbin/sshd  1       /dev/null
256     /usr/sbin/sshd  2       /dev/null
256     /usr/sbin/sshd  3       socket:[5687]
270     /bin/busybox    0       /dev/null
270     /bin/busybox    1       /dev/null
270     /bin/busybox    2       /dev/null
270     /bin/busybox    3       socket:[5721]
286     /bin/busybox    0       /dev/null
286     /bin/busybox    1       /dev/console
286     /bin/busybox    2       /dev/console
286     /bin/busybox    3       pipe:[5768]
286     /bin/busybox    4       pipe:[5768]
[... 332 lines cut ...]
[    4.017120] udevd[288]: failed to execute '/lib/udev/v4l_id' 'v4l_id /dev/video2': No such file or directory
[    4.020591] udevd[289]: failed to execute '/lib/udev/v4l_id' 'v4l_id /dev/video3': No such file or directory
[    4.021912] udevd[290]: failed to execute '/lib/udev/v4l_id' 'v4l_id /dev/video4': No such file or directory
[    4.023172] udevd[291]: failed to execute '/lib/udev/v4l_id' 'v4l_id /dev/video5': No such file or directory
[    4.046352] udevd[293]: failed to execute '/lib/udev/v4l_id' 'v4l_id /dev/video6': No such file or directory
[    4.301744] file system registered
[    4.339536] using random self ethernet address
[    4.339572] using random host ethernet address
[    4.361857] Mass Storage Function, version: 2009/09/11
[    4.361958] LUN: removable file: (no medium)
[    4.451589] read descriptors
[    4.451665] read strings
[    5.455443] usb0: HOST MAC 76:f4:f0:6b:c6:9f
[    5.455470] usb0: MAC 1e:2b:60:e1:b9:67
[    5.565791] dwc3 ffb00000.usb: device reset
[    5.669456] android_work: sent uevent USB_STATE=CONNECTED
[    5.752315] android_work: sent uevent USB_STATE=CONFIGURED
[   13.343789] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /serial@ff4d0000/pinctrl-0
[   13.343826] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /serial@ff4d0000/status
[   13.348257] ff4d0000.serial: ttyS3 at MMIO 0xff4d0000 (irq = 63, base_baud = 1500000) is a 16550A
[   13.907562] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /serial@ff4c0000/pinctrl-0
[   13.907603] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /serial@ff4c0000/status
[   13.914709] ff4c0000.serial: ttyS2 at MMIO 0xff4c0000 (irq = 58, base_baud = 1500000) is a 16550A
[   14.226458] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /pwm@ff360030/pinctrl-0
[   14.226498] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /pwm@ff360030/statu`

luckfox-eng33, 2024-08-09:
This log is not readable
It is best to start recording with a single boot until you encounter a bug