LuckfoxTECH/luckfox-pico #155: Why can't I control GPIO3_B0

Game106, 2024-08-23:
Development board model:  Luckfox Pico Pro/Max

My operation process is as follows:
echo 104 > /sys/class/gpio/export       success
echo out > /sys/class/gpio/gpio104/direction     success
echo 1 > /sys/class/gpio/gpio104/value      fail,The value has not changed, it has always been 0

Check the "rv1106-pinctrl.dtsi" file, where the GPIO3_B0 pin is referenced in "vicapm0_pins" and "mipi_pins"
Then I checked the files "rv1106-luckfox-pico-pro-max-ipc.dtsi" and "rv1106g-luckfox-pico-pro-max.dts" and found that "vicapm0_pins" was not referenced
Then in the file "rv1106-luckfox-pico-pro-max-ipc.dtsi", it was found that "mipi_pins" was referenced but commented out by me
Then I recompiled the kernel and burned it into RV1106, but still couldn't control the GPIO3_B0 pin. Why is this?

luckfox-eng33, 2024-08-23:
GPIO3_B0 is not exported, how do you use it?
You can use the io command to directly read the gpio register to determine the io function. Please refer to the documentation（[Rockchip_RV1106_User_Manual_GPIO.pdf](https://github.com/user-attachments/files/16725839/Rockchip_RV1106_User_Manual_GPIO.pdf)） for details.

Game106, 2024-08-23:
> GPIO3_B0 没有导出，如何使用？可以使用 io 命令直接读取 gpio 读取来确定 io 功能。具体 请参考文档（[Rockchip_RV1106_User_Manual_GPIO.pdf ）。](https://github.com/user-attachments/files/16725839/Rockchip_RV1106_User_Manual_GPIO.pdf)

感谢回复，不过GPIO3_B0我已经导出了，根据luckfox wiki给出的引脚编号公式，它引脚编号是104。公式如下：
GPIO 的命名方式为 GPIO{bank}_{group}{X}, pin = bank * 32 + (group * 8 + X)
GPIO3_B0 = 3*32 + (1*8+0) = 104

[image]

Game106, 2024-08-23:
> GPIO3_B0 is not exported, how do you use it? You can use the io command to directly read the gpio register to determine the io function. Please refer to the [...]

Thank you for your reply, but I have already exported GPO3_B0. According to the formula provided on the luckfox wiki, its pin number is 104。The formula is as follows：
GPIOs are named in the format GPIO{bank}_{group}{X}, pin = bank * 32 + (group * 8 + X)
So, GPIO3_B0 = 3x32 + (1x8 + 0 ) = 104
[image]

luckfox-eng33, 2024-08-23:
I may have misdescribed it, to be precise I meant that this pin is not connected out through the PCB (in the physical sense)

Game106, 2024-08-27:
The reason why GPO3_B0 cannot be operated is that it has not been configured as a GPIO function under DTS

Game106, 2024-08-28:
GPO3_B0 to GPO3_C3 only support input mode

Soheiljbb, 2025-12-30:
Hi @Game106,

Thank you for your previous explanations regarding GPIO3_B0 to GPIO3_C3. I have a question about your comment on “commenting out the mipi_pins.”

I tried doing that in my config/dts_config as follows:

&rkcif_mipi_lvds {
        pinctrl-0 = <>;
};

However, when I read the pins (e.g., GPIO115), the values are always 0 even when I apply 1.8 V to the pin:
#
root@rv1106:~# cat /sys/class/gpio/gpio115/direction
in
root@rv1106:~# cat /sys/class/gpio/gpio115/value
0

I also disabled the entire MIPI subsystem and its dependencies in "rv1106-luckfox-pico-pro-max-ipc.dtsi" again in config/dts_config as below:

&mipi0_csi2_hw{ status = "disabled"; };
&mipi1_csi2_hw{ status = "disabled"; };
&csi2_dphy_hw { status = "disabled"; };
&csi2_dphy0 { status = "disabled"; };
&mipi0_csi2 { status = "disabled"; };
&rkcif { status = "disabled"; };
&rkcif_mipi_lvds { status = "disabled"; pinctrl-0 = <>; };
&rkcif_mipi_lvds_sditf { status = "disabled"; };
&rkisp { status = "disabled"; };
&rkisp_vir0 { status = "disabled"; };

Despite this, the GPIO still reads 0.

So my question is: was the commenting out of mipi_pins the only step required on your side to successfully read GPIO3_B0 to GPIO3_C values, or is there something else I am missing?

Thank you in advance for your guidance!

Game106, 2025-12-30:
来信我已收到，谢谢。