LuckfoxTECH/luckfox-pico #156: Wifi adapter for Pico mini B

fredsco, 2024-08-25:
I want to create a hotspot on my Pico Mini B so that other devices can discover and connect to it from the Wifi list and then they can see the video stream (rtsp) 

So far :

- Camera module works
- recompiled kernel with host mode enabled and verified this on device. 

Now I ordered a TP-Link TL-WN725N this is a wifi adapter which uses the RTL8188EUS chipset. If I look in this repository I do see a driver for RTL8188EU... 

My questions are: 

- Will my adapter work given that mine uses the **RTL8188EUS** chipset and the repository only has **RTL8188EU** that is close to it. 
- If not what adapter would you recommend?
- Do I need to perform any specific steps to ensure this driver is loaded / included during compilation.. for now I assume it is already there in the OS.

luckfox-eng33, 2024-08-26:
We have not tested the RTL8188EUS module and we cannot be sure it will work properly.
Generally speaking, module adaptation requires technical support from the module supplier, so it is recommended that you contact your module supplier to provide technical support, and it may be possible to adapt.
We transplanted AIC8800DC on [Luckfox-Pico-Ultra-W](https://www.luckfox.com/EN-Luckfox-Pico-Ultra-W), maybe you can refer to this

fredsco, 2024-08-28:
@luckfox-eng33 Thank you I will try a aic8800dc dongle. 

Do you know if I need to modify something in the Luckfox Pico build / folder to activate /load the aic8800dc on Luckfox Pico Mini B?

luckfox-eng33, 2024-08-28:
The simplest and stupidest method is to compile Luckfox-Pico-Ultra-W, then copy the corresponding aic-related ko, manually copy it to the board and load it
The smart way is to check the[ Luckfox-Pico-Ultra-W configuration file](https://github.com/LuckfoxTECH/luckfox-pico/blob/main/project/cfg/BoardConfig_IPC/BoardConfig-EMMC-Ubuntu-RV1106_Luckfox_Pico_Ultra_W-IPC.mk) and modify your configuration file

fredsco, 2024-09-03:
@luckfox-eng33 Thank you. I have decided to get the RTL8192CU as during compilation the exact same drivers are build. However after booting up while the device is available under lsusb. there is no wlan0. i

In Luckfox_rv1106_linux_defconfig I have;

``` 
CONFIG_RTL8192CU=y
```

in the MK file for rv1103 (Luckfox Pico mini b) I have:

```
export RK_ENABLE_WIFI=y
```

fredsco, 2024-09-03:
@luckfox-eng33 I have managed to make the Wifi work using a RTL8192CU. CONFIG_RTL8192CU  was first set to m.

noon92, 2024-10-26:
@fredsco after setting to y, did wlan0 just appear when the usb device was connected?