LuckfoxTECH/luckfox-pico #158: Driver assistant RK not working on ARM windows

fredsco, 2024-08-25:
I'm unable to install the drivers on  Windows 11 ARM (Virtual Machine) I have tried to install the drivers manually but they are not recognized... 

When installing drivers using [DriverAssitant-RK](https://files.luckfox.com/wiki/Luckfox-Pico/Software/DriverAssitant_v5.12.zip).

 I get the following error in Windows 11 ARM

"Running installation Failed" 

```
10:34:14 389	"\\Mac\Home\Downloads\DriverAssitant_v5.12\DriverAssitant_v5.12\bin\x64\RKDrvPreinst_x64.exe" /i "\\Mac\Home\Downloads\DriverAssitant_v5.12\DriverAssitant_v5.12\Driver\x64\win10\rockusb.inf"  /log
10:34:15 034	"\\Mac\Home\Downloads\DriverAssitant_v5.12\DriverAssitant_v5.12\bin\x64\RKDrvPreinst_x64.exe" /i "\\Mac\Home\Downloads\DriverAssitant_v5.12\DriverAssitant_v5.12\ADBDriver\android_winusb.inf"  /log
10:34:15 477	Install driver ok.

10:34:15 477	Info:InstallDriverProc-->RunOhterExe failed
```

it would be great to add support for Windows 11 ARM.

luckfox-eng33, 2024-08-26:
The driver is officially maintained by RK. We will report this issue back later, but we are not sure whether they will adapt it in time.
In addition, can you provide us with the model of the equipment you are using?

FusionPlmH, 2024-09-03:
Same here i running at 8cx gen2 laptop which is a native windows 11 arm

X-212, 2025-10-24:
You have to disable driver signature enforcement and then you can install your drivers. It resets every reboot, but the driver stays.