LuckfoxTECH/luckfox-pico #159: Cannot find kernel headers.

fredsco, 2024-08-26:
Where can I find the kernel headers for :https://drive.google.com/drive/folders/18ONAwpvh6cACw5CUql7ec-UeZcl0N4sN?usp=share_link 

```
uname -r
5.10.110
```

I can't find the specific version, also would like in .deb format if possible?

luckfox-eng33, 2024-08-28:
We do not provide kernel headers directly
If you need kernel headers, you can cross-program the kernel headers yourself via PC instead of using luckfox pico (which is too difficult for it)

closed 2024-09-04 by fredsco

noon92, 2024-10-22:
@fredsco did you ever figure out how go get the headers?

fredsco, 2024-10-26:
@noon92 No but have you tried running kernelconfig from PC?

noon92, 2024-10-26:
Yes, but adding headers via kernelconfig just doesn't work. After build it's deselected.

AliMoal, 2025-02-05:
@luckfox-eng33 Hi, Cross-program sometimes is impossible, I want to write some kernel modules. how to do it?

luckfox-eng33, 2025-02-07:
I think [the wiki](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-SDK/#53-application-example) has explained it clearly.
@AliMoal