LuckfoxTECH/luckfox-pico #160: How to connect the DVP pin of RV1106

Game106, 2024-08-27:
I want to use the DVP function of RV1106, and the camera has an 8-bit signal input. I would like to ask if the D0~D7 pins of the camera should be connected to the VI_CIF_D0~VI_CIF_D7 pins of RV1106, or to the VI_CIF_D8~VI_CIF_D15 pins? I don't have the relevant data manual for RV1106, so I don't know how to connect it.

luckfox-eng33, 2024-08-28:
Currently, luckfox pico series does not support DVP cameras.

denismanus, 2025-02-02:
@luckfox-eng33 and what about now?

Game106, 2025-02-02:
来信我已收到，谢谢。

luckfox-eng33, 2025-02-05:
We do not intend to adapt to DVP cameras for the time being @denismanus

Maobuff, 2025-04-30:
> I want to use the DVP function of RV1106, and the camera has an 8-bit signal input. I would like to ask if the [D0:D7] pins of the camera should be connected to the [VI_CIF_D0:VI_CIF_D7] pins of [...]

[image]

Game106, 2025-04-30:
来信我已收到，谢谢。

Game106, 2025-07-06:
来信我已收到，谢谢。

Maobuff, 2025-07-06:
@mrigi Only rockchip rv1106g2/rv1106g3 have support for dvp. rv1103g1/rv1103b have only mipi-csi.

mrigi, 2025-07-06:
Okay, did you found how to wire to rv1106?
I have rv1106g3 also in form of "pico pi a" board.
Idk which gpio to wire.

Maobuff, 2025-07-06:
@mrigi Custom board, I will provide schematics later. You can look at screenshot what I provided earlier. Basically you have to use upper bits of DVP interface if you are not using full bus width.

mrigi, 2025-07-07:
Ok, thanks. So you mapped to D2-D9 instead of D8-D15. Got working dts file for it?

Maobuff, 2025-07-08:
@mrigi Yes, vicapm1_pins are located in rv1106-pinctrl.dtsi file
```
&pinctrl {
...
    vicap {
    ...
/omit-if-no-ref/
		vicapm1_pins: vicapm1-pins {
			rockchip,pins =
				/* vicap_clkin_m1 */
				<1 RK_PD0 2 &pcfg_pull_none>,
				/* vicap_d0_m1 */
				<1 RK_PA2 3 &pcfg_pull_none>,
				/* vicap_d1_m1 */
				<1 RK_PB1 4 &pcfg_pull_none>,
				/* vicap_d2_m1 */
				<1 RK_PC0 2 &pcfg_pull_none>,
				/* vicap_d3_m1 */
				<1 RK_PC1 2 &pcfg_pull_none>,
				/* vicap_d4_m1 */
				<1 RK_PC2 2 &pcfg_pull_none>,
				/* vicap_d5_m1 */
				<1 RK_PC3 2 &pcfg_pull_none>,
				/* vicap_d6_m1 */
				<1 RK_PC4 2 &pcfg_pull_none>,
				/* vicap_d7_m1 */
				<1 RK_PC5 2 &pcfg_pull_none>,
				/* vicap_d8_m1 */
				<1 RK_PC6 2 &pcfg_pull_none>,
				/* vicap_d9_m1 */
				<1 RK_PC7 2 &pcfg_pull_none>,
				/* vicap_hsync_m1 */
				<1 RK_PD1 2 &pcfg_pull_none>,
				/* vicap_vsync_m1 */
				<1 RK_PD2 2 &pcfg_pull_none>;
		};
    ...
    };
...
};
```

Take note that this pinctrl will occupy D0-D1 even if you are using 8bit bus.
I advise to create your own definition. Also HSYNC/VSYNC can be dropped if you are using BT656