LuckfoxTECH/luckfox-pico #162: CSI Camera does not work when manually compiling buildroot

fredsco, 2024-09-03:
When I manually compile the build root no /dev/video0 is available and no files in /userdata/ is present. When using the images from google drive the camera does work. Do I need to enable something in kernel menu config?

luckfox-eng33, 2024-09-04:
Except for Ultra/Ultra W, all other boards are turned on by default, and you don’t need to do anything else.

fredsco, 2024-09-04:
@luckfox-eng33 Hi I recompiled again and the camera is simply not working. If I use the img files from the google drive link everything works fine. 

I think during compilation the drivers are not loaded. 

How Can I ensure it is on in the luck fox-pico SDK?

luckfox-eng33, 2024-09-04:
If you can adb or ssh into the luckfox pico system, you can try `luckfox config` (the quick configuration tool we provide) and turn on CSI
For details on how to use it, you can check the [wiki](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-RV1106/Luckfox-Pico-Ultra-W/Luckfox-config/)

fredsco, 2024-09-04:
@luckfox-eng33 Thanks for the tip. Since I have successfully setup wlan0 I indeed can SSH into the device. I have done so and ran the Luckfox-config command. 

I have enabled CSI but I still do not see any files in /userdata/ 

The files such as video0 etc. do show up when I use the image files from the google drive link....

luckfox-eng33, 2024-09-04:
1. Please provide complete system log information, such as "dmesg"
2. Based on your questions on other issues, I reasonably doubt whether you have used root permissions and modified other files at will. I suggest you re-pull the sdk and recompile to test it.

fredsco, 2024-09-04:
I have done the following things:

1. clone the pico SDK.
2. set USB to host mode using the following instructions: https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-USB
3.`run sudo ./build.sh buildrootconfig`  select right board, SD Card, build root -> and enable wpa_supplicant , dhcpcd, dnsmasq
4. lastly run `sudo ./build.sh` and wait for a while until output/image

Dmesg output: 

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.110 (root@kitty) (arm-rockchip830-linux-uclibcgnueabihf-gcc (crosstool-NG 1.24.0) 8.3.0, GNU ld (crosstool-NG 1.24.0) 2.32) #1 Tue Sep 3 22:24:34 CEST 2024
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=50c53c7d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Luckfox Pico Mini B
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 24576 KiB at 0x02800000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: user_debug=31 storagemedia=sd androidboot.storagemedia=sd androidboot.mode=normal  rootwait earlycon=uart8250,mmio32,0xff4c0000 console=ttyFIQ0 snd_soc_core.prealloc_buffer_size_kbytes=16 coherent_pool=0 root=/dev/mmcblk1p7 rootfstype=ext4 rk_dma_heap_cma=24M blkdevparts=mmcblk1:32K(env),512K@32K(idblock),256K(uboot),32M(boot),512M(oem),256M(userdata),6G(rootfs),-(media)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 33000K/65536K available (4236K kernel code, 367K rwdata, 1980K rodata, 196K init, 149K bss, 7960K reserved, 24576K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x1fb/0x360 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000006] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000017] Switching to timer-based delay loop, resolution 41ns
[    0.000409] Console: colour dummy device 80x30
[    0.000455] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=80000)
[    0.000473] pid_max: default: 4096 minimum: 301
[    0.000642] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000655] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001269] CPU: Testing write buffer coherency: ok
[    0.001782] Setting up static identity map for 0x81c0 - 0x81f8
[    0.002185] devtmpfs: initialized
[    0.008906] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.009097] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6370867519511994 ns
[    0.009122] futex hash table entries: 16 (order: -5, 192 bytes, linear)
[    0.009209] pinctrl core: initialized pinctrl subsystem
[    0.010037] NET: Registered protocol family 16
[    0.010484] DMA: failed to allocate 0 KiB pool for atomic coherent allocation
[    0.011262] thermal_sys: Registered thermal governor 'step_wise'
[    0.011272] thermal_sys: Registered thermal governor 'user_space'
[    0.011520] cpuidle: using governor menu
[    0.011557] Registered FIQ tty driver
[    0.023712] rockchip-gpio ff380000.gpio: probed /pinctrl/gpio@ff380000
[    0.024478] rockchip-gpio ff530000.gpio: probed /pinctrl/gpio@ff530000
[    0.025211] rockchip-gpio ff550000.gpio: probed /pinctrl/gpio@ff550000
[    0.025854] rockchip-gpio ff560000.gpio: probed /pinctrl/gpio@ff560000
[    0.025956] rockchip-pinctrl pinctrl: probed pinctrl
[    0.035799] fiq_debugger fiq_debugger.0: IRQ uart_irq not found
[    0.035824] fiq_debugger fiq_debugger.0: IRQ wakeup not found
[    0.036095] printk: console [ttyFIQ0] enabled
[... 139 lines cut ...]
[    0.543393] usb 1-1: Direct firmware load for rtlwifi/rtl8192cufw.bin failed with error -2
[    0.543419] rtlwifi: Loading alternative firmware rtlwifi/rtl8192cufw.bin
[    0.543428] rtlwifi: Selected firmware is not available
[    0.804428] udevd[88]: starting version 3.2.7
[    0.810667] udevd[88]: specified group 'input' unknown
[    0.811796] udevd[88]: specified group 'kvm' unknown
[    0.844012] udevd[89]: starting eudev-3.2.7
[    1.959303] EXT4-fs (mmcblk1p5): recovery complete
[... cut ...]
[    2.996850] rtl8192cu: MAC auto ON okay!
[    3.013046] rtl8192cu: Tx queue select: 0x05
[    3.574380] rtl8192c_common: Polling FW ready fail! REG_MCUFWDL:0x00000006.
[    3.574407] rtl8192c_common: Firmware is not ready to run!
[    4.346638] file system registered
[    4.376150] using random self ethernet address
[    4.376189] using random host ethernet address
[    4.399858] Mass Storage Function, version: 2009/09/11
[    4.399971] LUN: removable file: (no medium)
[    4.512548] read descriptors
[    4.512664] read strings
```

luckfox-eng33, 2024-09-05:
From the log, I think you lost part of the SDK configuration, causing the camera driver to not register properly.
Please do not make any modifications and use the original SDK for testing.
Also, when compiling buildroot or busybox, please do not use sudo, which may cause permission confusion.

fredsco, 2024-09-05:
@luckfox-eng33 Thank you, so you are saying I cannot  enable wpa_supplicant and USB HOST mode?

luckfox-eng33, 2024-09-05:
I want you to use the original SDK for testing, and first eliminate problems with our SDK.
Then you can modify it step by step and then determine the specific problem. It does not restrict you from modifying the corresponding configuration.
Based on the existing information, I just think there is a problem with your configuration. You need to solve the specific problem by yourself.

fredsco, 2024-09-05:
@luckfox-eng33 I deleted the pico-idk folder and cloned it again. when building I can already see some new builds regarding to drivers/video which I didn't see previously. 

Configuring the buildroot and kernel still show me that drivers/video is being built 

Perhaps as you said using sudo messed things up. I will keep this thread updated once my built is done and everything works.

fredsco, 2024-10-02:
@luckfox-eng33 Even after following steps above Camera is not working

xt8288, 2024-10-08:
[image]
The latest SDK compiled image can successfully detect the camera after enabling CSI.

xt8288, 2024-10-08:
The latest SDK compiled image can successfully detect the camera after enabling CSI.
[image]

fredsco, 2024-10-08:
> The latest SDK compiled image can successfully detect the camera after enabling CSI. [...]

Can you please tell me how you enabled CSI? was it pre built or post? Thank you

xt8288, 2024-10-08:
I simply downloaded the latest SDK without making any modifications, compiled the image, and flashed it onto the development board. After logging into the development board, I enabled it in luckfox-config and rebooted for the changes to take effect.

fredsco, 2024-10-08:
Ok. I just did everything again

literally the only thing I have done are the following steps:

1. git clone https://github.com/LuckfoxTECH/luckfox-pico.git
2.  cd {SDK_PATH}/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/
3. source env_install_toolchain.sh
4. cd ~/{SDK_PATH}
5. ./build.sh
6. select pico mini B -> SD card

**This works, I see videoX now under userdata**

but once I :

1. ./build.sh clean
2. enable USB host mode -> https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-USB
3. enable RTL8192 in kernel config (./build.sh kernelconfig)
4. and enable wpa_supplicant in buildroot config (./build.sh buildrootconfig)
5. recompile.. (./build.sh)
6. Flash to SD -> no videoX is shown in /userdata/

Why won't this work? is it due to resources? and/or USB HOST mode?

luckfox-eng33, 2024-10-09:
First of all, different issues call for different issues.
Then, voltage jitter is probably caused by the fact that your module requires a large instantaneous startup current

fredsco, 2024-10-26:
short summary: 

It turns out that I forgot to set the necessary environment variables (see README for details). Even if you set the environment variables after first build it will still not work.

```
source env_install_toolchain.sh
```

If this happens to you, do not retry it after you already compiled. Reclone the SDK because it still won't work if you set the environment variables after compilation (even when running a -clean all)

closed 2024-10-26 by fredsco