LuckfoxTECH/luckfox-pico #164: Build failed after changing build root config

fredsco, 2024-09-03:
After changing the build root config with:

```
./build.sh buildrootconfig
```

I get the following error when doing:

```
sudo ./build.sh
```

```
Run-time dependency libffi found: YES 3.4.4
Run-time dependency zlib found: YES 1.2.13
Checking for function "ngettext" : NO
Library intl found: NO

output/build/libglib2-2.72.3/meson.build:2102:4: ERROR: Automatic wrap-based subproject downloading is disabled

A full log can be found at /home/fredsco/dev/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/libglib2-2.72.3/build/meson-logs/meson-log.txt
make[1]: *** [package/pkg-generic.mk:283: /home/fredsco/dev/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/libglib2-2.72.3/.stamp_configured] Error 1
make[1]: Leaving directory '/home/fredsco/dev/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6'
make: *** [Makefile:573: buildroot] Error 2
make: Leaving directory '/home/fredsco/dev/luckfox-pico/sysdrv'
[build.sh:error] Running build_sysdrv failed!
[build.sh:error] exit code 2 from line 653:
[build.sh:info]     make -C ${SDK_SYSDRV_DIR}
```

I have enabled some networking tools including wpa_supplicant...

luckfox-eng33, 2024-09-04:
Please refer to [wiki](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-RV1103/Luckfox-Pico-SDK/)

closed 2024-09-04 by fredsco

prototypicalpro, 2025-05-22:
If anyone else encounters this issue, the solution is to ensure `BR2_PACKAGE_GETTEXT` and `BR2_PACKAGE_GETTEXT_GNU` are enabled. For me this meant enabling `BR2_SYSTEM_ENABLE_NLS`, which had been disabled by something while I was messing with the toolchain.

[image]