LuckfoxTECH/luckfox-pico #165: python or c library to stream video

fredsco, 2024-09-06:
Are there any libraries available to stream the video from the CSI Camera? I have managed to install wifi driver and enable the camera under buildroot and would like to stream the video over the web / localhost.

luckfox-eng33, 2024-09-10:
Maybe you can refer to [wiki](https://wiki.luckfox.com/Luckfox-Pico/RKMPI-example)

fredsco, 2024-10-01:
@luckfox-eng33 Thank you. this might be exactly what I need. How can I modify the IP address for the rtsp ? I can't find a constant in code.. 

Also what are the differences between `luckfox_pico_rtsp_opencv` and `luckfox_pico_rtsp_opencv_capture`

Thank you

luckfox-eng33, 2024-10-09:
Usually you can use your router to assign a fixed IP or refer to [Setting the IP address in Linux](https://www.ibm.com/docs/en/power9?topic=notebook-setting-ip-address-in-linux)

fredsco, 2024-10-09:
@luckfox-eng33 Thank you. I do notice some 3-4 seconds delay when using the luckfox_pico_rtsp_opencv_capture_demo on the Luckfox Pico mini B. Any chance to decrease the delay to ms?

luckfox-eng33, 2024-10-09:
Usually around 1-2S latency is the normal range, 3-4S latency is usually a problem with the playback software

fredsco, 2024-10-09:
> Usually around 1-2S latency is the normal range, 3-4S latency is usually a problem with the playback software

Thank you, that's correct. after modifying the VLC settings and decreasing the network-caching value I immediately noticed lower latency. I have also played with the bitrate.. but for now the latency is pretty acceptable.