LuckfoxTECH/luckfox-pico #168: 取消samba导致无法编译系统

WolfGenerals, 2024-09-22:
```
>>>   Generating filesystem image rootfs.tar
mkdir -p /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/images
rm -rf /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar
mkdir -p /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar
rsync -auH --exclude=/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/target/ /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/target
echo '#!/bin/sh' > /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
echo "set -e" >> /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
echo "chown -h -R 0:0 /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/target" >> /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
PATH="/sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/host/bin:/sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/host/sbin:/sdk/luckfox-pico/sysdrv/tools/pc/toolkits:/sdk/luckfox-pico/sysdrv/tools/pc/uboot_tools:/sdk/luckfox-pico/sysdrv/out/bin/pc:/sdk/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin:/sdk/luckfox-pico/output/out/sysdrv_out/pc:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/support/scripts/mkusers /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/full_users_table.txt /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/target >> /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
echo "/sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/host/bin/makedevs -d /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/full_devices_table.txt /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/target" >> /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
printf '        rm -rf /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/target/usr/lib/udev/hwdb.d/ /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/target/etc/udev/hwdb.d/\n' >> /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
echo "find /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/target/run/ -mindepth 1 -prune -print0 | xargs -0r rm -rf --" >> /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
echo "find /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/target/tmp/ -mindepth 1 -prune -print0 | xargs -0r rm -rf --" >> /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
printf '   \n' >> /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
printf '   \n' >> /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
printf '        (cd /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/target; find -print0 | LC_ALL=C sort -z | tar  --pax-option=exthdr.name=%%d/PaxHeaders/%%f,atime:=0,ctime:=0 -cf /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/images/rootfs.tar --null --xattrs-include='\''*'\'' --no-recursion -T - --numeric-owner)\n' >> /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
chmod a+x /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
PATH="/sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/host/bin:/sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/host/sbin:/sdk/luckfox-pico/sysdrv/tools/pc/toolkits:/sdk/luckfox-pico/sysdrv/tools/pc/uboot_tools:/sdk/luckfox-pico/sysdrv/out/bin/pc:/sdk/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin:/sdk/luckfox-pico/output/out/sysdrv_out/pc:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" FAKEROOTDONTTRYCHOWN=1 /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/host/bin/fakeroot -- /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/fakeroot
rootdir=/sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/tar/target
table='/sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/build/buildroot-fs/full_devices_table.txt'
ln -snf /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/host/arm-buildroot-linux-uclibcgnueabihf/sysroot /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/staging
make[1]: Leaving directory '/sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6'
tar xf /sdk/luckfox-pico/sysdrv/source/buildroot/buildroot-2023.02.6/output/images/rootfs.tar -C /sdk/luckfox-pico/sysdrv/out/rootfs_uclibc_rv1106
cp /sdk/luckfox-pico/sysdrv/tools/board/buildroot/shadow_defconfig /sdk/luckfox-pico/sysdrv/out/rootfs_uclibc_rv1106/etc/shadow
cp /sdk/luckfox-pico/sysdrv/tools/board/buildroot/sshd_defconfig /sdk/luckfox-pico/sysdrv/out/rootfs_uclibc_rv1106/etc/ssh/sshd_config
cp /sdk/luckfox-pico/sysdrv/tools/board/buildroot/samba_defconfig /sdk/luckfox-pico/sysdrv/out/rootfs_uclibc_rv1106/etc/samba/smb.conf
cp: cannot create regular file '/sdk/luckfox-pico/sysdrv/out/rootfs_uclibc_rv1106/etc/samba/smb.conf': No such file or directory
make: *** [Makefile:577: buildroot] Error 1
make: Leaving directory '/sdk/luckfox-pico/sysdrv'
[build.sh:error] Running build_rootfs failed!
[build.sh:error] exit code 2 from line 697:
[build.sh:info]     make rootfs -C ${SDK_SYSDRV_DIR}
```
Luckfox pico Mini B

luckfox-eng33, 2024-09-24:
https://github.com/LuckfoxTECH/luckfox-pico/blob/main/sysdrv/Makefile
注释第601行
`cp $(SYSDRV_DIR)/tools/board/buildroot/samba_defconfig $(SYSDRV_DIR_OUT_ROOTFS)/etc/samba/smb.conf`

WolfGenerals, 2024-09-25:
第602行也要注释
cp $(SYSDRV_DIR)/tools/board/buildroot/smbpasswd_defconfig $(SYSDRV_DIR_OUT_ROOTFS)/etc/samba/smbpasswd

WolfGenerals, 2024-09-25:
希望后续加上条件判断