LuckfoxTECH/luckfox-pico #169: RV1106G3适配RK628D的DSI模式(RGB888)问题

qq8322302, 2024-09-30:
您好，目前状态：
    Luckfox Pico Max (RV1106G3)的MIPI接口接入了RK628D，用于HDMI-IN，我已经在RK628D-CSI模式调试通过，并且可以正常的在CIF节点获取到YUV422的图像数据。(ISP不支持YUV数据，所以我跳过了ISP，直接在CIF节点获取数据)

遇到的问题：
    但由于RK628D在HDMI-IN的CSI模式下有色差，需要切换到DSI模式。我已在DTS中切换到DSI模式，[ media-ctl -p -d /dev/media0 ] 已经可以看到切换为了RGB888_1X24，但是直接在CIF节点取数据会报错。
    取数据的命令[ v4l2-ctl --set-fmt-video=width=1920,height=1080,pixelformat=RGB3 --stream-mmap=4 --stream-to=/tmp/RGB3.yuv --stream-count=1 -d /dev/video0
 ]
    希望能够帮忙解决一下，实现CIF支持RGB888_1X24，然后可以送到ISP进行处理。这样可以直接通过rockit库在ISP中获取到RK628 HDMI-IN的图像。

[image]

luckfox-eng33, 2024-10-09:
1. RV1106支持CSI，不支持DSI
2. 关于色差，因为硬件不是我们提供的，我们没有硬件进行调试，通过RK628D供应商联系原厂进行调试

closed 2024-11-19 by qq8322302

Wisthen, 2025-05-09:
@qq8322302 使用RK628D模块购买连接能发一下吗

qq8322302, 2025-05-10:
@Wisthen    https://item.taobao.com/item.htm?id=743508124573
淘宝上面有很多商家在卖。。
需要注意： 接口与Luckfox 不一样，需要自己再制作一个转接板。

Wisthen, 2025-05-10:
@qq8322302 能加你一下微信吗 我现在遇到一些问题 想请你帮助一下

Wisthen, 2025-05-12:
@qq8322302 我给你发邮件了 这样可以保护一下隐私

qq8322302, 2025-05-13:
@Wisthen   

这个是转接板的原理图+PCB图，使用立创EDA专业版导入即可。
[ProPrj_RV1106-FPC20P-RK628D_2025-05-13.zip](https://github.com/user-attachments/files/20180688/ProPrj_RV1106-FPC20P-RK628D_2025-05-13.zip)

[image]

由于太久没有弄了，只能给你提供这些了。

Wisthen, 2025-05-16:
@qq8322302 可以买你做好的转接板吗