LuckfoxTECH/luckfox-pico #174: how to remove saradc pullup resistor

KaliAssistant, 2024-10-10:
i have a pico max 
but i want read my battery voltage
but the pull-up resistor of the ADC will affect my voltage divider resistors.

luckfox-eng33, 2024-10-10:
The corresponding voltages can be calculated using [Kirchhoff's circuit laws](https://en.wikipedia.org/wiki/Kirchhoff%27s_circuit_laws)

KaliAssistant, 2024-10-10:
@luckfox-eng33 
> The corresponding voltages can be calculated using [Kirchhoff's circuit laws](https://en.wikipedia.org/wiki/Kirchhoff%27s_circuit_laws)

W     H     A     T
i mean pullup resistors in chip , can change by devicetree, kernel ...etc
What if the temperature changes?

KaliAssistant, 2024-10-12:
@luckfox-eng33 
I cant find any config about saradc pullup in iio/adc/rockchip_saradc.c or dts
when my luckfox pico max boot up , the adc val always "1023" ,until the adc  pin conn  to my signal generator

KaliAssistant, 2024-10-12:
@luckfox-eng33 Are you the EE on this project?
dOnT Us3 KiRcHH0Ff`s CiRcUiT lAws
temperature changes will make pull-up resistors(in adc) be different, and Not every die has the same pull-up resistor

KaliAssistant, 2024-10-12:
@luckfox-eng33 
还是你看得懂中文 
~~還是你看得懂繁體中文~~

luckfox-eng33, 2024-10-12:
通常只需要考虑外部电阻即可，但是通常来着说SOC的ADC可能存在一定偏差是正常现象。
如果你对温漂很敏感或者对电压敏感，建议使用外部ADC芯片或者专用的锂电池管理芯片进行处理，通常此类芯片有温漂矫正。

KaliAssistant, 2024-10-12:
但是通常ADC內部在還沒設定的時候應該不會上拉
我的電壓會經過分壓電路，上電阻就會與我的分壓電組並聯

luckfox-eng33, 2024-10-12:
请查看原理图，两个ADC引脚都有外部10K上拉

KaliAssistant, 2024-10-12:
🤦🏻‍♂️

KaliAssistant, 2024-10-12:
我只好拿出我的烙鐵了

luckfox-eng33, 2024-10-12:
请注意 ADC0上电或者重启会检测其电压，
如果过低，会进入下载模式

KaliAssistant, 2024-10-12:
謝啦

closed 2024-10-12 by KaliAssistant

dwhacks, 2025-04-25:
What resistor did you remove @KaliAssistant, I'm trying to do the same thing on SARDC1

KaliAssistant, 2025-04-25:
> What resistor did you remove [@KaliAssistant](https://github.com/KaliAssistant), I'm trying to do the same thing on SARDC1

[image]

KaliAssistant, 2025-04-25:
R57, 5%10K

KaliAssistant, 2025-04-25:
> What resistor did you remove [@KaliAssistant](https://github.com/KaliAssistant), I'm trying to do the same thing on SARDC1

And DO NOT REMOVE SARADC0's resistor, OR chip will always in download mode