LuckfoxTECH/luckfox-pico #175: Changed EMMC from 8GB to 32GB, unable to Boot now.

HTerminal, 2024-10-10:
Hello,

I changed the RV1106 EMMC to 32GB from 8GB, flashed the update.bin it flashed with no problem even flash was also detected.

Started reducing the clock speed of EMMC to be sure if hardware is correct - 200Mhz to 15 Mhz
still unable to boot.

Here is the output logs from serial debug
Please suggest what can be done

Regards 
Hitesh
```
S5P1
f967
rgef1
DDRConf2
DDR3, BW=16 Col=10 Bk=8 CS0 Row=14 CS=1 Die BW=16 Size=256MB
924MHz
DDR bin out

U-Boot SPL board init
U-Boot SPL 2017.09 (Oct 10 2024 - 15:43:54)
unrecognized JEDEC id bytes: 00, 00, 00
unknown raw ID 0 0 0
Trying to boot from MMC2
Card did not respond to voltage select!
mmc_init: -95, time 16
spl: mmc init failed with error: -95
Trying to boot from MMC1
ENVF: Primary 0x00000000 - 0x00008000
ENVF: OK
Trying fit image at 0x440 sector
## Verified-boot: 0
## Checking uboot 0x00200000 (lzma @0x00400000) ... sha256(7be34be5c1...) + sha256(963dff0d3b...) + OK
## Checking fdt 0x0025bed0 ... sha256(d4cf0ee956...) + OK
Total: 327.698 ms

Jumping to U-Boot(0x00200000)

U-Boot 2017.09 (Oct 10 2024 - 15:43:54 +0530)

Model: Rockchip RV1106 EVB Board
PreSerial: 2, raw, 0xff4c0000
DRAM:  256 MiB
Sysmem: init
Relocation Offset: 0fd85000
Relocation fdt: 0edfacb0 - 0edfede0
CR: M/C/I
Using default environment

mmc@ffa90000: 0, mmc@ffaa0000: 1
ENVF: Primary 0x00000000 - 0x00008000
ENVF: OK
Bootdev(atags): mmc 0
MMC0: HS200, 15Mhz
PartType: ENV
Hotkey: ctrl+s
DM: v2
No misc partition
boot mode: None
FIT: no signed, no conf required
DTB: rk-kernel.dtb
HASH(c): OK
Model: Luckfox Pico Ultra
CLK: (sync kernel. arm: enter 816000 KHz, init 816000 KHz, kernel 0N/A)
  apll 816000 KHz
  dpll 924000 KHz
  gpll 1188000 KHz
  cpll 1000000 KHz
  aclk_peri_root 400000 KHz
  hclK_peri_root 200000 KHz
  pclk_peri_root 100000 KHz
  aclk_bus_root 300000 KHz
  pclk_top_root 100000 KHz
  pclk_pmu_root 100000 KHz
  hclk_pmu_root 200000 KHz
Net:   eth0: ethernet@ffa80000
Hit key to stop autoboot('CTRL+C'):  0
## Booting FIT Image at 0xe907680 with size 0x002f2600
Fdt Ramdisk skip relocation
No misc partition
Sysmem Warn: kernel 'reserved-memory' "rkisp@860000"(0x00860000 - 0x012ec000) is overlap with "FDT" (0x00c00000 - 0x00c0a204)
Sysmem Warn: kernel 'reserved-memory' "rtos@40000"(0x00040000 - 0x00080000) is overlap with "KERNEL" (0x00008000 - 0x002efc04)
Sysmem Warn: kernel 'reserved-memory' "mmc@3f000"(0x0003f000 - 0x00040000) is overlap with "KERNEL" (0x00008000 - 0x002efc04)
Sysmem Warn: kernel 'reserved-memory' "mmc@100000"(0x00100000 - 0x00200000) is overlap with "KERNEL" (0x00008000 - 0x002efc04)

sysmem_dump_all:
    --------------------------------------------------------------------
    memory.rgn[0].addr     = 0x00000000 - 0x10000000 (size: 0x10000000)

    memory.total           = 0x10000000 (256 MiB. 0 KiB)
    --------------------------------------------------------------------
    allocated.rgn[0].name  = "UBOOT"
                    .addr  = 0x0edfac90 - 0x10000000 (size: 0x01205370)
    allocated.rgn[1].name  = "STACK"
                    .addr  = 0x0ebfac90 - 0x0edfac90 (size: 0x00200000)
    allocated.rgn[2].name  = "FIT"
                    .addr  = 0x0e907680 - 0x0ebf9c84 (size: 0x002f2604)
    allocated.rgn[3].name  = "FDT"
                    .addr  = 0x00c00000 - 0x00c0a204 (size: 0x0000a204)
    allocated.rgn[4].name  = "KERNEL"
                    .addr  = 0x00008000 - 0x002efc04 (size: 0x002e7c04)

    kmem-resv.rgn[0].name  = "rtos@40000"
                    .addr  = 0x00040000 - 0x00080000 (size: 0x00040000)
    kmem-resv.rgn[1].name  = "meta@800000"
                    .addr  = 0x00800000 - 0x00860000 (size: 0x00060000)
    kmem-resv.rgn[2].name  = "rkisp@860000"
                    .addr  = 0x00860000 - 0x012ec000 (size: 0x00a8c000)
    kmem-resv.rgn[3].name  = "ramdisk_r"
                    .addr  = 0x012ec000 - 0x01cec000 (size: 0x00a00000)
    kmem-resv.rgn[4].name  = "ramdisk_c"
                    .addr  = 0x01cec000 - 0x021ec000 (size: 0x00500000)
    kmem-resv.rgn[5].name  = "mmc@3f000"
                    .addr  = 0x0003f000 - 0x00040000 (size: 0x00001000)
    kmem-resv.rgn[6].name  = "mmc@100000"
                    .addr  = 0x00100000 - 0x00200000 (size: 0x00100000)

    framework malloc_r     =  16 MiB
    framework malloc_f     = 512 KiB

    allocated.total        = 0x0351677c (53 MiB. 89 KiB)
    --------------------------------------------------------------------
    LMB.allocated[0].addr  = 0x00008000 - 0x002efc04 (size: 0x002e7c04)
    LMB.allocated[1].addr  = 0x00c00000 - 0x00c0a204 (size: 0x0000a204)
    LMB.allocated[2].addr  = 0x0e907680 - 0x0ebf9cc0 (size: 0x002f2640)
    LMB.allocated[3].addr  = 0x0ebfac90 - 0x10000000 (size: 0x01405370)

    reserved.core.total    = 0x019e97b8 (25 MiB. 933 KiB)
    --------------------------------------------------------------------

## Loading kernel from FIT Image at 0e907680 ...
   Using 'conf' configuration
## Verified-boot: 0
   Trying 'kernel' kernel subimage
     Description:  unavailable
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x0e912080
     Data Size:    3046136 Bytes = 2.9 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x00008000
     Entry Point:  0x00008000
     Hash algo:    sha256
     Hash value:   c611a8a95f63cd61b82f8fc0abd1d4094f98a707cd0fc580a49cf10108e95ef1
   Verifying Hash Integrity ... sha256+ OK
## Loading fdt from FIT Image at 0e907680 ...
   Using 'conf' configuration
   Trying 'fdt' fdt subimage
     Description:  unavailable
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x0e907e80
     Data Size:    41451 Bytes = 40.5 KiB
     Architecture: ARM
     Load Address: 0x00c00000
     Hash algo:    sha256
     Hash value:   ab0b43fa00b05eb20f87436c0159242553cdb6c47d94324df13b59ac16e66268
   Verifying Hash Integrity ... sha256+ OK
   Loading fdt from 0x0e907680 to 0x00c00000
   Booting using the fdt blob at 0x00c00000
   Loading Kernel Image from 0x0e912080 to 0x00008000 ... OK
   kernel loaded at 0x00008000, end = 0x002efaf8
  'reserved-memory' rtos@40000: addr=40000 size=40000
  'reserved-memory' meta@800000: addr=800000 size=60000
  'reserved-memory' rkisp@860000: addr=860000 size=a8c000
  'reserved-memory' ramdisk_r: addr=12ec000 size=a00000
  'reserved-memory' ramdisk_c: addr=1cec000 size=500000
  'reserved-memory' mmc@3f000: addr=3f000 size=1000
  'reserved-memory' mmc@100000: addr=100000 size=100000
   Using Device Tree in place at 00c00000, end 00c0d1ea
Adding bank: 0x00000000 - 0x10000000 (size: 0x10000000)
Total: 1183.229 ms

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.110 (ml@ml-ts) (arm-rockchip830-linux-uclibcgnueabihf-gcc (crosstool-NG 1.24.0) 8.3.0, GNU ld (crosstool-NG 1.24.0) 2.32) #1 Thu Oct 10 12:46:04 IST 2024
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5

```

luckfox-eng33, 2024-10-12:
What is your EMMC model number?
At the moment it looks like there is a problem with your kernel or EMMC is not supported

HTerminal, 2024-10-12:
Hello

part number is 

KLMBG2JETD-B041

luckfox-eng33, 2024-10-14:
This emmc may support it, we don't have this emmc, so you need to test it yourself
Is the firmware you are burning compiled by you? We recommend burning our firmware first to test it.