LuckfoxTECH/luckfox-pico #176: SPI NAND flash is programmed, but the board still fails to boot.

smzahraee, 2024-10-12:
Hi
I used the following command line to program the board:
`./tools/linux/Linux_Upgrade_Tool/upgrade_tool uf ../Luckfox_Pico_Pro_Max_Flash-20241012T074752Z-001/Luckfox_Pico_Pro_Max_Flash/update.img`

and the output is:

```
Using /mnt/rockchip/rv1106/luckfox-pico/tools/linux/Linux_Upgrade_Tool/config.ini
Loading firmware...
Support Type:1106	FW Ver:0.0.00	FW Time:2024-07-10 14:43:58
Loader ver:1.01	Loader Time:2024-07-10 14:42:28
Start to upgrade firmware...
Download Boot Start
Download Boot Success
Wait For Maskrom Start
Wait For Maskrom Success
Test Device Start
Test Device Success
Check Chip Start
Check Chip Success
Get FlashInfo Start
Get FlashInfo Success
Prepare IDB Start
Prepare IDB Success
Download IDB Start
Download IDB Success
Download Firmware Start
Download Image... (100%)
Download Firmware Success
Upgrade firmware ok.
```

but after flashing nothing boot
```
RKUARTDDR V1.10 ec2fae0c96 wesley.yao 22/11/15-10:58:09
S5P1
f967
rgef1
DDRConf2
DDR3, BW=16 Col=10 Bk=8 CS0 Row=14 CS=1 Die BW=16 Size=256MB
924MHz
DDR bin out

DDR V1.10 ec2fae0c96 wesley.yao 22/11/15-10:58:09
S5P1
f967
F
DDRConf2
DDR3, BW=16 Col=10 Bk=8 CS0 Row=14 CS=1 Die BW=16 Size=256MB
924MHz
DDR bin out

RKUART
```

luckfox-eng33, 2024-10-14:
With the log messages, we learn that uboot is not booting properly.
First, let's make sure that the download is slow or fast.
If it's fast, there may be a problem with flash.
If it's slow, it could be a firmware issue.
Did you compile your firmware yourself? Try using the firmware we provide first

smzahraee, 2024-10-14:
The firmware upload is slow. 
I tried both versions: one compiled with my firmware and the other with your firmware.

luckfox-eng33, 2024-10-16:
If you burned the firmware we provided, this problem should not occur under normal circumstances, please make sure that the image you burned is the corresponding hardware

smzahraee, 2024-10-19:
Before proceeding with burning the image to the SPI NAND flash, please perform the following steps:

1. Run this command on your device to zero out the target memory:
   ```bash
   dd if=/dev/zero of=/dev/mtd0 bs=1k
   ```

2. After that, proceed with burning the image onto the SPI NAND flash.

Let me know if you need further adjustments!