LuckfoxTECH/luckfox-pico #177: linux upgrade_tool not working?!

pottendo, 2024-10-13:
Hello,
I'm new to the Luckfox pico world, so I may have overlooked some documents maybe fixing my issue.
My target setup:
- Lickfox Pico Max
- SC3336 3MP Camera

My host setup:
- Arch linux

I got the Pico Max boooting with its default image from the NAND flash. I can log in via `ssh -l root@192....`
```
# uname -a
Linux luckfox 5.10.110 #1 Tue Nov 14 18:06:04 CST 2023 armv7l GNU/Linux
```
Trying to use `upgrade_tool' to retrieve some information fails:
```
(...)/luckfox-pico/tools/linux/Linux_Upgrade_Tool(main)$ sudo ./upgrade_tool -v
Using /opt/work/src/luckfox/luckfox-pico/tools/linux/Linux_Upgrade_Tool/config.ini
Upgrade Tool v2.17
(...)/luckfox-pico/tools/linux/Linux_Upgrade_Tool(main)$ lsusb
[...]
Bus 001 Device 043: ID 2207:110c Fuzhou Rockchip Electronics Company 
[...]
(...)/luckfox-pico/tools/linux/Linux_Upgrade_Tool(main)$ sudo ./upgrade_tool LD
Using /opt/work/src/luckfox/luckfox-pico/tools/linux/Linux_Upgrade_Tool/config.ini
List of rockusb connected(1)
DevNo=1	Vid=0x2207,Pid=0x110c,LocationID=11	Mode=Maskrom	SerialNo=
(...)/luckfox-pico/tools/linux/Linux_Upgrade_Tool(main)$ sudo ./upgrade_tool RSN
Using /opt/work/src/luckfox/luckfox-pico/tools/linux/Linux_Upgrade_Tool/config.ini
ERROR:failed to read sn,err=-3!
(...)/luckfox-pico/tools/linux/Linux_Upgrade_Tool(main)$ sudo ./upgrade_tool PL
Using /opt/work/src/luckfox/luckfox-pico/tools/linux/Linux_Upgrade_Tool/config.ini
Reading partition info from device failed!
(...)/luckfox-pico/tools/linux/Linux_Upgrade_Tool(main)$ 
```

So I didn't want to try to flash an `update.img` because it may render the device (at least) non-bootable.

**Any ideas what I can try to program the NAND flash?**

Using [this](https://gist.github.com/Spiritdude/da36d2cf064e49094c870e0a8b9f972f) script (thanx @Spiritdude) I could boot a more recent version from a SD Card:
```
[root@luckfox bin]# uname -a
Linux luckfox 5.10.110 #1 Wed Jul 10 13:57:24 CST 2024 armv7l GNU/Linux
```
A more general question:

Using `https://drive.google.com/drive/folders/169oiWZ6LlDxpu2Rnq3CVlCpBA0iY3ZQ-` and here `update.img` (Luckfox_pico_Pro_Max_Flash) is supposed to be the one to be flashed on the NAND flash using `sudo upgrade_tool uf update.img` (as described [here](https://wiki.luckfox.com/Luckfox-Pico/Linux-MacOS-Burn-Image))
**This would be the equivalent to using the script above to write a bootable SDCard. Correct?**

thanks, pottendo

PS: some more information: 
```
$ cat /opt/work/src/luckfox/luckfox-pico/tools/linux/Linux_Upgrade_Tool/config.ini
#lba_parity=
#rb_check_off=
#nor_single_idb=
#log_off=
```
`upgrade_tool`'s log shows:
```
17:27:25 	Upgrade Tool v2.17
17:27:25 	Current process in /opt/work/src/luckfox/luckfox-pico/tools/linux/Linux_Upgrade_Tool
17:27:25 	Using /opt/work/src/luckfox/luckfox-pico/tools/linux/Linux_Upgrade_Tool/config.ini
17:27:25 	Command Line: /opt/work/src/luckfox/luckfox-pico/tools/linux/Linux_Upgrade_Tool/upgrade_tool RSN
17:27:25 	Current Device Location ID:11
17:27:30 	Error:RKU_Write failed,err=-7
17:27:30 	Error:read_vendor_sn->Fail to read sn,err=-3
17:27:30 	Command mode out
17:29:27 	Upgrade Tool v2.17
17:29:27 	Current process in /opt/work/src/luckfox/luckfox-pico/tools/linux/Linux_Upgrade_Tool
17:29:27 	Using /opt/work/src/luckfox/luckfox-pico/tools/linux/Linux_Upgrade_Tool/config.ini
17:29:27 	Command Line: /opt/work/src/luckfox/luckfox-pico/tools/linux/Linux_Upgrade_Tool/upgrade_tool PL
17:29:27 	Current Device Location ID:11
17:29:32 	Error:RKU_Write failed,err=-7
17:29:32 	Error:get_parameter_device->Fail to readlba on checking gpt or parameter,err=-3
17:29:32 	Error:partition_list->Fail to read partition info from device
17:29:32 	Command mode out
```

pottendo, 2024-10-14:
Thanks for the feedback,
Unfortunately I've recently changed my dev-setup from Ubuntu to Arch-Linux for various reasons. So I won't go back.
I hope that the community has some ideas how to make it working also on an Archi-Linux development host.

Again, for me to clarify: 
My attempts to read e.g. the serial number (`sudo ./upgrade_tool RSN`) is supposed to work, right?
If not even this works, there's little chance that the flashing works. Is this assumption correct? As written above I've **not tried** to flash, because it may kill the existing image on the NAND flash, making the device un-usable.

Can you confirm that functions from the upgrade_tool _RSN, PL_, etc.  work on Ubuntu?

Last question:
As I'm successfully booting from an SD card (pre-built image from your google drive _and_ my self-built buildroot image). Is there a way to access the NAND flash, when booted from the SD card? I didn't find the necessary block device to mount the flash... (well, not _yet_! ;-) )

thanks, pottendo

pottendo, 2024-10-15:
hi,
sorry to be impatient: can someone here (@luckfox-eng33 ?) confirm that the _upgrade_tool_ succeeds with commands like
  $ sudo upgrade_tool RSN
or
  $ sudo upgrade_tool PL
on Ubuntu Linux, when the default image is still flashed on the NAND flash?
thanks, pottendo

luckfox-eng33, 2024-10-16:
The upgrade tool is usually maintained directly by the chip manufacturer, and ours is provided by ROCKCHIP, but its porting to other platforms we can't predict what has happened to it.
We usually directly burned image, and RSN and PL may not work with RV1106.
Since you haven't tried burning an image, why not give it a try?

luckfox-eng33, 2024-10-16:
1. 我们测试RSN和PL在RV1106上无法使用
2. 设备名应该为mtd*，你需要使用mtd-utils进行设置

pottendo, 2024-10-17:
Hi @luckfox-eng33,
thanks for the inputs - I now flashed successfully, the device boots and I could connect via ssh.
```
[root@luckfox etc]# cat /proc/version 
Linux version 5.10.110 (hxj@luckfox-System-Product-Name) (arm-rockchip830-linux-uclibcgnueabihf-gcc (crosstool-NG 1.24.0) 8.3.0, GNU ld (crosstool-NG 1.24.0) 2.32) #1 Wed Jul 10 13:57:24 CST 2024
```
Sorry for the noise, you may add a respective note that certain commands of _upgrade_tool_ may not work on the RV1106 chip.
bye, pottendo