LuckfoxTECH/luckfox-pico #179: unable to run rtsp_opencv_capture when using startup script

fredsco, 2024-10-14:
I have created a startup script following the documentation : https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-Autostart/

When I run luckfox_pico_rtsp_opencv_capture : (a.k.a) when my S99myservices runs on startup it outputs:

`/oem/luckfox_pico_rtsp_opencv_capture_demo/luckfox_pico_rtsp_opencv_capture: can't load library 'librockiva.so'
`
running /oem/luckfox_pico_rtsp_opencv_capture_demo/luckfox_pico_rtsp_opencv_capture after boot works fine.. I assume some things are not loaded. What would be the best place to run luckfox_pico_rtsp_opencv_capture on boot..

I tried to change orders of other services but without luck.

My init.d is as follows:

```
S01seedrng            S41dhcpcd             S90rtcinit
S01syslogd            S50sshd               S90usb0config
S02klogd              S50telnet             S99myservices
S02sysctl             S50usbdevice          rcK
S10udev               S80dnsmasq            rcS
S20linkmount          S90_auto_reboot       tbd_S49ntp
S20urandom            S90hciinit            tbd_S91smb
S21appinit            S90luckfoxconfigload
S40network            S90python
```

luckfox-eng33, 2024-10-16:
I suspect you didn't turn off RKIPC.

fredsco, 2024-10-17:
@luckfox-eng33 the problem was that 

```
export LD_LIBRARY_PATH=/oem/usr/lib:$LD_LIBRARY_PATH

```

was not in my script.

closed 2024-10-17 by fredsco