LuckfoxTECH/luckfox-pico #182: How to use rndis on 4G module,such as air780e ?

starry-m, 2024-10-23:
I want to use the 4G module's RNDIS to access the internet.

luckfox-eng33, 2024-10-24:
Open the corresponding kernel module and I think it should be available

starry-m, 2024-10-27:
Thank you for your reply. I have opened all RNDIS related kernels, and changed usb to host in the device tree. However, I inserted 4g module (), but rndis access is still not displayed. This 4g module can recognize rndis on other linux
> Open the corresponding kernel module and I think it should be available

vjoco, 2024-10-28:
I used usb 4G Huawei sticks with alpine, there were some udev rules required to see it as RNDIS device

starry-m, 2024-10-29:
Hi, if you could share your changes？
Thank you
>  I used usb 4G Huawei sticks with alpine, there were some udev rules required to see it as RNDIS device

vjoco, 2024-10-29:
Not sure if this answers your question: 
On alpine I have installed (apk add):
eudev-3.2.14-r0 armv7 {eudev} (GPL-2.0-or-later) [installed]
eudev-libs-3.2.14-r0 armv7 {eudev} (GPL-2.0-or-later) [installed]
eudev-openrc-3.2.14-r0 armv7 {eudev} (GPL-2.0-or-later) [installed]

Then created a udev rule for the modem (same as ubuntu, but adding restart networking at the end)

```sh
cat /etc/udev/rules.d/40-huawei.rules

# This is part of USB_ModeSwitch version 1.x.x
#
ACTION!="add", GOTO="modeswitch_rules_end"
SUBSYSTEM!="usb", GOTO="modeswitch_rules_end"

# All known install partitions are on interface 0
ATTRS{bInterfaceNumber}!="00", GOTO="modeswitch_rules_end"

# only storage class devices are handled; negative
# filtering here would exclude some quirky devices
ATTRS{bDeviceClass}=="e0", GOTO="modeswitch_rules_begin"
ATTRS{bInterfaceClass}=="e0", GOTO="modeswitch_rules_begin"
GOTO="modeswitch_rules_end"

LABEL="modeswitch_rules_begin"
# Huawei E3372-325
ATTRS{idVendor}=="3566", ATTRS{idProduct}=="2001", RUN+="/usr/sbin/usb_modeswitch -v 3566 -p 2001 -W -R -w 400"
ATTRS{idVendor}=="3566", ATTRS{idProduct}=="2001", RUN+="/usr/sbin/usb_modeswitch -v 3566 -p 2001 -W -R"

ATTRS{idVendor}=="3566", ATTRS{idProduct}=="2001", RUN+="/sbin/rc-service networking restart"

LABEL="modeswitch_rules_end"

```

vjoco, 2024-10-29:
of course you need also usb_modeswitch. If you are in a buildroot environment, I'm not sure how complex is to make it.

starry-m, 2024-10-29:
Thanks for your answer, even though I used builroot, you gave me a new idea, I will try it!