LuckfoxTECH/luckfox-pico #183: Yolo5 demo not working anymroe - Regression?

pottendo, 2024-10-26:
hi,

My HW: luckfox pico max + SC3336 camera

A few days ago (Oct. 13'th) I cloned the luckfox-pic SDK and built both for SDCard and NAND flash - images boot.
I further cloned some demos: e.g.
```
# yolov5 object detection
git clone https://github.com/luckfox-eng29/luckfox_pico_rtsp_yolov5.git  
```
following the instructions to run `/oem/usr/bin/RkLunch-stop.sh` and then running the demo works as expected.

Yesterday I forked the SDK repo, cloned my fork and using an Ubuntu docker the build went again flawlessly and boots from the NAND flash (https://github.com/pottendo/pottendo-luckfox-pico/tree/main/project)

I can stream to my host PC after start
But when using  `/oem/usr/bin/RkLunch-stop.sh` and then again **launching the Yolo5 demo some errors appear and the demo isn't working**.

Here the console 
```
[root@luckfox demos]# [ 1153.902457] venc_release 51
[ 1185.950803] stream_cif_mipi_id0: s_power 1, entity use_count 1
[ 1185.951274] stream_cif_mipi_id0: s_power 0, entity use_count 0
[ 1186.100289] stream_cif_mipi_id0: s_power 1, entity use_count 1
[ 1186.101199] stream_cif_mipi_id0: open video, entity use_countt 2
[ 1186.101305] stream_cif_mipi_id1: open video, entity use_countt 1
[ 1186.109158] rkisp rkisp-vir0: first params buf queue
[ 1186.120776] vsys dev open 27
[ 1186.129883] stream_cif_mipi_id0: open video, entity use_countt 3
[ 1186.130047] stream_cif_mipi_id0: close video, entity use_count 2
[ 1186.131333] stream_cif_mipi_id0: open video, entity use_countt 3
[ 1186.133625] stream_cif_mipi_id0: close video, entity use_count 2
[ 1186.141804] stream_cif_mipi_id0: open video, entity use_countt 3
[ 1186.141889] stream_cif_mipi_id0: close video, entity use_count 2
[ 1186.154480] stream_cif_mipi_id0: open video, entity use_countt 3
[ 1186.154568] stream_cif_mipi_id0: close video, entity use_count 2
[ 1186.158278] rkisp queue buf done
[ 1186.159223] rkisp_hw ffa00000.rkisp: set isp clk = 198000000Hz
[ 1186.161161] rkcif-mipi-lvds: stream[0] start streaming
[ 1186.161287] rockchip-mipi-csi2 mipi0-csi2: stream on, src_sd: 46747aed, sd_name:rockchip-csi2-dphy0
[ 1186.161301] rockchip-mipi-csi2 mipi0-csi2: stream ON
[ 1186.161354] rockchip-csi2-dphy0: dphy0, data_rate_mbps 506
[ 1186.161388] rockchip-csi2-dphy csi2-dphy0: csi2_dphy_s_stream stream on:1, dphy0, ret 0
[ 1186.189431] mpp_vcodec: 45: num_chan = 0
[ 1186.189536] mpp_vcodec: 387: attr->max_width = 720, attr->max_height = 480
[ 1186.191163] mpp_vcodec: 103: create channel 0 handle 73df573b online 0
[ 1186.193357] 836: MPP_ENC_SET_CFG in
[ 1186.193388] 532: MPP_ENC_SET_RC_CFG bps 3888000 [3645000 : 4131000] fps [30:30] gop 60
[ 1186.193403] 335: MPP_ENC_SET_PREP_CFG w:h [720:480] stride [2160:480]
[ 1186.193418] 270: send header for set cfg change input/format
[ 1186.193576] 836: MPP_ENC_SET_CFG in
[ 1186.193592] 532: MPP_ENC_SET_RC_CFG bps 3072000 [2880000 : 3264000] fps [30:30] gop 1
[ 1186.193602] 270: send header for set cfg change rc_mode/fps/gop
[ 1186.216930] rga_mm: failed to calculating buffer size!
[ 1186.216955] rga_common: memory param: w = 0, h = 0, f = YCbCr420SP(0xa), size = 0
[ 1186.216963] rga_mm: rga_mm_map_buffer map dma_buf error!
[ 1186.217100] rga_mm: This handle[-22] is illegal.
[ 1186.217135] rga_mm: Image size cal error! width = 0, height = 0, format = YCbCr420SP
[ 1186.217144] rga_mm: Can't get src buffer info from handle!
[ 1186.217151] rga_mm: failed to get buffer from handle
[ 1186.217159] rga_job: rga_job_commit: failed to map job info
[ 1186.217178] rga_job: failed to commit job!
[ 1186.217188] rga: rga_mpi_commit, commit mpi job failed
```

and here the launch for the demo app:
```
[root@luckfox yolo5_demo]# ./luckfox_rtsp_yolov5 
  index=0, name=images, n_dims=4, dims=[1, 640, 640, 3], n_elems=1228800, size=1228800, fmt=NHWC, type=INT8, qnt_type=AFFINE, zp=-128, scale=0.003922
  index=0, name=output0, n_dims=4, dims=[1, 80, 80, 255], n_elems=1632000, size=1632000, fmt=NHWC, type=INT8, qnt_type=AFFINE, zp=-128, scale=0.003922
  index=1, name=286, n_dims=4, dims=[1, 40, 40, 255], n_elems=408000, size=408000, fmt=NHWC, type=INT8, qnt_type=AFFINE, zp=-128, scale=0.003922
[... 154 lines cut ...]
                [3] RV1103_Luckfox_Pico_Plus
                [4] RV1106_Luckfox_Pico_Pro_Max
                [5] RV1106_Luckfox_Pico_Ultra
                [6] RV1106_Luckfox_Pico_Ultra_W
                [7] custom
Which would you like? [0~7][default:0]: 4
  Lunch menu...pick the boot medium:
  选择启动媒介:
                [0] SD_CARD
                [1] SPI_NAND
Which would you like? [0~1][default:0]: 1
  Lunch menu...pick the system version:
  选择系统版本:
                [0] Buildroot(Support Rockchip official features) 
Which would you like? [0~1][default:0]: 0
[build.sh:info] Lunching for Default BoardConfig_IPC/BoardConfig-SPI_NAND-Buildroot-RV1106_Luckfox_Pico_Pro_Max-IPC.mk boards...
[build.sh:info] Running build_select_board succeeded.

```

**Can someone confirm that with the most recent builds, the Yolo5 demo still works? Maybe the log information tells what's wrong.**

Running `/oem/usr/bin/RkLunch.sh` again makes the video-stream working.

thanks, pottendo

luckfox-eng33, 2024-10-28:
RGA version not updated, please re-pull https://github.com/LuckfoxTECH/luckfox_pico_rkmpi_example

pottendo, 2024-10-29:
Thanks @luckfox-eng33, 
the demo works again with this update - also on my recent builds.
happy hacking, pottendo

closed 2024-10-29 by pottendo