LuckfoxTECH/luckfox-pico #184: Kernel crash with running rgaImDemo example

smzahraee, 2024-10-27:
The running kernel crashes if I execute the following example

```
[root@luckfox root]# rgaImDemo -w6 --copy
Start selecting mode 0
im2d while .. 6 time.
Start selecting mode 80
im2d copy ..
MODE = 1
open file
rga_api version 1.9.1_[0]
[  185.151101] rga_mm: Current NONE_MMU[0] cannot su[  185.151827] 8<--- cut here ---
[  185.160491] Unable to handle kernel NULL pointer dereference at virtual address 00000014
[  185.168612] pgd = 1dca415c
[  185.171327] [00000014] *pgd=01104835, *pte=00000000, *ppte=00000000
[  185.177605] Internal error: Oops - BUG: 17 [#1] THUMB2
[  185.182743] Modules linked in: rockit(O) rknpu mpp_vcodec(O) rga3 os04a10 phy_rockchip_csi2_dphy phy_rockchip_csi2_dphy_hw video_rkisp video_rkcif rk_dvbm [last unloaded: sc3336]
[  185.198666] CPU: 0 PID: 795 Comm: rgaImDemo Tainted: G           O      5.10.110 #6
[  185.206319] Hardware name: Generic DT based system
[  185.211176] PC is at rga_mm_map_buffer+0x5fc/0x76e [rga3]
[  185.216596] LR is at rga_mm_map_buffer+0x2a9/0x76e [rga3]
[  185.221987] pc : [<af894626>]    lr : [<af8942d3>]    psr: 60000033
[  185.228245] sp : b0d9dda0  ip : b0d9c03c  fp : b05378f0
[  185.233467] r10: b1106000  r9 : 00000000  r8 : b1c0c0d0
[  185.238689] r7 : b1fc6a10  r6 : b1c5db40  r5 : 00000000  r4 : b1c5dd40
[  185.245212] r3 : 00000000  r2 : 00000000  r1 : 00000001  r0 : b157e000
[  185.251739] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA Thumb  Segment user
[  185.259045] Control: 50c53c7d  Table: 01fec059  DAC: 00000055
[  185.264787] Process rgaImDemo (pid: 795, stack limit = 0x7a6f7416)
[  185.270964] Stack: (0xb0d9dda0 to 0xb0d9e000)
[  185.275323] dda0: 00385000 00000cc0 00000000 00000001 b0d9ddfc 00000001 b157e000 00385000
[  185.283499] ddc0: a6a65008 00000008 00000000 b157e040 00000000 00000dc0 00002813 00000000
[  185.291675] dde0: b159ba00 b1c5dd40 b1c128a0 b1c128ac b05f3008 b159ba00 b1fc6a00 af898c77
[  185.299851] de00: b159ba00 af895583 b159ba00 b1fc6a00 00000000 00000000 00000000 00000000
[  185.308026] de20: b159ba00 b05f3008 b1fc6a00 af89ee48 af898c77 af88e1ad 00000001 b0d9de67
[  185.316202] de40: b071c000 00000000 00000000 b04ef773 aee45860 00000000 00000001 00000000
[  185.324376] de60: 00000002 b0d37a00 b0d37b44 00000001 00000001 b0d9dea4 b0d37b44 00000001
[  185.332552] de80: 00000000 60000013 0000001a b0b6e3d0 b0d9c000 b00294df 00000004 b0d9dea4
[  185.340727] dea0: b0d9f600 00000000 b05f3008 00000000 b0d9deb0 b0d9deb0 b0d37a00 0000001a
[  185.348903] dec0: 0000001a 00000000 00000000 b00294fb 00000000 b0d37a00 0000001a b01d6cff
[  185.357078] dee0: 00000000 00000000 00000000 00000000 00000254 0000001a b0d9f600 0000001a
[  185.365254] df00: b0d9df80 a6e636cc b0d9c000 6d24fe94 6d24fe94 b1c04e88 c0107203 aee45850
[  185.373430] df20: b2490f00 00000003 b0d9c000 b2490f00 00000000 b0080ff1 00002000 b00815db
[  185.381605] df40: b0d9f600 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  185.389779] df60: 00000000 00000000 b0d9f600 b0d9df88 b0d9f600 00000000 b0d9f600 00000000
[  185.397956] df80: 00000000 aee45850 aee45850 a6fe657c 00000036 b0008424 b0d9c000 00000036
[  185.406131] dfa0: 00000000 b0008201 aee45850 aee45850 00000003 c0107203 aee45850 00000000
[  185.414307] dfc0: aee45850 aee45850 a6fe657c 00000036 a6fe655c 0002a358 a6ffef70 00000000
[  185.422483] dfe0: a6fe6118 aee457f8 a6fc4300 a6dfd558 60000010 00000003 00000000 00000000
[  185.430720] [<af894626>] (rga_mm_map_buffer [rga3]) from [<af895583>] (rga_mm_import_buffer+0x15b/0x1e0 [rga3])
[  185.440853] [<af895583>] (rga_mm_import_buffer [rga3]) from [<af88e1ad>] (rga_ioctl+0x25d/0x464 [rga3])
[  185.450270] [<af88e1ad>] (rga_ioctl [rga3]) from [<b0080ff1>] (vfs_ioctl+0x11/0x1c)
[  185.457937] [<b0080ff1>] (vfs_ioctl) from [<b00815db>] (sys_ioctl+0x73/0x47c)
[  185.465071] [<b00815db>] (sys_ioctl) from [<b0008201>] (ret_fast_syscall+0x1/0x5a)
[  185.472629] Exception stack(0xb0d9dfa8 to 0xb0d9dff0)
[  185.477685] dfa0:                   aee45850 aee45850 00000003 c0107203 aee45850 00000000
[  185.485859] dfc0: aee45850 aee45850 a6fe657c 00000036 a6fe655c 0002a358 a6ffef70 00000000
[  185.494032] dfe0: a6fe6118 aee457f8 a6fc4300 a6dfd558
root
[  185.499091] Code: f8c4 9008 60e3 6862 (e9d2) 3205 
[  185.503936] ---[ end trace 4d8226234e5e70ab ]---
pport virtual address!
```

luckfox-eng33, 2024-10-28:
Please try with the latest SDK, the one you're using now is an older version!

smzahraee, 2024-10-30:
Compiling the latest kernel version results in numerous compilation errors. Please investigate and provide a fix.
your recent sdk is broken

closed 2024-10-31 by smzahraee