LuckfoxTECH/luckfox-pico #185: How to disable MAC randomization on Alpine Linux for Pico Pro?

3xtant, 2024-10-27:
I am running the Alpine Linux Luckfox image from the getting started wiki, but I cannot disable the MAC randomization and it is causing networking problems with a static IP address. DNS stops working in my intranet and the ability to connect via SSH using a local host name does not work as expected.

The only solution I have at the moment is logging into my DHCP server, changing the expected MAC address of the static assignment to match the current MAC address of the Pico Pro, which is very inconvenient. I have to do this every time I restart my Pico Pro.

I tried editing `/etc/network/interfaces` by adding a line `HWADDR ...` but networking breaks complaining that there is a duplicate assignment. I also tried `MACADDR` but that directive is ignored.

Help is appreciated

3xtant, 2024-10-29:
@davidcoulson thanks for your reply. I had not, but I installed it to check it out and it does not help my situation. The mac changes (`ifconfig` or `macchanger`) do not survive a reboot.
I had tried using `ifconfig hw ether <mac address>` but the problem is that I can only do that from the local startup script, which executes after `eth0` has gone up and requested an IP address. Using `macchanger` would have the same result.
Even restarting `networking`, DNS is broken after a mac change. It was interesting, though, to see that `macchanger` displays both the physical and the current mac addresses. They are identical (and random) in my device.

I am looking in the Luckfox wiki at the directions to build a kernel. It might be my only option, but I need to understand how the PHY driver works and there are no instructions anywhere. I need to enable a GPIO pin and change UART config settings, so I will be building a new kernel either way.

davidcoulson, 2024-10-29:
This worked for me:

https://www.reddit.com/r/Ubuntu/s/f1GxYjYsNt

davidcoulson, 2024-10-29:
Here is instructions how to use systemd to run macchanger at boot: https://www.linuxuprising.com/2018/05/how-to-permanently-change-mac-address.html

3xtant, 2024-10-29:
Thank you! I'll give it a try tonight.

3xtant, 2024-11-04:
@davidcoulson thanks again for your help, but I could not fix the issue following your instructions. Alpine Linux does not have `systemd` and the earliest I could run my script was using startup local scripts. These scripts run after the interface has already been up. I could take it down, change the MAC address and bring it back up, but it would not have internet and other devices could not find it by name, only static IP address.
These guides only got me part of the way there:
https://wiki.alpinelinux.org/wiki/Configure_Networking
https://unix.stackexchange.com/questions/669813/alpine-linux-startup-script

Today, though, I was able to get it working by switching from the default networking package to NetworkManager. The latter manages the interface from boot time and can create a cloned MAC address that persists through reboots. I followed these steps:
1. I created a random private MAC address using [a script](https://serverfault.com/a/40720) and saved the address for later use.
2. installed NetworkManager and disabled Networking. I also added NetworkManager as a default service after removing Networking.
3. Removed `eth0` from `/etc/networking/interfaces`
4. Created a config file under `/etc/NetworkManager/system-connections` using hints from:
   * https://docs.redhat.com/en/documentation/red_hat_enterprise_linux/8/html/configuring_and_managing_networking/assembly_networkmanager-connection-profiles-in-keyfile-format_configuring-and-managing-networking#con_the-keyfile-format-of-networkmanager-profiles_assembly_networkmanager-connection-profiles-in-keyfile-format
   *  https://michlstechblog.info/blog/linux-set-a-static-fixed-ip-with-network-manager-cli/
 5. I finally added a `NetworkManager.conf` file following [this format](https://www.networkmanager.dev/docs/api/latest/NetworkManager.conf.html) using the `ethernet.cloned-mac-address` directive with the address I had created in step 1.

closed 2024-11-04 by 3xtant

davidcoulson, 2024-11-04:
Have you tried using this in /etc/network/interfaces - Not sure if Alpine works like Deiban.

pre-up ifconfig <interface> hw ether xx:xx:xx:yy:yy:yy

3xtant, 2024-11-05:
Yes, I had tried that, but it actually broke networking. Very strange. NetworkManager is very stable and has been working reliably since I set it up. Reboots are no longer an issue for me