LuckfoxTECH/luckfox-pico #189: Update `open-cv-mobile` to support `cv::imshow("fb", bgr)` for direct framebuffer rendering

pottendo, 2024-10-31:
hello,

I got an ili9341 display working nicely with a little bit of tweaking on
my HW: luckfox pico max + SC3336 camera
A little demo shows nice and bright colors:
  
[image]

I also got the yolo5 and luckfox_pico_retinaface_facenet_demo running and showing the live camera stream on the display.
The capture camera pictures are shifted to blue quite intensively - even if cover the lense I get a blue screen instead of black. Not yet understood, what can be the reason for this (see [here](https://photos.app.goo.gl/T9ss6XpvDXiCyJGB7)). This display uses a 16bit color format per pixel BGR565.

The `luckfox_pico_retinaface_facenet_demo` converts images and memcpys to the framebuffer: 
```
            //LCD Show 
            if( pixel_size == 4 ) 
                cv::cvtColor(bgr, disp, cv::COLOR_BGR2BGRA);
            else if( pixel_size == 2 )
                cv::cvtColor(bgr, disp, cv::COLOR_BGR2BGR565);
            memcpy(framebuffer, disp.data, disp_width * disp_height * pixel_size);
```

Now I found this: opencv-mobile promises [cv::imshow supports Linux framebuffer and Windows](https://github.com/nihui/opencv-mobile?tab=readme-ov-file#cvimshow-supports-linux-framebuffer-and-windows)

The version of opencv-mobile shipped with the rknn demos won't allow direct rendering to the display using `cv::imshow("fb", bgr)`. This just writes fb.png files (which actually are the cam-pictures in perfect nice colors).

I think the build of **opencv-mobile needs just to be refreshed to latest master version of opencv-mobile to enable this feature**.
The code seems to be in the master branch: [here](https://github.com/nihui/opencv-mobile/blob/master/highgui/src/display_fb.cpp)

I don't know if this is complex or not, and I didn't find an easy way to cross-compile opencv-mobile for the luckfox buildroot target. I can't build on the device itself, as it's not running gcc etc. **So I ask here if one could support this.** Writing the real-time camera display like suggested by the example linked above would be very elegant and convenient and one wouldn't be bothered with the various framebuffer layouts.

thanks, pottendo

pottendo, 2024-11-02:
A small update: I found the reason for the wrong blue'ish color shift. It's been a simple BGR vs. RGB twist.
Still the direct rendering into framebuffers would be nice. ;-)

luckfox-eng33, 2024-11-04:
I think updating opencv-mobile isn't very difficult when it comes to just replacing libs and cmake

pottendo, 2024-11-04:
Normally, one wants process the incoming frames, so the the function is probably anyway of limited use - maybe I'll have a look into it (one day). thanks anyway, pottendo

closed 2024-11-04 by pottendo

pottendo, 2024-12-06:
fyi, I built the open-cv-mobile 4.10.0 for my pico max target and the direct framebuffer rendering ("imshow("bf", ...)) and other opencv functions work as expected. Tested (very loosly) in combination with my touch screen (Ilitek 9341) - see here: https://github.com/LuckfoxTECH/luckfox-pico/pull/204
have fun, pottendo