LuckfoxTECH/luckfox-pico #190: Issue During the build of the Luckfox Pico project

oresterosso75, 2024-11-01:
Issue: During the build of the Luckfox Pico project, I encounter a copy error that halts the process. The cp command fails to copy the sshd_defconfig file to the destination ${PROJECT_PATH}/sysdrv/out/rootfs_uclibc_rv1106/etc/ssh/sshd_config, returning a "No such file or directory" error.

Details:

I verified that the path exists, but out/rootfs_uclibc_rv1106/etc/ssh is deleted during the build process.
I tried recreating the directory manually using mkdir -p and rerunning the build, but the error persists.
Even when attempting to copy the file manually, I still get the same error.
Questions:

Is the path /sysdrv/out/rootfs_uclibc_rv1106 being deleted by a specific command during the build?
Is there an alternative way to prevent this deletion, or a specific configuration to apply in the Makefile or build.sh to resolve this issue?

luckfox-eng33, 2024-11-02:
How are you?
This problem usually occurs when you modify the default kernel driver, do you have SSH turned off? otherwise this problem shouldn't occur!
If SSH is turned off, please turn off the override of SSH related files by yourself.https://github.com/LuckfoxTECH/luckfox-pico/blob/main/sysdrv/Makefile

oresterosso75, 2024-11-02:
I get it, I just want to be able to choose what to use in ./build.sh buildrootconfig.
instead of ssh I wanted to add dropbear (a lightweight SSH server)
use udhcpc to automatically assign an ip address:
remove Samba, Python, etc:
I enjoy customizing my build.
I'm noticing that if you don't use the default settings, the compilation stops:
First with SSH, now with Samba.
I assume it can't be customized?

luckfox-eng33, 2024-11-04:
You are free to customize that.
It's just to remove some of the override operations as we are thinking more about customers adding components on top of ours, and we'll address this later on.