LuckfoxTECH/luckfox-pico #206: BT656 camera driver example

smzahraee, 2024-12-07:
I am trying to write a b656 camera driver. 
Is there any working example camera driver in your camera drivers that you tested?
I wonder how to config device tree and driver configuration to enable `V4L2_MBUS_BT656`
good luck

luckfox-eng33, 2024-12-09:
Sorry, we have not tested the DVP camera driver and we are unable to guide you on how to configure the device tree and driver at this time.
But usually you can find linux drivers and tutorials for the corresponding camera or seek help from the camera vendor.

smzahraee, 2024-12-10:
No Problem.
There is Rockchip Driver Guide VI v1.0.8, In that document there is some guides on running dvp interface.
but I could not find the right kernel for that document.
could you guide me how I can find that kernel to check
Thanks

smzahraee, 2024-12-11:
I am searching for the source code corresponding to the specified kernel version mentioned in the document.
The document lists some clock names that I was unable to locate in your kernel source code.

For instance, I cannot determine which clock to use for the DVP. In a CSI camera, the clock is specified as `clocks = <&cru MCLK_REF_MIPI0>;`, but the appropriate clock for the DVP remains unclear. I came across `CLK_VICAP_M0`, but I am unsure if it is the correct choice.

denismanus, 2025-02-02:
Any updates on this?

Maobuff, 2025-04-30:
> I am searching for the source code corresponding to the specified kernel version mentioned in the document. The document lists some clock names that I was unable to locate in your kernel source [...]

you dont need to set clocks for dvp.
just enable rkcif, enable rkcif_dvp with endpoint for your camera and select pinctrl
something like this:

```
&rkcif {
   status = "okay";
};
&rkcif {
   status = "okay";
   pinctrl-names = "default";
   pinctrl-0 = <&vicap_m0>; //check in rv1106-pinctrl.dtsi
   port {
		/* Parallel bus endpoint */
		cif_para_in: endpoint {
			remote-endpoint = <&sensor_out>;
		};
	};
};
```

disable mipi0_csi2_hw and mipi1_csi2_hw if you are planing to use m1. But keep in mind that now you need to load video_rkcif manualy
```
&mipi0_csi2_hw {
   status = "disabled";
};
&mipi1_csi2_hw {
   status = "disabled";
};
```
Also make sure that you are connecting to the right pins of vicap

[image]