LuckfoxTECH/luckfox-pico #207: Unable to build: //arch/arm/mach-rockchip/fit_nodes.sh EXEC format error

Kulasekaran-148, 2024-12-09:
I'm using Ubuntu 22.04 as host. I ran the source command for sourcing the toolchain inside {SDK_PATH}/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/ directoy. Then I ran ./build.sh lunch from the root directory. Configured it for my luckfox pico ultra w board. Chose emmc. Chose Ubuntu. Then finally I ran sudo ./build.sh and the building process started. But each time it gets failed when reaching //arch/arm/mach-rockchip/fit_nodes.sh. Please help. 

On the other hand, I flashed the board with the default Ubuntu image provided by luckfox. The dmesg log shows the following error: [    0.202414] rfkill_bt: probe of wireless-bluetooth failed with error -1. I want to use the aic8800 module. But due to the above error, I'm not able to access it. When referring to the DTS file, the clk sources seems to be commented out. But anyway, even if could modify it,  I need to take care of the build issue first :)

Kulasekaran-148, 2024-12-09:
Just FYI, I'm using WSL on windows. Not sure if this info is helpful.

luckfox-eng33, 2024-12-09:
Can you provide the compilation log?
If it's WSL, please make sure you have the right version of WSL, WSL2 is working fine!

Kulasekaran-148, 2024-12-09:
./build.sh lunch
-bash: ./build.sh: No such file or directory
kulasekaran_wsl@LAPTOP-AOF8L2VJ:/LuckFox/luckfox-pico/arm-rockchip830-linux-uclibcgnueabihf$ cd ..
kulasekaran_wsl@LAPTOP-AOF8L2VJ:/LuckFox/luckfox-pico$ ./bui
ld.sh lunch
ls: cannot access 'BoardConfig*.mk': No such file or directory
You're building on Linux
  Lunch menu...pick the Luckfox Pico hardware version:
  选择 Luckfox Pico 硬件版本:
                [0] RV1103_Luckfox_Pico
                [1] RV1103_Luckfox_Pico_Mini_A
                [2] RV1103_Luckfox_Pico_Mini_B
                [3] RV1103_Luckfox_Pico_Plus
                [4] RV1103_Luckfox_Pico_WebBee
                [5] RV1106_Luckfox_Pico_Pro
                [6] RV1106_Luckfox_Pico_Max
                [7] RV1106_Luckfox_Pico_Ultra
                [8] RV1106_Luckfox_Pico_Ultra_W
                [9] custom
Which would you like? [0~9][default:0]: 8
  Lunch menu...pick the boot medium:
  选择启动媒介:
                [0] EMMC
Which would you like? [0][default:0]: 0
  Lunch menu...pick the system version:
  选择系统版本:
                [0] Buildroot(Support Rockchip official features)
                [1] Ubuntu(Support for the apt package management tool)
Which would you like? [0~1][default:0]: 1
[build.sh:info] Lunching for Default BoardConfig_IPC/BoardConfig-EMMC-Ubuntu-RV1106_Luckfox_Pico_Ultra_W-IPC.mk boards...
[build.sh:info] switching to board: /LuckFox/luckfox-pico/project/cfg/BoardConfig_IPC/BoardConfig-EMMC-Ubuntu-RV1106_Luckfox_Pico_Ultra_W-IPC.mk
[build.sh:info] Running build_select_board succeeded.
kulasekaran_wsl@LAPTOP-AOF8L2VJ:/LuckFox/luckfox-pico$ sudo
./build.sh
[sudo] password for kulasekaran_wsl:
[build.sh:info] switch to DTS: /LuckFox/luckfox-pico/sysdrv/source/kernel/arch/arm/boot/dts/rv1106g-luckfox-pico-ultra-w.dts
[build.sh:info] switch to kernel defconfig: /LuckFox/luckfox-pico/sysdrv/source/kernel/arch/arm/configs/luckfox_rv1106_linux_defconfig
[build.sh:info] The root file system used is not buildroot
[build.sh:info] /LuckFox/luckfox-pico/sysdrv/tools/board/ubuntu is not empty, skipping submodule update!
**************************************
Check [OK]: dtc --version
**************************************
Check [OK]: makeinfo --version
**************************************
Check [OK]: gperf --version
**************************************
Check [OK]: dpkg --list |grep g++-.*-multilib
**************************************
Check [OK]: dpkg --list |grep gcc-.*-multilib
**************************************
Check [OK]: make -v
GLOBAL_PARTITIONS: 0x8000@0x0(env),0x80000@0x8000(idblock),0x40000@0x88000(uboot),0x2000000@0xC8000(boot),0x20000000@0x20C8000(oem),0x10000000@0x220C8000(userdata),0x180000000@0x320C8000(rootfs)
[build.sh:info] Partition Filesystem Type Configure: rootfs@IGNORE@ext4,userdata@/userdata@ext4,oem@/oem@ext4
============================================
TARGET_ARCH=arm
TARGET_UBOOT_CONFIG=luckfox_rv1106_uboot_defconfig rk-emmc.config
TARGET_SPL_CONFIG=
TARGET_KERNEL_CONFIG=luckfox_rv1106_linux_defconfig rv1106-bt.config
TARGET_KERNEL_DTS=rv1106g-luckfox-pico-ultra-w.dts
TARGET_RECOVERY_CONFIG=
TARGET_RAMBOOT_CONFIG=
============================================
============Start building sysdrv============
============Start building uboot============
TARGET_UBOOT_CONFIG=luckfox_rv1106_uboot_defconfig rk-emmc.config
=========================================
make: Entering directory '/LuckFox/luckfox-pico/sysdrv'
 ==sysdrv== build uboot
make[1]: Entering directory '/LuckFox/luckfox-pico/sysdrv/source/uboot/u-boot'
#
[... 61 lines cut ...]
  COPY    u-boot.bin
  ALIGN   u-boot.bin
  CC      spl/common/spl/spl.o
  COPY    spl/u-boot-spl.dtb
  CC      tpl/arch/arm/mach-rockchip/tpl.o
  CC      spl/arch/arm/mach-rockchip/spl.o
  LD      tpl/arch/arm/mach-rockchip/built-in.o
  LD      spl/common/spl/built-in.o
  LD      spl/arch/arm/mach-rockchip/built-in.o
  CC      spl/lib/display_options.o
  LD      tpl/u-boot-tpl
  OBJCOPY tpl/u-boot-tpl-nodtb.bin
  COPY    tpl/u-boot-tpl.bin
  LD      spl/lib/built-in.o
  LD      spl/u-boot-spl
  OBJCOPY spl/u-boot-spl-nodtb.bin
  CAT     spl/u-boot-spl-dtb.bin
  COPY    spl/u-boot-spl.bin
  CFGCHK  u-boot.cfg
make[1]: Leaving directory '/LuckFox/luckfox-pico/sysdrv/source/uboot/u-boot'
.//arch/arm/mach-rockchip/fit_nodes.sh: line 44: /usr/local/bin/openssl: cannot execute binary file: Exec format error
make: *** [Makefile:446: uboot] Error 1
make: Leaving directory '/LuckFox/luckfox-pico/sysdrv'
[build.sh:error] Running build_uboot failed!
[build.sh:error] exit code 2 from line 748:

Kulasekaran-148, 2024-12-09:
@luckfox-eng33 I'm using WSL2 only. Just now verified. :)

luckfox-eng33, 2024-12-09:
Have you read our wiki or readme?
You need to install openssl
`sudo apt-get install -y git ssh make gcc gcc-multilib g++-multilib module-assistant expect g++ gawk texinfo libssl-dev bison flex fakeroot cmake unzip gperf autoconf device-tree-compiler libncurses5-dev pkg-config bc python-is-python3 passwd openssl openssh-server openssh-client vim file cpio rsync`

Kulasekaran-148, 2024-12-09:
```
openssh-client is already the newest version (1:8.9p1-3ubuntu0.10).
openssh-server is already the newest version (1:8.9p1-3ubuntu0.10).
```

Kulasekaran-148, 2024-12-09:
```
file
/usr/local/bin/openssl
/usr/local/bin/openssl: ELF 32-bit LSB pie executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, BuildID[sha1]=8ced8550a5f3ff2ab092e82bde905b08f9173f16, for GNU/Linux 3.2.0, not stripped
```

Kulasekaran-148, 2024-12-09:
@luckfox-eng33 No, I'm using windows machine x64. I'm using WSL2 which is running Ubuntu 22.04

```
wsl -l -v
  NAME      STATE           VERSION
* Ubuntu    Running         2
```

```
uname -a
Linux LAPTOP-AOF8L2VJ 5.15.167.4-microsoft-standard-WSL2 #1 SMP Tue Nov 5 00:21:55 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

```
uname -r
5.15.167.4-microsoft-standard-WSL2
```

```
Distributor ID: Ubuntu
Description: Ubuntu 22.04.5 LTS
Release: 22.04
Codename: jammy
```

Kulasekaran-148, 2024-12-09:
@luckfox-eng33 The main problem I'm facing is not being able to use the bluetooth in my luckfox pico ultra w board. 

```
pico@luckfox:~$ sudo hciattach -s 115200 /dev/ttyS1 any 115200 flow nosleep&
[1] 1480
pico@luckfox:~$ Device setup complete

[1]+  Done                    sudo hciattach -s 115200 /dev/ttyS1 any 115200 flow nosleep
pico@luckfox:~$
pico@luckfox:~$ sudo hciconfig hci0 up
Can't init device hci0: Connection timed out (110)
pico@luckfox:~$
```

luckfox-eng33, 2024-12-09:
> @luckfox-eng33 The main problem I'm facing is not being able to use the bluetooth in my luckfox pico ultra w board.
>
> ```
> pico@luckfox:~$ sudo hciattach -s 115200 /dev/ttyS1 any 115200 flow nosleep& [...]

This message is not registered wake-up pin reported, the actual use does not affect the

luckfox-eng33, 2024-12-09:
> @luckfox-eng33 The main problem I'm facing is not being able to use the bluetooth in my luckfox pico ultra w board.
>
> ```
> pico@luckfox:~$ sudo hciattach -s 115200 /dev/ttyS1 any 115200 flow nosleep& [...]

Typically, our mirrors should automatically turn on Bluetooth when they are turned on, without the need to manually turn it on

luckfox-eng33, 2024-12-09:
> @luckfox-eng33 No, I'm using windows machine x64. I'm using WSL2 which is running Ubuntu 22.04
>
> ```
> wsl -l -v
>   NAME      STATE           VERSION
> * Ubuntu    Running         2
> ```
>
> ```
> uname -a [...]

Can you run `openssl`? I think you have some problem with your environment.

Kulasekaran-148, 2024-12-09:
@luckfox-eng33 You are right. The problem was with `openssl` only. I removed and re-installed openssl alone and that problem is now gone. But still the build has failed with the following error. Here's the latest build log:
```
/LuckFox/luckfox-pico$ sudo ./build/sh
sudo: ./build/sh: command not found
kulasekaran_wsl@LAPTOP-AOF8L2VJ:/LuckFox/luckfox-pico$ sudo ./build.sh
[build.sh:info] switch to DTS: /LuckFox/luckfox-pico/sysdrv/source/kernel/arch/arm/boot/dts/rv1106g-luckfox-pico-ultra-w.dts
[build.sh:info] switch to kernel defconfig: /LuckFox/luckfox-pico/sysdrv/source/kernel/arch/arm/configs/luckfox_rv1106_linux_defconfig
[build.sh:info] The root file system used is not buildroot
[build.sh:info] /LuckFox/luckfox-pico/sysdrv/tools/board/ubuntu is not empty, skipping submodule update!
**************************************
Check [OK]: dtc --version
**************************************
Check [OK]: makeinfo --version
**************************************
Check [OK]: gperf --version
**************************************
Check [OK]: dpkg --list |grep g++-.*-multilib
**************************************
Check [OK]: dpkg --list |grep gcc-.*-multilib
**************************************
Check [OK]: make -v
GLOBAL_PARTITIONS: 0x8000@0x0(env),0x80000@0x8000(idblock),0x40000@0x88000(uboot),0x2000000@0xC8000(boot),0x20000000@0x20C8000(oem),0x10000000@0x220C8000(userdata),0x180000000@0x320C8000(rootfs)
[build.sh:info] Partition Filesystem Type Configure: rootfs@IGNORE@ext4,userdata@/userdata@ext4,oem@/oem@ext4
============================================
TARGET_ARCH=arm
TARGET_UBOOT_CONFIG=luckfox_rv1106_uboot_defconfig rk-emmc.config
TARGET_SPL_CONFIG=
TARGET_KERNEL_CONFIG=luckfox_rv1106_linux_defconfig rv1106-bt.config
TARGET_KERNEL_DTS=rv1106g-luckfox-pico-ultra-w.dts
TARGET_RECOVERY_CONFIG=
TARGET_RAMBOOT_CONFIG=
============================================
============Start building sysdrv============
============Start building uboot============
TARGET_UBOOT_CONFIG=luckfox_rv1106_uboot_defconfig rk-emmc.config
=========================================
make: Entering directory '/LuckFox/luckfox-pico/sysdrv'
 ==sysdrv== build uboot
make[1]: Entering directory '/LuckFox/luckfox-pico/sysdrv/source/uboot/u-boot'
#
# configuration written to .config
#
Using .config as base
Merging ./arch/../configs/rk-emmc.config
Value of CONFIG_ROCKCHIP_EMMC_IOMUX is redefined by fragment ./arch/../configs/rk-emmc.config:
Previous value: # CONFIG_ROCKCHIP_EMMC_IOMUX is not set
New value: CONFIG_ROCKCHIP_EMMC_IOMUX=y

#
# merged configuration written to .config (needs make)
#
scripts/kconfig/conf  --oldconfig Kconfig
#
# configuration written to .config
#
make[1]: Leaving directory '/LuckFox/luckfox-pico/sysdrv/source/uboot/u-boot'
pushd /LuckFox/luckfox-pico/sysdrv/source/uboot/u-boot;/LuckFox/luckfox-pico/sysdrv/source/uboot/u-boot/make.sh --spl-new CROSS_COMPILE=arm-rockchip830-linux-uclibcgnueabihf- || exit 1;popd
/LuckFox/luckfox-pico/sysdrv/source/uboot/u-boot /LuckFox/luckfox-pico/sysdrv
make[1]: Entering directory '/LuckFox/luckfox-pico/sysdrv/source/uboot/u-boot'
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config.h
  CFG     u-boot.cfg
  GEN     include/autoconf.mk.dep
  CFG     spl/u-boot.cfg
  CFG     tpl/u-boot.cfg
  GEN     spl/include/autoconf.mk
  GEN     tpl/include/autoconf.mk
  GEN     include/autoconf.mk
  CHK     include/config/uboot.release
  CHK     include/generated/timestamp_autogenerated.h
[... 105 lines cut ...]
    TO      /LuckFox/luckfox-pico/output/image

 [INSTALL]  /LuckFox/luckfox-pico/sysdrv/out/bin/board_uclibc_rv1106/uboot.debug.tar.bz2
    TO      /LuckFox/luckfox-pico/output/out/sysdrv_out/board_uclibc_rv1106

make: Leaving directory '/LuckFox/luckfox-pico/sysdrv'
[build.sh:info] Running build_uboot succeeded.
============Start building kernel============
TARGET_ARCH          =arm
TARGET_KERNEL_CONFIG =luckfox_rv1106_linux_defconfig
TARGET_KERNEL_DTS    =rv1106g-luckfox-pico-ultra-w.dts
TARGET_KERNEL_CONFIG_FRAGMENT =rv1106-bt.config
==========================================
make: Entering directory '/LuckFox/luckfox-pico/sysdrv'
 ==sysdrv== build kernel
make -C /LuckFox/luckfox-pico/sysdrv/source/kernel ARCH=arm mrproper
make[1]: Entering directory '/LuckFox/luckfox-pico/sysdrv/source/kernel'
make[2]: *** Documentation/Kbuild: Is a directory.  Stop.
make[1]: *** [Makefile:1942: _clean_Documentation] Error 2
make[1]: Leaving directory '/LuckFox/luckfox-pico/sysdrv/source/kernel'
make: *** [Makefile:485: kernel] Error 2
make: Leaving directory '/LuckFox/luckfox-pico/sysdrv'
[build.sh:error] Running build_kernel failed!
[build.sh:error] exit code 2 from line 876:
```

Kulasekaran-148, 2024-12-09:
@luckfox-eng33 I had to rename the `kbuild` directory inside **{SDK_PATH}/sysdrv/source/kernel/Documentation** in order proceed ahead with the build. Not sure why, but the build process failed as it was looking for a file `kbuild` and found a directory with the same name instead. Will have to flash this new generated image onto the board and see if that resolves the bluetooth issue that I'm facing. Thanks for your help today.

```
Make firmware OK!
------ OK ------
********rkImageMaker ver 2.2********
Generating new image, please wait...
Writing head info...
Writing boot file...
Writing firmware...
Generating MD5 data...
MD5 data generated successfully!
New image generated successfully!
[mk-update_pack.sh:info] Making -RK1106 update.img OK.
[build.sh:info] Running build_updateimg succeeded.
[build.sh:info] Running build_firmware succeeded.
[build.sh:info] Running build_all succeeded.
'uboot.debug.tar.bz2' -> '/LuckFox/luckfox-pico/IMAGE/IPC_EMMC_UBUNTU_RV1106_LUCKFOX_PICO_ULTRA_W_20241209.2110_RELEASE_TEST/DEBUG_FILES/uboot/uboot.debug.tar.bz2'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
save to /LuckFox/luckfox-pico/IMAGE/IPC_EMMC_UBUNTU_RV1106_LUCKFOX_PICO_ULTRA_W_20241209.2110_RELEASE_TEST
[build.sh:info] Running build_save succeeded.
[build.sh:info] Running build_allsave succeeded.
```

closed 2024-12-09 by Kulasekaran-148