LuckfoxTECH/luckfox-pico #208: Boot fails after flashing custom built Ubuntu image

Kulasekaran-148, 2024-12-10:
Here's the system boot log:

```
[  240.110326] reboot: Restarting system
DDR 306b9977f5 wesley.yao 23/12/21-09:28:37,fwver: v1.15
S5P1
4x
f967
F
DDRConf2
DDR3, BW=16 Col=10 Bk=8 CS0 Row=14 CS=1 Size=256MB
924MHz
DDR bin out

U-Boot SPL board init
U-Boot SPL 2017.09 (Dec 09 2024 - 19:46:04)
unknown raw ID 0 0 0
Trying to boot from MMC2
Card did not respond to voltage select!
mmc_init: -95, time 20
spl: mmc init failed with error: -95
Trying to boot from MMC1
Best phase range 79-23 (28 len)
Successfully tuned phase to 237, used 4ms
ENVF: Primary 0x00000000 - 0x00008000
ENVF: Primary 0x00000000 - 0x00008000
No misc partition
Trying fit image at 0x440 sector
## Verified-boot: 0
## Checking uboot 0x00200000 (lzma @0x00400000) ... sha256(241c59051b...) + sha256(cbf7ce689a...) + OK
## Checking fdt 0x00260998 ... sha256(9f596c5683...) + OK
Total: 360.776/387.654 ms

Jumping to U-Boot(0x00200000)

U-Boot 2017.09 (Dec 09 2024 - 19:46:04 +0530)

Model: Rockchip RV1106 EVB Board
MPIDR: 0xf00
PreSerial: 2, raw, 0xff4c0000
DRAM:  256 MiB
Sysmem: init
Relocation Offset: 0fd81000
Relocation fdt: 0edfa778 - 0edfede0
CR: M/C/I
Using default environment

mmc@ffa90000: 0, mmc@ffaa0000: 1
Best phase range 79-23 (28 len)
Successfully tuned phase to 237, used 3ms
ENVF: Primary 0x00000000 - 0x00008000
ENVF: Primary 0x00000000 - 0x00008000
Bootdev(atags): mmc 0
MMC0: HS200, 200Mhz
PartType: ENV
DM: v2
No misc partition
boot mode: normal
RESC: 'boot', blk@0x00001ebe
resource: sha256+
FIT: no signed, no conf required
DTB: rk-kernel.dtb
HASH(c): OK
Model: Luckfox Pico Ultra W
## retrieving sd_update.txt ...
Card did not respond to voltage select!
mmc_init: -95, time 20
CLK: (uboot. arm: enter 816000 KHz, init 816000 KHz, kernel 0N/A)
  apll 816000 KHz
  dpll 924000 KHz
[... 636 lines cut ...]
```

I want to use the AIC8800 module on my Luckfox Pico Ultra W board. At first, I flashed the board with the default Ubuntu image provided. in the dmesg log, I could find the following error:

```
rfkill_bt: probe of wireless-bluetooth failed with error -1
```

Unsure of what to do, I tried the steps in the Luckfox [documentation] (https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-Ultra-W-WIFI) to use Bluetooth and it results in the following:

```
pico@luckfox:~$ sudo hciattach -s 115200 /dev/ttyS1 any 115200 flow nosleep&
[1] 1480
pico@luckfox:~$ Device setup complete

[1]+  Done                    sudo hciattach -s 115200 /dev/ttyS1 any 115200 flow nosleep
pico@luckfox:~$
pico@luckfox:~$ sudo hciconfig hci0 up
Can't init device hci0: Connection timed out (110)
pico@luckfox:~$
```

So, I tried to build the image myself with some changes to the **`rv1106g-luckfox-pico-ultra-w.dts`** file present in **`{SDK_PATH}/sysdrv/source/kernel/arch/arm/boot/dts`** directory. I could the see the **cru** and **npu** nodes commented out in the file by default. I uncommented them out in a hope that, that could fix things.

Please also do refer to this [thread](https://github.com/LuckfoxTECH/luckfox-pico/issues/207), where I faced build fail issues and I had to rename a folder in-order to make the build succeed. Not sure if this is relevant. I'm completely new to this.

luckfox-eng29, 2024-12-11:
您好 根据您的日志来看ubuntu镜像出现问题应该是没有使用 sudo ./build.sh 来编译镜像，部分文件的权限错误导致ubuntu的服务启动失败。

luckfox-eng29, 2024-12-11:
您好 根据您的log来看发生错误的原因应该是编译镜像时没有使用
```
sudo ./build.sh
```
导致部分系统文件权限错误无法启动相关服务；

luckfox-eng29, 2024-12-11:
Hello, according to your logs, the reason for the failure to start the Ubuntu system may be that the 
``` 
sudo ./build.sh 
```
 is not used when compiling the image, resulting in the wrong permissions of some files and the related services cannot be started; Bluetooth boots correctly after driver upgrade command is
```
hciattach -s 1500000 /dev/ttyS1 any 1500000 flow nosleep &
```
, which will been modified in the wiki