LuckfoxTECH/luckfox-pico #209: git reset --hard destroyed your sdk environment

Q1anWan, 2024-12-10:
I have successfully built an image:
```
[mk-update_pack.sh:info] Making -RK1106 update.img OK.
[build.sh:info] Running build_updateimg succeeded.
[build.sh:info] Running build_firmware succeeded.
[build.sh:info] Running build_all succeeded.
'uboot.debug.tar.bz2' -> '/home/qianwan/sdk/luckfox-pico/IMAGE/IPC_EMMC_UBUNTU_RV1106_LUCKFOX_PICO_ULTRA_20241210.1700_RELEASE_TEST/DEBUG_FILES/uboot/uboot.debug.tar.bz2'
save to /home/qianwan/sdk/luckfox-pico/IMAGE/IPC_EMMC_UBUNTU_RV1106_LUCKFOX_PICO_ULTRA_20241210.1700_RELEASE_TEST
[build.sh:info] Running build_save succeeded.
[build.sh:info] Running build_allsave succeeded.
```

But if I run ```git reset --hard``` after that, and run ```sudo ./build```, my sdk got into crashed and report errors like that:

```
In file included from /home/qianwan/ws/sdk/luckfox-pico/sysdrv/source/kernel/drivers/soc/rockchip/iomux.c:17:
/home/qianwan/ws/sdk/luckfox-pico/sysdrv/source/kernel/drivers/soc/rockchip/../../pinctrl/pinctrl-rockchip.h:338:20: error: field 'gpio_chip' has incomplete type
  struct gpio_chip  gpio_chip;
                    ^~~~~~~~~
/home/qianwan/ws/sdk/luckfox-pico/sysdrv/source/kernel/drivers/soc/rockchip/../../pinctrl/pinctrl-rockchip.h:339:28: error: field 'grange' has incomplete type
  struct pinctrl_gpio_range grange;
                            ^~~~~~
/home/qianwan/ws/sdk/luckfox-pico/sysdrv/source/kernel/drivers/soc/rockchip/../../pinctrl/pinctrl-rockchip.h:430:24: error: field 'param' has incomplete type
  enum pin_config_param param;
                        ^~~~~
/home/qianwan/ws/sdk/luckfox-pico/sysdrv/source/kernel/drivers/soc/rockchip/../../pinctrl/pinctrl-rockchip.h:467:23: error: field 'pctl' has incomplete type
  struct pinctrl_desc  pctl;
                       ^~~~
make[6]: *** [/home/qianwan/ws/sdk/luckfox-pico/sysdrv/source/kernel/scripts/Makefile.build:273: drivers/soc/rockchip/iomux.o] Error 1
make[5]: *** [/home/qianwan/ws/sdk/luckfox-pico/sysdrv/source/kernel/scripts/Makefile.build:516: drivers/soc/rockchip] Error 2
make[4]: *** [/home/qianwan/ws/sdk/luckfox-pico/sysdrv/source/kernel/scripts/Makefile.build:516: drivers/soc] Error 2
make[4]: *** Waiting for unfinished jobs....
  AR      drivers/video/fbdev/core/built-in.a
  AR      drivers/video/fbdev/built-in.a
  AR      drivers/video/built-in.a
make[3]: *** [/home/qianwan/ws/sdk/luckfox-pico/sysdrv/source/kernel/Makefile:1935: drivers] Error 2
make[2]: *** [arch/arm/Makefile:375: rv1106g-luckfox-pico-ultra.img] Error 2
make[2]: Leaving directory '/home/qianwan/ws/sdk/luckfox-pico/sysdrv/source/objs_kernel'
make[1]: *** [Makefile:185: __sub-make] Error 2
make[1]: Leaving directory '/home/qianwan/ws/sdk/luckfox-pico/sysdrv/source/kernel'
make: *** [Makefile:489: kernel] Error 1
make: Leaving directory '/home/qianwan/ws/sdk/luckfox-pico/sysdrv'
[build.sh:error] Running build_kernel failed!
[build.sh:error] exit code 2 from line 876:
[build.sh:info]     make kernel -C ${SDK_SYSDRV_DIR} $kernel_build_options KERNEL_DTS=${RK_KERNEL_DTS} KERNEL_CFG=${RK_KERNEL_DEFCONFIG} KERNEL_CFG_FRAGMENT=${RK_KERNEL_DEFCONFIG_FRAGMENT}
```
And I have to remove the whole SDK and download it again to fix it.

Q1anWan, 2024-12-10:
Some mistake leds to a same issue in [#210](https://github.com/LuckfoxTECH/luckfox-pico/issues/210)

closed 2024-12-10 by Q1anWan

fengchunbo, 2024-12-12:
I can't use it even after updating, how should I handle it

Q1anWan, 2024-12-13:
> I can't use it even after updating, how should I handle it

Very sad, you shall run ```sudo rm -rf path/to/luckfox-pico```, and then clone it again.
It's vital to remember, never run ```git reset --hard``` anymore.