LuckfoxTECH/luckfox-pico #218: I2C-3 on Pico Plus

MxFxM, 2024-12-19:
Hi,

according to the documentation on [https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-RV1103/Luckfox-Pico-pinout/Luckfox-Pico-I2C](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-RV1103/Luckfox-Pico-pinout/Luckfox-Pico-I2C) the I2C 3 interface should be enabled on the Pico Plus.

However, when running `ls /sys/bus/i2c/devices/` as recommended, I only see i2c-4

I also tried older precompiled firmware versions using buildroot from the google drive at [https://drive.google.com/drive/folders/1_tR-Rj0fY5y_aLzWKHja7OtaYhUSg_NF](https://drive.google.com/drive/folders/1_tR-Rj0fY5y_aLzWKHja7OtaYhUSg_NF) but with no success.

When looking for the dts file as mentioned in the docs about the PWM, I cannot find the Pico Plus dts file here, so I can not check it...

Best regards, Max

luckfox-eng33, 2024-12-20:
You have an easy way to use our lukfox-config, refer to [the wiki](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-RV1103/Luckfox-Pico-Plus-Mini/Luckfox-config/) for details on how to use it!
And DTS you can find inside the entire SDK

MxFxM, 2024-12-23:
Oh, I did not see that.
That is great actually!
Maybe you can link that page on the peripherals page as well.

Thank you very much.

closed 2024-12-23 by MxFxM