LuckfoxTECH/luckfox-pico #219: How can we use jtag to debug the ARM7 chip?

bwooster0, 2024-12-29:
From uboot we can do the following:

mm.l 0xff538008

and then enter

0xff001100

This should set us up to use jtag to debug the ARM7 chip.

But now we need a target file for openocd to connect to debug the ARM7 chip.

Thanks for any help.

luyi1888, 2024-12-30:
RV1106 ARM openocd cfg

`
if { [info exists CHIPNAME] } {
  set _CHIPNAME $CHIPNAME
} else {
  set _CHIPNAME rv1106
}

#
# Main DAP
#
if { [info exists DAP_TAPID] } {
   set _DAP_TAPID $DAP_TAPID
} else {
   set _DAP_TAPID 0x2ba01477
#}

adapter speed 1000

transport select swd

reset_config srst_only srst_nogate

# declare the one SWD tap to access the DAP
swd newdap $_CHIPNAME cpu -expected-id $_DAP_TAPID -ignore-version

# create the DAP
dap create $_CHIPNAME.dap -chain-position $_CHIPNAME.cpu
target create $_CHIPNAME.apb mem_ap -dap $_CHIPNAME.dap -ap-num 0

target create $_CHIPNAME.core cortex_a -dap $_CHIPNAME.dap -dbgbase 0x80110000

`

bwooster0, 2024-12-30:
1) I got into the luckfox uboot and did "mm.l 0xff538008"

and then enter

0xff001100

2) I flashed your dirtyjtag file onto a pi pico. I then hooked it up to the luckfox pico, TMS -> uart2_rx, and TCK -> uart2_tx.

I ran openocd -f dirtyjtag.cfg -f file_with_above_config

I got the following error:

Info : only one transport option; autoselecting 'jtag'
Info : DirtyJTAG: cJTAG protocol set to 0
Can't change session's transport after the initial selection was made

luyi1888, 2024-12-31:
ARM is SWD Protocol。
You can use PicoProbe/J-Link/BlackMagic.