LuckfoxTECH/luckfox-pico #221: Opkg architecture misunderstand

B3B3K, 2025-01-05:
Device: Luckfox Pico Mini B, boot on sd card

Problem: opkg refuses install ipk file due architecture error

Detail: From mirrors, installed gcc package and ran tryed install it (opkg install gcc.ipk) but it gave this error:
`Not selecting gcc 8.4.0 due to incompatible architecture.`

Question: I m not being able to choose architecture and package source. 

Used ipk soruces:

[https://archive.openwrt.org/releases/packages-24.10/arm_cortex-a7/packages/](arm_cortex-a7)
[https://bin.entware.net/armv5sf-k3.2/](armv5sf-k3.2)
[https://bin.entware.net/armv7sf-k3.2/Packages.html](armv7sf-k3.2)

luckfox-eng33, 2025-01-10:
Are you using openwrt? Make sure you are using uclibc toolchain, we didn't test openwrt and may not be able to help much!

B3B3K, 2025-01-10:
> Are you using openwrt? Make sure you are using uclibc toolchain, we didn't test openwrt and may not be able to help much!

I have used uclibc, i compiled and installed in buildroot successfully.  Also configured it. But i have problem with installk packages. Which architecture should I use?

B3B3K, 2025-01-20:
This version of opkg works for me.

http://pkg.entware.net/binaries/armv7/