LuckfoxTECH/luckfox-pico #223: what makes it impossible to use CSI camera with ubuntu

djdisodo, 2025-01-08:
i understand the ubuntu image is unable to use CSI camera because RKMPI is not available on ubuntu

however, without RKMPI, getting raw camera access, isn't v4l2 device  created by linux kerne ?

and because kernel is independent from userspace, by using buildroot kernel with ubuntu rootfs, will that allow me to use camera?

luckfox-eng33, 2025-01-08:
You can get raw image, but ubuntu doesn't work with rkmpi and rknpu

djdisodo, 2025-01-08:
> You can get raw image, but ubuntu doesn't work with rkmpi and rknpu

can you tell me how can i get raw images from ubuntu?, i have to test my camera

also is this rkmpi? (is it worth try compiling this?)

https://github.com/rockchip-linux/mpp

luckfox-eng33, 2025-01-08:
Just use v4l2 to read video0 I think (I kinda forgot)
rk didn't open source 1106, so the repository you posted doesn't work.

djdisodo, 2025-01-08:
i noticed that one thing about the issue is that prebuilt mpp is built for ulibc, not glibc which ubuntu uses,

i'll try to run this on ubuntu, by chroot or linking ulibc statically

but before that i want to make sure my camera works

my devboard is not rv1106 but rv1103,

```
root@luckfox:~# v4l2-ctl --list-devices
rkcif-mipi-lvds (platform:rkcif):
        /dev/media0

Failed to open /dev/video0: No such device
root@luckfox:~# v4l2-ctl --list-formats /dev/video0
Failed to open /dev/video0: No such device
root@luckfox:~# v4l2-ctl --list-formats /dev/media0
Failed to open /dev/video0: No such device
root@luckfox:~# v4l2-ctl --list-formats /dev/media0
Failed to open /dev/video0: No such device
root@luckfox:~# dmesg > log

```

here's the full dmesg
```

[    3.963625] systemd[1]: Reached target Local Encrypted Volumes.
[    3.969335] systemd[1]: Reached target Remote File Systems.
[    3.975039] systemd[1]: Reached target Slice Units.
[    3.980618] systemd[1]: Reached target Local Verity Protected Volumes.
[    3.988334] systemd[1]: Listening on Syslog Socket.
[    3.993424] systemd[1]: Listening on initctl Compatibility Named Pipe.
[    4.025292] systemd[1]: Condition check resulted in Journal Audit Socket being skipped.
[    4.026500] systemd[1]: Listening on Journal Socket (/dev/log).
[    4.033185] systemd[1]: Listening on Journal Socket.
[    4.041514] systemd[1]: Listening on udev Control Socket.
[    4.049157] systemd[1]: Listening on udev Kernel Socket.
[    4.056491] systemd[1]: Condition check resulted in Huge Pages File System being skipped.
[    4.057390] systemd[1]: Condition check resulted in POSIX Message Queue File System being skipped.
[    4.062142] systemd[1]: Mounting Kernel Debug File System...
[    4.069836] systemd[1]: Condition check resulted in Kernel Trace File System being skipped.
[    4.074586] systemd[1]: systemd-journald.service: unit configures an IP firewall, but the local system does not support BPF/cgroup firewalling.
[    4.074630] systemd[1]: (This warning is only shown for the first unit using IP firewalling.)
[    4.084127] systemd[1]: Starting Journal Service...
[    4.092871] systemd[1]: Condition check resulted in Create List of Static Device Nodes being skipped.
[    4.116644] systemd[1]: Starting Load Kernel Module configfs...
[    4.162673] systemd[1]: Starting Load Kernel Module drm...
[    4.209871] systemd[1]: Starting Load Kernel Module efi_pstore...
[    4.258607] systemd[1]: Starting Load Kernel Module fuse...
[    4.294010] systemd[1]: Starting Load Kernel Modules...
[    4.324765] systemd[1]: Starting Remount Root and Kernel File Systems...
[    4.362126] systemd[1]: Starting Coldplug All udev Devices...
[    4.417765] systemd[1]: Mounted Kernel Debug File System.
[    4.433834] systemd[1]: modprobe@configfs.service: Deactivated successfully.
[    4.450894] systemd[1]: Finished Load Kernel Module configfs.
[    4.504507] systemd[1]: modprobe@drm.service: Deactivated successfully.
[    4.507081] systemd[1]: Finished Load Kernel Module drm.
[    4.523378] systemd[1]: modprobe@efi_pstore.service: Deactivated successfully.
[    4.531446] EXT4-fs (mmcblk1p7): re-mounted. Opts: (null)
[    4.536237] systemd[1]: Finished Load Kernel Module efi_pstore.
[    4.567061] systemd[1]: modprobe@fuse.service: Deactivated successfully.
[    4.574078] systemd[1]: Finished Load Kernel Module fuse.
[    4.592503] systemd[1]: Finished Load Kernel Modules.
[    4.608215] systemd[1]: Finished Remount Root and Kernel File Systems.
[    4.638978] systemd[1]: Activating swap /swapfile...
[    4.646925] systemd[1]: Condition check resulted in FUSE Control File System being skipped.
[    4.670423] systemd[1]: Mounting Kernel Configuration File System...
[    4.682084] systemd[1]: Condition check resulted in Platform Persistent Storage Archival being skipped.
[    4.715642] systemd[1]: Starting Load/Save Random Seed...
[... 172 lines cut ...]
[  134.882916] rkcif_tools_id2: update sensor info failed -19
[  134.902178] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[  134.902223] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[  134.902233] stream_cif_mipi_id0: update sensor info failed -19
[  174.675762] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[  174.675802] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[  174.675813] stream_cif_mipi_id0: update sensor info failed -19
[  178.489475] wlx90de8078226d: authenticate with 28:4e:e9:01:72:93
[  178.529963] wlx90de8078226d: send auth to 28:4e:e9:01:72:93 (try 1/3)
[  178.731370] wlx90de8078226d: send auth to 28:4e:e9:01:72:93 (try 2/3)
[  178.753821] wlx90de8078226d: authenticated
[  178.759017] wlx90de8078226d: associate with 28:4e:e9:01:72:93 (try 1/3)
[  178.961369] wlx90de8078226d: associate with 28:4e:e9:01:72:93 (try 2/3)
[  178.995344] wlx90de8078226d: RX AssocResp from 28:4e:e9:01:72:93 (capab=0x411 status=0 aid=19)
[  178.995449] wlx90de8078226d: AP has invalid WMM params (AIFSN=1 for ACI 3), will use 2
[  178.996378] usb 1-1: rtl8xxxu_bss_info_changed: HT supported
[  178.997613] wlx90de8078226d: associated
[  179.605138] wlx90de8078226d: AP has invalid WMM params (AIFSN=1 for ACI 3), will use 2
[  194.090900] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[  194.090950] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[  194.090966] stream_cif_mipi_id0: update sensor info failed -19
[  198.197066] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[  198.197106] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[  198.197117] stream_cif_mipi_id0: update sensor info failed -19
```

djdisodo, 2025-01-08:
repluged connector:
```
[    0.824693] systemd[1]: systemd 249.11-0ubuntu3.11 running in system mode (+PAM +AUDIT +SELINUX +APPARMOR +IMA +SMACK +SECCOMP +GCRYPT +GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN +IPTC +KMOD +LIBCRYPTSETUP +LIBFDISK +PCRE2 -PWQUALITY -P11KIT -QRENCODE +BZIP2 +LZ4 +XZ +ZLIB +ZSTD -XKBCOMMON +UTMP +SYSVINIT default-hierarchy=unified)
[    0.825796] systemd[1]: Detected architecture arm.
[    0.833817] systemd[1]: Hostname set to <luckfox>.
[    1.966233] systemd[1]: Queued start job for default target Graphical Interface.
[    1.972067] systemd[1]: Created slice Slice /system/getty.
[    1.976398] systemd[1]: Created slice Slice /system/modprobe.
[    1.982633] systemd[1]: Created slice Slice /system/serial-getty.
[    1.988262] systemd[1]: Created slice User and Session Slice.
[    1.994156] systemd[1]: Started Dispatch Password Requests to Console Directory Watch.
[    2.002175] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[    2.010175] systemd[1]: Condition check resulted in Arbitrary Executable File Formats File System Automount Point being skipped.
[    2.010868] systemd[1]: Reached target Local Encrypted Volumes.
[    2.016618] systemd[1]: Reached target Remote File Systems.
[    2.022306] systemd[1]: Reached target Slice Units.
[    2.027898] systemd[1]: Reached target Local Verity Protected Volumes.
[    2.035624] systemd[1]: Listening on Syslog Socket.
[    2.040725] systemd[1]: Listening on initctl Compatibility Named Pipe.
[    2.075564] systemd[1]: Condition check resulted in Journal Audit Socket being skipped.
[    2.076840] systemd[1]: Listening on Journal Socket (/dev/log).
[    2.083498] systemd[1]: Listening on Journal Socket.
[    2.091963] systemd[1]: Listening on udev Control Socket.
[    2.099514] systemd[1]: Listening on udev Kernel Socket.
[    2.106733] systemd[1]: Condition check resulted in Huge Pages File System being skipped.
[    2.107578] systemd[1]: Condition check resulted in POSIX Message Queue File System being skipped.
[    2.112466] systemd[1]: Mounting Kernel Debug File System...
[    2.120426] systemd[1]: Condition check resulted in Kernel Trace File System being skipped.
[    2.125275] systemd[1]: systemd-journald.service: unit configures an IP firewall, but the local system does not support BPF/cgroup firewalling.
[    2.125322] systemd[1]: (This warning is only shown for the first unit using IP firewalling.)
[    2.134106] systemd[1]: Starting Journal Service...
[    2.144777] systemd[1]: Condition check resulted in Create List of Static Device Nodes being skipped.
[    2.168488] systemd[1]: Starting Load Kernel Module configfs...
[    2.216005] systemd[1]: Starting Load Kernel Module drm...
[    2.265234] systemd[1]: Starting Load Kernel Module efi_pstore...
[    2.309438] systemd[1]: Starting Load Kernel Module fuse...
[    2.349182] systemd[1]: Starting Load Kernel Modules...
[    2.379269] systemd[1]: Starting Remount Root and Kernel File Systems...
[    2.417721] systemd[1]: Starting Coldplug All udev Devices...
[    2.468111] systemd[1]: Mounted Kernel Debug File System.
[    2.489273] systemd[1]: modprobe@configfs.service: Deactivated successfully.
[    2.504041] systemd[1]: Finished Load Kernel Module configfs.
[    2.562789] systemd[1]: modprobe@drm.service: Deactivated successfully.
[    2.566365] EXT4-fs (mmcblk1p7): re-mounted. Opts: (null)
[    2.571671] systemd[1]: Finished Load Kernel Module drm.
[    2.597399] systemd[1]: modprobe@efi_pstore.service: Deactivated successfully.
[    2.612475] systemd[1]: Finished Load Kernel Module efi_pstore.
[    2.632725] systemd[1]: modprobe@fuse.service: Deactivated successfully.
[    2.634129] systemd[1]: Finished Load Kernel Module fuse.
[    2.650002] systemd[1]: Finished Load Kernel Modules.
[    2.660433] systemd[1]: Finished Remount Root and Kernel File Systems.
[    2.688840] systemd[1]: Activating swap /swapfile...
[    2.697682] systemd[1]: Condition check resulted in FUSE Control File System being skipped.
[    2.725175] systemd[1]: Mounting Kernel Configuration File System...
[    2.736086] systemd[1]: Condition check resulted in Platform Persistent Storage Archival being skipped.
[    2.769284] systemd[1]: Starting Load/Save Random Seed...
[    2.773824] Adding 1048572k swap on /swapfile.  Priority:-2 extents:3 across:1196028k SS
[    2.809265] systemd[1]: Starting Apply Kernel Variables...
[    2.844272] systemd[1]: Starting Create System Users...
[    2.865315] systemd[1]: Activated swap /swapfile.
[    2.977136] systemd[1]: Started Journal Service.
[    3.182472] systemd-journald[65]: Received client request to flush runtime journal.
[    3.187718] systemd-journald[65]: File /var/log/journal/1fe5d22f36814c0598da7e315031b67d/system.journal corrupted or uncleanly shut down, renaming and replacing.
[    4.792434] rockchip-csi2-dphy csi2-dphy0: failed attach dphy0 hw
[    4.792473] rockchip-csi2-dphy csi2-dphy0: csi2 dphy hw can't be attached, register dphy0 failed!
[    4.932103] rockchip-csi2-dphy-hw ff3e8000.csi2-dphy-hw: csi2 dphy hw probe successfully!
[    5.283202] sc4336 4-0030-1: driver version: 00.01.01
[    5.283269] sc4336 4-0030-1: Failed to get reset-gpios
[    5.283319] sc4336 4-0030-1: could not get default pinstate
[    5.283350] sc4336 4-0030-1: could not get sleep pinstate
[... 141 lines cut ...]
[  100.571325] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[... cut ...]
[  100.579984] stream_cif_mipi_id0: update sensor info failed -19
[  109.433366] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[  109.433406] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[  109.433417] stream_cif_mipi_id0: update sensor info failed -19
[  115.023819] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[  115.023860] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[  115.023870] stream_cif_mipi_id0: update sensor info failed -19
```

djdisodo, 2025-01-08:
> Try using rootfs from ubuntu and other images from original img. May it be solve bcs drivers is not loaded in the rootfs.

i'll try that, however reason i can't do much on buildroot, is because i don't have ethernet port on my boart so i have to use wifi dongle, and it's hard to set up on buildroot

however i got drivers from /oem/usr/ko

still /dev/video0 is unavailable
 rkcif-mipi-lvds is not working

djdisodo, 2025-01-08:
@B3B3K
maybe it has something to do with env.img

on ubuntu `rk_dma_heap_cma=1M`

on buildroot `rk_dma_heap_cma=24M`

i'm finding a way without reflashing it, since i can't connect to uart now

djdisodo, 2025-01-08:
i could change that value using mkenv.sh, 24MB ubuntu not booting, 8MB, it booted but still not working

djdisodo, 2025-01-09:
found that it has something to do with order and how the kernel modules are loaded 

for me:

added 
```
blacklist video_rkcif
blacklist video_rkisp
blacklist phy-rockchip-csi2-dphy-hw
blacklist phy-rockchip-csi2-dphy
blacklist sc3336
blacklist sc4336
blacklist sc530i
```
to /etc/modprobe.d/blacklist.conf

ran
```
/oem/usr/ko/insmod_ko.sh
```

now drivers are "properly" loaded
and i can get raw images

djdisodo, 2025-01-09:
there's too little ram left on ubuntu, can't add for cma reserve,

i'll try alpine linux, i couldn't find major distro that supports uclibc

djdisodo, 2025-01-10:
i did make it work, tho it's on alpine linux, not ubuntu
it's just because ubuntu uses too mcuh ram

```
# copy /lib from buildroot to /uclibc
cd /oem/usr/bin
for f in * ; do patchelf --set-interpreter /uclibc/ld-uClibc.so.0 --set-rpath /uclibc/ "$f" ; done
```
huh

B3B3K, 2025-01-10:
Nich work! I have luckfox pico mini but i cant install adb to alpline linux. And missing ethernet port, keeps rndis one solution for internet (and it is avaible on just in buildroot). So keeping the way that buildroot haves,  may be easyest solution for now.

djdisodo, 2025-01-10:
https://github.com/djdisodo/luckfox-pico-alpine

closed 2025-01-10 by djdisodo