LuckfoxTECH/luckfox-pico #224: Using SDK with commit ID 40639ec to build image but "exit code 2 from line 882"

VoLinhTruc, 2025-01-09:
Firstly I'm sorry for this low quality of the console log captured pictured.

I tried to clone the newest version of the sdk but build failed.

I check that the TARGET_KERNEL_CONFIG_FRAGMENT value is empty as shown below.

[image]

I tried build using docker image and also try to build on Ubuntu 22.04 with full tool set by following this guideline
https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-SDK/

But the result is still failed.

I remember that in November, I can build pass with the same procedure.

Please support

luckfox-eng33, 2025-01-10:
Hello.
Can you provide more complete log information?
So far I see that you have a problem with the path
* `/home/luckfox-pico/sysdrv` seems to put the sdk directly into the user directory, 
* normally the sdk needs to be in a separate folder

So I think it might be a compilation environment problem!

Xmaksss, 2025-03-25:
Hey there! I have same problem :(
I used this docker image: luckfoxtech/luckfox_pico:1.0 (also tried on my own docker image but had same problem)

here is logs:
u-boot was successfully build, but with kernel and rootfs I have problems
```sh
make[1]: Leaving directory '/home/sysdrv/source/uboot/u-boot'
pack u-boot.itb okay! Input: /home/sysdrv/source/uboot/rkbin/RKTRUST/RV1106TOS.ini

FIT description: FIT Image with ATF/OP-TEE/U-Boot/MCU
Created:         Wed Mar 26 00:58:18 2025
 Image 0 (uboot)
  Description:  U-Boot
  Created:      Wed Mar 26 00:58:18 2025
  Type:         Standalone Program
  Compression:  lzma compressed
  Data Size:    181424 Bytes = 177.17 KiB = 0.17 MiB
  Architecture: ARM
  Load Address: 0x00200000
  Entry Point:  unavailable
  Hash algo:    sha256
  Hash value:   0f08751430dc0f919c579ccd7016f645bef453eb294957a83fdb3d13ddc40f8a
 Image 1 (fdt)
  Description:  U-Boot dtb
  Created:      Wed Mar 26 00:58:18 2025
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    9670 Bytes = 9.44 KiB = 0.01 MiB
  Architecture: ARM
  Hash algo:    sha256
  Hash value:   9f596c5683acf8a637b6e1a3cb1669bfc33f229b4cef5782c67070aaf0bc543e
 Default Configuration: 'conf'
 Configuration 0 (conf)
  Description:  rv1106-evb
  Kernel:       unavailable
  FDT:          fdt
  Loadables:    uboot
********boot_merger ver 1.35********
Info:Pack loader ok.
creating new idblock from loader...
idblock binary saving at rv1106_idblock_v1.15.102.img
pack loader(SPL) okay! Input: /home/sysdrv/source/uboot/rkbin/RKBOOT/RV1106MINIALL.ini

/home/sysdrv/source/uboot/u-boot
pack loader with new: spl/u-boot-spl.bin

Image(no-signed, version=0): uboot.img (FIT with uboot, trust...) is ready
Image(no-signed): rv1106_idblock_v1.15.102.img (with spl, ddr...) is ready
pack uboot.img okay! Input: /home/sysdrv/source/uboot/rkbin/RKTRUST/RV1106TOS.ini

Platform RV1106 is build OK, with exist .config
arm-rockchip830-linux-uclibcgnueabihf-
Wed Mar 26 00:58:19 HKT 2025
/home/sysdrv
'/home/sysdrv/source/uboot/u-boot/uboot.img' -> '/home/sysdrv/out/image_uclibc_rv1106/uboot.img'
'/home/sysdrv/source/uboot/u-boot/rv1106_idblock_v1.15.102.img' -> '/home/sysdrv/out/image_uclibc_rv1106/idblock.img'
'/home/sysdrv/source/uboot/u-boot/rv1106_download_v1.15.108.bin' -> '/home/sysdrv/out/image_uclibc_rv1106/download.bin'
/home/sysdrv/source/uboot/u-boot /home/sysdrv
/home/sysdrv

 [INSTALL]  /home/sysdrv/out/image_uclibc_rv1106/idblock.img /home/sysdrv/out/image_uclibc_rv1106/uboot.img /home/sysdrv/out/image_uclibc_rv1106/download.bin 
    TO 	    /home/output/image 

 [INSTALL]  /home/sysdrv/out/bin/board_uclibc_rv1106/uboot.debug.tar.bz2 
    TO 	    /home/output/out/sysdrv_out/board_uclibc_rv1106 

make: Leaving directory '/home/sysdrv'
[build.sh:info] Running build_uboot succeeded.
============Start building kernel============
TARGET_ARCH          =arm
TARGET_KERNEL_CONFIG =luckfox_rv1106_linux_defconfig
TARGET_KERNEL_DTS    =rv1103g-luckfox-pico.dts
TARGET_KERNEL_CONFIG_FRAGMENT =
==========================================
make: Entering directory '/home/sysdrv'
 ==sysdrv== build kernel  
make -C /home/sysdrv/source/kernel ARCH=arm mrproper
make[1]: Entering directory '/home/sysdrv/source/kernel'
make[2]: *** Documentation/Kbuild: Is a directory.  Stop.
make[1]: *** [Makefile:1942: _clean_Documentation] Error 2
make[1]: Leaving directory '/home/sysdrv/source/kernel'
make: *** [Makefile:485: kernel] Error 2
make: Leaving directory '/home/sysdrv'
[build.sh:error] Running build_kernel failed!
[build.sh:error] exit code 2 from line 847:
[build.sh:info]     make kernel -C ${SDK_SYSDRV_DIR} $kernel_build_options KERNEL_DTS=${RK_KERNEL_DTS} KERNEL_CFG=${RK_KERNEL_DEFCONFIG} KERNEL_CFG_FRAGMENT=${RK_KERNEL_DEFCONFIG_FRAGMENT}
```

Xmaksss, 2025-03-29:
I copied this script `sysdrv/source/kernel/scripts/Makefile.clean` from linux kernel v6 and now everything works properly

imic, 2025-08-21:
> 嘿！我:(使用这个 docker 映像时遇到了同样的问题：luckfoxtech/luckfox_pico：1.0（也尝试了我自己的 docker 映像，但遇到了同样的问题）
> 
> 这是日志：u-boot 已成功构建，但是使用 kernel 和 rootfs 时我遇到了问题
> 
> make[1]: Leaving directory '/home/sysdrv/source/uboot/u-boot'
> pack u-boot.itb okay! Input: /home/sysdrv/source/uboot/rkbin/RKTRUST/RV1106TOS.ini
> 
> FIT description: FIT Image with ATF/OP-TEE/U-Boot/MCU
> Created:         Wed Mar 26 00:58:18 2025
>  Image 0 (uboot)
>   Description:  U-Boot
>   Created:      Wed Mar 26 00:58:18 2025
>   Type:         Standalone Program
>   Compression:  lzma compressed
>   Data Size:    181424 Bytes = 177.17 KiB = 0.17 MiB
>   Architecture: ARM
>   Load Address: 0x00200000
>   Entry Point:  unavailable
>   Hash algo:    sha256
>   Hash value:   0f08751430dc0f919c579ccd7016f645bef453eb294957a83fdb3d13ddc40f8a
>  Image 1 (fdt)
>   Description:  U-Boot dtb
>   Created:      Wed Mar 26 00:58:18 2025
>   Type:         Flat Device Tree
>   Compression:  uncompressed
>   Data Size:    9670 Bytes = 9.44 KiB = 0.01 MiB
>   Architecture: ARM
>   Hash algo:    sha256
>   Hash value:   9f596c5683acf8a637b6e1a3cb1669bfc33f229b4cef5782c67070aaf0bc543e
>  Default Configuration: 'conf'
>  Configuration 0 (conf)
>   Description:  rv1106-evb
>   Kernel:       unavailable
>   FDT:          fdt
>   Loadables:    uboot
> ********boot_merger ver 1.35********
> Info:Pack loader ok.
> creating new idblock from loader...
> idblock binary saving at rv1106_idblock_v1.15.102.img
> pack loader(SPL) okay! Input: /home/sysdrv/source/uboot/rkbin/RKBOOT/RV1106MINIALL.ini
> 
> /home/sysdrv/source/uboot/u-boot
> pack loader with new: spl/u-boot-spl.bin
> 
> Image(no-signed, version=0): uboot.img (FIT with uboot, trust...) is ready
> Image(no-signed): rv1106_idblock_v1.15.102.img (with spl, ddr...) is ready
> pack uboot.img okay! Input: /home/sysdrv/source/uboot/rkbin/RKTRUST/RV1106TOS.ini
> 
> Platform RV1106 is build OK, with exist .config
> arm-rockchip830-linux-uclibcgnueabihf-
> Wed Mar 26 00:58:19 HKT 2025
> /home/sysdrv
> '/home/sysdrv/source/uboot/u-boot/uboot.img' -> '/home/sysdrv/out/image_uclibc_rv1106/uboot.img'
> '/home/sysdrv/source/uboot/u-boot/rv1106_idblock_v1.15.102.img' -> '/home/sysdrv/out/image_uclibc_rv1106/idblock.img'
> '/home/sysdrv/source/uboot/u-boot/rv1106_download_v1.15.108.bin' -> '/home/sysdrv/out/image_uclibc_rv1106/download.bin'
> /home/sysdrv/source/uboot/u-boot /home/sysdrv
> /home/sysdrv
> 
> 
>  [INSTALL]  /home/sysdrv/out/image_uclibc_rv1106/idblock.img /home/sysdrv/out/image_uclibc_rv1106/uboot.img /home/sysdrv/out/image_uclibc_rv1106/download.bin 
>     TO 	    /home/output/image 
> 
> 
> 
> 
>  [INSTALL]  /home/sysdrv/out/bin/board_uclibc_rv1106/uboot.debug.tar.bz2 
>     TO 	    /home/output/out/sysdrv_out/board_uclibc_rv1106 
> 
> 
> make: Leaving directory '/home/sysdrv'
> [build.sh:info] Running build_uboot succeeded.
> ============Start building kernel============
> TARGET_ARCH          =arm
> TARGET_KERNEL_CONFIG =luckfox_rv1106_linux_defconfig
> TARGET_KERNEL_DTS    =rv1103g-luckfox-pico.dts
> TARGET_KERNEL_CONFIG_FRAGMENT =
> ==========================================
> make: Entering directory '/home/sysdrv'
>  ==sysdrv== build kernel  
> make -C /home/sysdrv/source/kernel ARCH=arm mrproper
> make[1]: Entering directory '/home/sysdrv/source/kernel'
> make[2]: *** Documentation/Kbuild: Is a directory.  Stop.
> make[1]: *** [Makefile:1942: _clean_Documentation] Error 2
> make[1]: Leaving directory '/home/sysdrv/source/kernel'
> make: *** [Makefile:485: kernel] Error 2
> make: Leaving directory '/home/sysdrv'
> [build.sh:error] Running build_kernel failed!
> [build.sh:error] exit code 2 from line 847:
> [build.sh:info]     make kernel -C ${SDK_SYSDRV_DIR} $kernel_build_options KERNEL_DTS=${RK_KERNEL_DTS} KERNEL_CFG=${RK_KERNEL_DEFCONFIG} KERNEL_CFG_FRAGMENT=${RK_KERNEL_DEFCONFIG_FRAGMENT}
> 请🙏帮忙

What are the parameters you used to start Docker, and is volume mapping enabled?