LuckfoxTECH/luckfox-pico #226: u-boot build failed

niyazFattahov, 2025-01-11:
I was trying to build u-boot using `./build.sh uboot` command, but got many error. 
This is the last lines:
```
./tools/../common/image.c:327:16: note: previous implicit declaration of ‘genimg_get_arch_name’ with type ‘int()’
  327 |         arch = genimg_get_arch_name(image_get_arch(hdr));
      |                ^~~~~~~~~~~~~~~~~~~~
./tools/../common/image.c:716:13: error: conflicting types for ‘genimg_get_type_name’; have ‘const char *(uint8_t)’ {aka ‘const char *(unsigned char)’}
  716 | const char *genimg_get_type_name(uint8_t type)
      |             ^~~~~~~~~~~~~~~~~~~~
./tools/../common/image.c:328:16: note: previous implicit declaration of ‘genimg_get_type_name’ with type ‘int()’
  328 |         type = genimg_get_type_name(image_get_type(hdr));
      |                ^~~~~~~~~~~~~~~~~~~~
./tools/../common/image.c:721:48: error: unknown type name ‘table_entry_t’
  721 | static const char *genimg_get_short_name(const table_entry_t *table, int val)
      |                                                ^~~~~~~~~~~~~
./tools/../common/image.c: In function ‘genimg_get_short_name’:
./tools/../common/image.c:727:21: error: request for member ‘sname’ in something not a structure or union
  727 |         return table->sname;
      |                     ^~
./tools/../common/image.c: At top level:
./tools/../common/image.c:738:13: error: conflicting types for ‘genimg_get_comp_name’; have ‘const char *(uint8_t)’ {aka ‘const char *(unsigned char)’}
  738 | const char *genimg_get_comp_name(uint8_t comp)
      |             ^~~~~~~~~~~~~~~~~~~~
./tools/../common/image.c:329:16: note: previous implicit declaration of ‘genimg_get_comp_name’ with type ‘int()’
  329 |         comp = genimg_get_comp_name(image_get_comp(hdr));
      |                ^~~~~~~~~~~~~~~~~~~~
./tools/../common/image.c:773:30: error: unknown type name ‘table_entry_t’
  773 | int get_table_entry_id(const table_entry_t *table,
      |                              ^~~~~~~~~~~~~
./tools/../common/image.c: In function ‘get_table_entry_id’:
./tools/../common/image.c:776:15: error: unknown type name ‘table_entry_t’
  776 |         const table_entry_t *t;
      |               ^~~~~~~~~~~~~
./tools/../common/image.c:778:26: error: request for member ‘id’ in something not a structure or union
  778 |         for (t = table; t->id >= 0; ++t) {
      |                          ^~
./tools/../common/image.c:782:22: error: request for member ‘sname’ in something not a structure or union
  782 |                 if (t->sname && strcasecmp(t->sname, name) == 0)
      |                      ^~
./tools/../common/image.c:782:45: error: request for member ‘sname’ in something not a structure or union
  782 |                 if (t->sname && strcasecmp(t->sname, name) == 0)
      |                                             ^~
./tools/../common/image.c:784:34: error: request for member ‘id’ in something not a structure or union
  784 |                         return (t->id);
      |                                  ^~
./tools/../common/image.c: In function ‘genimg_get_cat_name’:
./tools/../common/image.c:643:1: warning: control reaches end of non-void function [-Wreturn-type]
  643 | }
      | ^
./tools/../common/image.c: In function ‘genimg_get_cat_short_name’:
./tools/../common/image.c:667:1: warning: control reaches end of non-void function [-Wreturn-type]
  667 | }
      | ^
./tools/../common/image.c: In function ‘genimg_get_cat_count’:
./tools/../common/image.c:672:1: warning: control reaches end of non-void function [-Wreturn-type]
  672 | }
      | ^
./tools/../common/image.c: In function ‘genimg_get_cat_desc’:
./tools/../common/image.c:677:1: warning: control reaches end of non-void function [-Wreturn-type]
  677 | }
      | ^
./tools/../common/image.c: In function ‘get_table_entry_name’:
./tools/../common/image.c:703:1: warning: control reaches end of non-void function [-Wreturn-type]
  703 | }
      | ^
./tools/../common/image.c: In function ‘genimg_get_short_name’:
./tools/../common/image.c:731:1: warning: control reaches end of non-void function [-Wreturn-type]
  731 | }
      | ^
At top level:
./tools/../common/image.c:186:32: warning: ‘table_info’ defined but not used [-Wunused-variable]
  186 | static const struct table_info table_info[IH_COUNT] = {
      |                                ^~~~~~~~~~
make[2]: *** [scripts/Makefile.host:116: tools/common/image.o] Error 1
make[1]: *** [Makefile:1314: tools] Error 2
make[1]: Leaving directory '/home/niyaz/e-linux/rockchip/Luckfox-Pico-Pro/luckfox-pico/sysdrv/source/uboot/u-boot'
make: *** [Makefile:446: uboot] Error 1
make: Leaving directory '/home/niyaz/e-linux/rockchip/Luckfox-Pico-Pro/luckfox-pico/sysdrv'
[build.sh:error] Running build_uboot failed!
[build.sh:error] exit code 2 from line 748:
[build.sh:info]     make uboot -C ${SDK_SYSDRV_DIR} UBOOT_CFG=${RK_UBOOT_DEFCONFIG} UBOOT_CFG_FRAGMENT=${RK_UBOOT_DEFCONFIG_FRAGMENT} SYSDRV_UBOOT_RKBIN_OVERLAY_INI=$uboot_rkbin_ini
```
kernel and roots builds without problem, I was following this guide: https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-SDK/

I changed nothing in the sources.

luckfox-eng33, 2025-01-14:
Hello,
If you have not modified uboot, it may be an environmental problem. We have not encountered this problem before.
What device and operating system are you using?

niyazFattahov, 2025-01-14:
Hello. 
I have
PC: Laptop with intel 10750
OS: Kubuntu 22.04
I'm on zsh shell, but I was also tried on bash and got same errors.
But I also have another pc with kubuntu-22.04 and uboot builds successfully on zsh shell.
The difference I see at the moment that I did not install deps in the second pc from '1.1' step.

closed 2025-01-14 by niyazFattahov