LuckfoxTECH/luckfox-pico #227: "parameter is invalid, please check!" message during uboot flushing

inisider, 2025-01-11:
Hi there,

I am trying to flash luckfox pico plus using the following command but getting an error:
```
$sudo upgrade_tool di -uboot <...>/luckfox-pico/output/image/uboot.img
Using <...>/luckfox-pico/tools/linux/Linux_Upgrade_Tool/config.ini
parameter is invalid,please check!
```

It isn't really clear which parameter is invalid and how to check that. I checked that uboot.img exists and have a proper permissions:
```
ls -la <...>/luckfox-pico/output/image/uboot.img 
-rw-r--r--. 1 root root 262144 Jan 11 17:06 <...>/luckfox-pico/output/image/uboot.img
```

Could someone provide me any pointers?

luckfox-eng33, 2025-01-13:
Why not use `./rkflash.sh uboot`?

inisider, 2025-01-13:
@luckfox-eng33: I tried that before and it does't change result:
```
 ./rkflash.sh uboot
flash uboot: /run/media/ok/blue_disk/Projects/luckfox-pico/output/image/uboot.img
Using /run/media/ok/blue_disk/Projects/luckfox-pico/tools/linux/Linux_Upgrade_Tool/config.ini
parameter is invalid,please check!
```

SL-RU, 2025-06-03:
same issue