LuckfoxTECH/luckfox-pico #229: SD CARD分区未启用（SDK Version 1.4）

jinxfzz, 2025-01-13:
RV1106_Luckfox_Pico_Max
sdcard：64G
SDK Version：1.4
BoardConfig-SD_CARD-Buildroot-RV1106_Luckfox_Pico_Max-IPC.mk
﻿
其中分区配置变更如下：export RK_PARTITION_CMD_IN_ENV="32K(env),512K@32K(idblock),256K(uboot),32M(boot),512M(oem),256M(userdata),6G(rootfs)"
→
RK_PARTITION_CMD_IN_ENV="32K(env),512K@32K(idblock),256K(uboot),32M(boot),512M(oem),256M(userdata),**56G**(rootfs)"
﻿
设备启动后，使用命令"df -h"：
﻿
[root@luckfox ]# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               215.5M     73.1M    107.9M  40% /
devtmpfs                 90.6M         0     90.6M   0% /dev
tmpfs                    90.7M         0     90.7M   0% /dev/shm
tmpfs                    90.7M    392.0K     90.3M   0% /tmp
tmpfs                    90.7M    216.0K     90.5M   0% /run
/dev/block/by-name/userdata
                        540.0K     74.0K         0 100% /userdata
/dev/block/by-name/oem
                         18.3M     18.3M         0 100% /oem

[image]

/dev/root 只有215.5M，并没有56G。
使用默认的6G(rootfs)，也是一样的。
﻿看起来像是配置了256M的flash？

编译生成的文件“.env.txt”和“sd_update.txt”好像也是OK的。
.env.txt：
blkdevparts=mmcblk1:32K(env),512K@32K(idblock),256K(uboot),32M(boot),512M(oem),256M(userdata),**56G**(rootfs)
sys_bootargs= root=/dev/mmcblk1p7 rootfstype=ext4 rk_dma_heap_cma=66M
sd_parts=mmcblk0:16K@512(env),512K@32K(idblock),4M(uboot)
﻿
sd_update.txt：size：**0xE00000000 = 56G**
#env.img 0x0:0x0 0x40:0x8000 0x40:0x8000
mw.b ${ramdisk_addr_r} 0xff 0x8000; fatload mmc 1 ${ramdisk_addr_r} env.img; mmc write ${ramdisk_addr_r} 0x0 0x40;
﻿
#idblock.img 0x40:0x8000 0x400:0x80000 0x170:0x2E000
mw.b ${ramdisk_addr_r} 0xff 0x2E000; fatload mmc 1 ${ramdisk_addr_r} idblock.img; mmc write ${ramdisk_addr_r} 0x40 0x170;
﻿
#uboot.img 0x440:0x88000 0x200:0x40000 0x200:0x40000
mw.b ${ramdisk_addr_r} 0xff 0x40000; fatload mmc 1 ${ramdisk_addr_r} uboot.img; mmc write ${ramdisk_addr_r} 0x440 0x200;
﻿
#boot.img 0x640:0xC8000 0x10000:0x2000000 0x180C:0x301800
mw.b ${ramdisk_addr_r} 0xff 0x301800; fatload mmc 1 ${ramdisk_addr_r} boot.img; mmc write ${ramdisk_addr_r} 0x640 0x180C;
﻿
#oem.img 0x10640:0x20C8000 0x100000:0x20000000 0x12338:0x2467000
mw.b ${ramdisk_addr_r} 0xff 0x2467000; fatload mmc 1 ${ramdisk_addr_r} oem.img; mmc write ${ramdisk_addr_r} 0x10640 0x12338;
﻿
#userdata.img 0x110640:0x220C8000 0x80000:0x10000000 0x4C4A:0x989400
mw.b ${ramdisk_addr_r} 0xff 0x989400; fatload mmc 1 ${ramdisk_addr_r} userdata.img; mmc write ${ramdisk_addr_r} 0x110640 0x4C4A;
﻿
#rootfs.img 0x190640:0x320C8000 0x7000000:**0xE00000000** 0xF5C18:0x1EB83000
mw.b ${ramdisk_addr_r} 0xff 0x1EB83000; fatload mmc 1 ${ramdisk_addr_r} rootfs.img; mmc write ${ramdisk_addr_r} 0x190640 0xF5C18;
﻿
% <- this is end of file symbol
﻿
﻿
但是，使用SDK Version：1.3是OK的。

[root@luckfox ]# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                54.9G    179.3M     52.4G   0% /
devtmpfs                 90.6M         0     90.6M   0% /dev
tmpfs                    90.7M         0     90.7M   0% /dev/shm
tmpfs                    90.7M    172.0K     90.6M   0% /tmp
tmpfs                    90.7M    568.0K     90.2M   1% /run
/dev/mmcblk1p6          239.9M      2.0M    232.4M   1% /mnt/sdcard
/dev/block/by-name/userdata
                        239.9M      2.0M    232.4M   1% /userdata
/dev/block/by-name/oem
                        487.9M     28.0M    453.0M   6% /oem

[image]

SDK1.3的时候，Luckfox_Pico_Max和Pro好像使用相同的固件。

luckfox-eng33, 2025-01-14:
谢谢你的反馈，工程师已经在 #230 中解决了这个问题，请重新拉取最新SDK进行尝试

jinxfzz, 2025-01-14:
谢谢及时处理，目前OK了。
[image]

closed 2025-01-14 by jinxfzz