LuckfoxTECH/luckfox-pico #232: How can I change default 8G emmc to 4G emmc in sdk configs?

smzahraee, 2025-01-26:
I need to change the emmc size from 8G to 4G, where should I change the settings?

luckfox-eng33, 2025-01-26:
There are no 4G emmc products among the products we sell, so it makes no sense to switch to 4G.

KunYi, 2025-07-10:
just modify  RK_PARTITION_CMD_IN_ENV in your board config to change partition size

kunyi@kunyi-dev:/tmp/luckfox-pico$ git diff 
diff --git a/project/cfg/BoardConfig_IPC/BoardConfig-SD_CARD-Buildroot-RV1106_Luckfox_Pico_Pro_Max-IPC.mk b/project/cfg/BoardConfig_IPC/BoardConfig-SD_CARD-Buildroot-RV1106_Luckfox_Pico_Pro_Max-IPC.mk
index b9d6e5d39..bf388b847 100755
--- a/project/cfg/BoardConfig_IPC/BoardConfig-SD_CARD-Buildroot-RV1106_Luckfox_Pico_Pro_Max-IPC.mk
+++ b/project/cfg/BoardConfig_IPC/BoardConfig-SD_CARD-Buildroot-RV1106_Luckfox_Pico_Pro_Max-IPC.mk
@@ -35,7 +35,7 @@ export RK_UBOOT_DEFCONFIG_FRAGMENT=rk-emmc.config
 #       <partdef> := <size>[@<offset>](part-name)
 # Note:
 #   If the first partition offset is not 0x0, it must be added. Otherwise, it needn't adding.
-export RK_PARTITION_CMD_IN_ENV="32K(env),512K@32K(idblock),256K(uboot),32M(boot),512M(oem),256M(userdata),6G(rootfs)"
+export RK_PARTITION_CMD_IN_ENV="32K(env),512K@32K(idblock),256K(uboot),32M(boot),512M(oem),256M(userdata),2G(rootfs)"