LuckfoxTECH/luckfox-pico #233: Core1106 pin number mismatch & pad layout

DansDesigns, 2025-01-30:
Hey, really liking the features of the Core1106 and I would like to use it in my project for a replacement of the NanoPi Neo Air..

I'm having some issues making a footprint for the Core1106 in EasyEDA Pro, the wiki says the module has 120 pins however when counting the pins in the footprint.pdf and on the Core1106-PinOut.xls spreadsheet, the total number is 112 - which one is correct?

Also, what is the recommended pad size and layout for a carrier PCB?

I'm thinking of making a cutout on the PCB and using castellated holes to mount the Core1106 flush with my carrier board,
would this be acceptable or do you have another recommendation?

Thank you
Kind Regards

DansDesigns, 2025-02-01:
I've gone with the pcb layout from the pdf - 112 pins

closed 2025-02-01 by DansDesigns

luckfox-eng33, 2025-02-05:
It seems you have found the answer, wishing you a happy Chinese New Year