LuckfoxTECH/luckfox-pico #234: Reset & Boot Button Layout

DansDesigns, 2025-02-01:
I'm designing a board the uses the Core1106 and i'd like to understand the requirements for the Reset and Boot buttons.

I notice one goes to pin 26 (SARADC_IN0) according to the schematic for the LuckFox Pico Ultra, this is called "Recovery", 
The other button goes to pin 74 (NPOR), again according to the schematic pdf for the Pico Ultra, this is called "NPOR".

Which one (NPOR or Recovery) is BOOT and which is RESET?

I see in the same schematic that the SARADC_IN0 must always be pulled high, so i'm guessing that by pulling this low, it Resets the device?

Am I also correct in assuming that the USB_VBUSDET must have 3v3 the entire time it is needed to be in USB Device Mode?

DansDesigns, 2025-02-01:
ok so just found the CAD drawings for the Pico Ultra, comparing with the schematic pdf I have found that NPOR is RESET and GPIO4_C0_z (SARADC_IN0) is BOOT

closed 2025-02-01 by DansDesigns

luckfox-eng33, 2025-02-05:
It seems you have found the answer